openwrt/openwrt #18259: Realtek  kmod-r8168 with low uploading speed 800Mbps but  ssh command modprobe r8168 manually will have 920Mbps

heygo1345678, 2025-03-16:
### Describe the bug

my NIC is  RTL8168evl/8111evl
with below dmesg:
Sun Mar 16 11:26:17 2025 kern.info kernel: [    7.823992] r8168 Gigabit Ethernet driver 8.054.00-RSS loaded
Sun Mar 16 11:26:17 2025 kern.info kernel: [    7.827414] r8168 518b:00:00.0: enabling device (0000 -> 0002)
Sun Mar 16 11:26:17 2025 kern.notice kernel: [    7.918058] r8168 518b:00:00.0 eth1 (uninitialized): no netdev found for vf serial:34118
Sun Mar 16 11:26:17 2025 kern.notice kernel: [    7.918627] r8168 518b:00:00.0 eth1: no netdev found for vf serial:34118
Sun Mar 16 11:26:17 2025 kern.info kernel: [    7.919158] r8168: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
Sun Mar 16 11:26:17 2025 kern.warn kernel: [    7.919982] r8168  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com>
Sun Mar 16 11:26:21 2025 kern.info kernel: [   13.174450] r8168: eth1: Link is Up - 1Gbps/Full

there is a strange bug:
Realtek  kmod-r8168 with low  iperf3 uploading speed 800Mbps:
root@OpenWrt:~# iperf3 -c 192.168.85.111 -t 1000
Connecting to host 192.168.85.111, port 5201
[  5] local 192.168.85.40 port 48566 connected to 192.168.85.111 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  96.2 MBytes   807 Mbits/sec    0    479 KBytes       
[  5]   1.00-2.00   sec  95.6 MBytes   802 Mbits/sec    0    635 KBytes       
[  5]   2.00-3.00   sec  96.0 MBytes   805 Mbits/sec    0    673 KBytes       
[  5]   3.00-4.00   sec  95.5 MBytes   801 Mbits/sec    0    707 KBytes       
[  5]   4.00-5.00   sec  96.0 MBytes   805 Mbits/sec    0    707 KBytes       
[  5]   5.00-6.00   sec  94.4 MBytes   792 Mbits/sec    0    707 KBytes       
[  5]   6.00-7.00   sec  96.2 MBytes   807 Mbits/sec    0    707 KBytes       
[  5]   7.00-8.00   sec  94.5 MBytes   793 Mbits/sec    0    707 KBytes       
^C[  5]   8.00-8.87   sec  83.5 MBytes   804 Mbits/sec    0    707 KBytes    

after doing below:
root@OpenWrt:~# nano /etc/modules.d/r8168
blacklist r8168

then login ssh with execute  manually
root@OpenWrt:~# modprobe r8168

then the speed  look like normally
root@OpenWrt:~# iperf3 -c 192.168.85.111 -t 1000 
Connecting to host 192.168.85.111, port 5201
Reverse mode, remote host 192.168.85.111 is sending
[  5] local 192.168.85.40 port 45338 connected to 192.168.85.111 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   104 MBytes   874 Mbits/sec                  
[  5]   1.00-2.00   sec   112 MBytes   935 Mbits/sec                  
[  5]   2.00-3.00   sec   112 MBytes   936 Mbits/sec                  
[  5]   3.00-4.00   sec   111 MBytes   933 Mbits/sec                  
[  5]   4.00-5.00   sec   112 MBytes   936 Mbits/sec                  
[  5]   5.00-6.00   sec   112 MBytes   935 Mbits/sec                  
[  5]   6.00-7.00   sec   112 MBytes   937 Mbits/sec                  
[  5]   7.00-8.00   sec   112 MBytes   936 Mbits/sec                  
[  5]   8.00-9.00   sec   112 MBytes   937 Mbits/sec                  
[  5]   9.00-10.00  sec   112 MBytes   935 Mbits/sec    

### OpenWrt version

24.10.0

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

x86_64

### Device

RTL8168evl/8111evl

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Noltari, 2025-03-16:
This issue should be reported to Realtek.

You can also use kmod-r8169 instead.

brada4, 2025-03-16:
Gigabit is full duplex, should be 950 in and 950 out with tcp (iperf3 --bidir)