openwrt/openwrt #18284: [MR7350] Ethernet fails to negotiate gigabit, any-duplex, any port

starfoxdot64, 2025-03-17:
### Describe the bug

This issue is from pre-snapshot releases up to now, where the ethernet ports all refuse to negotiate anything higher than 100Mb/s, usually deferring to 10Mb/s. Initially, it was thought that this was because of a VLAN tagging issue, but using another device to test post-reset, it appears this may be a hardware/driver issue.

### OpenWrt version

r29028-adc4d95c74

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq60xx

### Device

Linksys MR7350

### Image kind

Official downloaded image

### Steps to reproduce

Reset the device to defaults, then connect it to the internet, or another ethernet device capable of using gigabit speeds. Check the connection with ethtool on the router's end, and check the speed of the connection on your device.

### Actual behaviour

The router's ports are hard-locked at 100Mb/s on WAN and 10Mb/s on LAN, while advertising up to 1000Mb/s full-duplex. When trying to manually set the speed of the port (eg. `ethtool -s lan1 autoneg on speed 1000 duplex full`), the router disconnects as expected, but refuses to communicate on that port again at the requested speed. 

During this time, the TX/RX lights of said port will blink on and off every few seconds, at the same time.

### Expected behaviour

The router should autonegotiate with the client device over ethernet, picking the fastest available speed based on advertised link modes from both it and the client.

### Additional info

It appears that dmesg and logread do not provide sufficient information to debug this, at least not myself. All they do is show the PHY speed flip-flopping between 100Mb/s and 10Mb/s.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

starfoxdot64, 2025-03-17:
Something I also forgot to mention, when referring to "pre-snapshot releases", I'm referring to an unsupported form called BackspaceWRT (now defunct for this model as of snapshot support being released), which touted support for this hardware in the months before official snapshots were announced. 

This issue was also present on the stable builds provided by that unsupported fork as well, leading me to believe it could be an issue with the Qualcomm QCA8075's implementation? That's above my knowledge, so I could be incorrect, but it's a good place to start looking, I think.

brada4, 2025-03-19:
So - no problem with OpenWrt whatsoever?

starfoxdot64, 2025-03-19:
> So - no problem with OpenWrt whatsoever?

I'm... not certain. On stock firmware, gigabit speeds work on all ports, so I'm not really sure what's going on here.

brada4, 2025-03-19:
Please get ethtool and mii-tool -v ooutputs from misnegotiated port(s) using official release
https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=qualcommax%2Fipq60xx&id=linksys_mr7350

starfoxdot64, 2025-03-21:
Sorry about the wait.

lan1:

```root@OpenWrt:~# ethtool lan1
Settings for lan1:

	Supported ports: [ TP MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full 
	Link partner advertised pause frame use: Symmetric
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 10Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 0
	Transceiver: external
	Auto-negotiation: on
	MDI-X: on (auto)
	Link detected: yes
root@OpenWrt:~# mii-tool lan1
lan1: negotiated 10baseT-FD flow-control, link ok```

lan2:

```root@OpenWrt:~# ethtool lan2
Settings for lan2:

	Supported ports: [ TP MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Supported pause frame use: Symmetric Receive-only
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full 
	Link partner advertised pause frame use: Symmetric
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 10Mb/s
	Duplex: Full
	Port: Twisted Pair
	PHYAD: 1
	Transceiver: external
	Auto-negotiation: on
	MDI-X: on (auto)
	Link detected: yes
root@OpenWrt:~# mii-tool lan2
lan2: negotiated 10baseT-FD flow-control, link ok```