openwrt/openwrt #18285: GL.iNet GL-MT3000 randomly rebooting in 24.10.0

mdimjasevic, 2025-03-17:
### Describe the bug

My GL.iNet GL-MT3000 router randomly reboots.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT3000

### Image kind

Official downloaded image

### Steps to reproduce

1. Flash GL.iNet GL-MT3000 with OpenWRT 24.10.0,
2. Connect with WiFi clients to an access point on the device,
3. Surf the web on the clients,
4. Observe a connection to the route rdrop on the clients.
5. Log in via LUCI to the router once it is available (after a restart) and observe that the uptime was reset less than a minute ago.

### Actual behaviour

The router restarts randomly for no apparent reason.

### Expected behaviour

The router does not reboot and keeps on serving WiFi clients.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-03-18:
This doesn't seem to be a widely reported problem with this device on the OpenWrt forums.

Is the problem apparent with 23.05.5?

This device apparently has a fan on the SoC - you might want to check that it is functioning and that your device isn't overheating.

mdimjasevic, 2025-03-18:
Hi @aimacintyre !

> Is the problem apparent with 23.05.5?
> 

I wouldn't know. I tested it with the latest stable release.

> This device apparently has a fan on the SoC - you might want to check that it is functioning and that your device isn't overheating.

I cannot find a mention of a fan anywhere in the device specs. Can you say more? How do I check its status? At `/sys/class/thermal/` I see three cooling devices named `cooling_device<0,1,2>`. Are they related?

lukasz1992, 2025-03-18:
Please check the crashlog if it is present (/sys/kernel/debug/crashlog or /sys/fs/pstore/*)

mdimjasevic, 2025-03-18:
The `/sys/kernel/debug/crashlog` does not exist on my router and the `/sys/fs/pstore/` folder is empty.

aimacintyre, 2025-03-19:
> > Is the problem apparent with 23.05.5?
> 
> I wouldn't know. I tested it with the latest stable release.

It wouldn't take much effort to test.

> I cannot find a mention of a fan anywhere in the device specs. Can you say more? How do I check its status? At `/sys/class/thermal/` I see three cooling devices named `cooling_device<0,1,2>`. Are they related?

[OEM docs](https://docs.gl-inet.com/router/en/4/interface_guide/system_overview/#cpu-average-load)
[forum thread](https://forum.openwrt.org/t/beryl-ax-and-openwrt-and-fan/192016)
[fan control](https://forum.gl-inet.com/t/beryl-ax-cooling-fan-control-in-vanilla-openwrt/38756/18) (24.10.0)

fildunsky, 2025-03-27:
I had this issue on stock factory ROMs sometimes. I am not really sure 100% but I thought it was because of a common issue with Filogic, and the workaround for me was adding to autorun:
```
ethtool -K eth0 tso off
```
For WAN port. But this thing should be fixed in 24.10.0 (at least I've heard so). Sorry, if it's irrelevant for you.

xirixiz, 2025-05-06:
I've experienced the same behaviour, validated with mulitple different power adapaters.
Now I upgraded to 24.10.1 with a factory reset, the problem seems to be resolved.

svilenstefanov, 2025-11-10:
I am experiencing same problem since updated to 24.10.0
Today updated to 24.10.4 and it is getting better. Still few restarts here and there. Now I am looking how to downgrade to 23