openwrt/openwrt #18286: dnsmasq-full crashes when ipset exceeds 62 entries

TheSainEyereg, 2025-03-17:
### Describe the bug

I need to add domains for routing via VPN (since my country is under sanctions there are a lot of them), however when the number of these domains exceeds 62, dnsmasq refuses to start.

Domain lists were made via `ipset` using [this guide (it's in russian)](https://itdog.info/tochechnaya-marshrutizaciya-po-domenam-na-routere-s-openwrt/#%D1%81%D0%B2%D0%BE%D0%B8-%D0%B4%D0%BE%D0%BC%D0%B5%D0%BD%D1%8B)

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

1. Create ipset for dnsmasq
2. Add 62 + 1 domains to it
3. `service dnsmasq restart` 

### Actual behaviour

dnsmasq refuses to start

### Expected behaviour

Everything works as expected

### Additional info

Error log:
```
root@OpenWrt:~# logread -f
Mon Mar 17 23:18:19 2025 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Mon Mar 17 23:18:24 2025 daemon.crit dnsmasq[1]: bad option at line 43 of /var/etc/dnsmasq.conf.cfg01411c
Mon Mar 17 23:18:24 2025 daemon.crit dnsmasq[1]: FAILED to start up
Mon Mar 17 23:18:29 2025 daemon.crit dnsmasq[1]: bad option at line 43 of /var/etc/dnsmasq.conf.cfg01411c
Mon Mar 17 23:18:29 2025 daemon.crit dnsmasq[1]: FAILED to start up
Mon Mar 17 23:18:34 2025 daemon.crit dnsmasq[1]: bad option at line 43 of /var/etc/dnsmasq.conf.cfg01411c
Mon Mar 17 23:18:34 2025 daemon.crit dnsmasq[1]: FAILED to start up
Mon Mar 17 23:18:39 2025 daemon.crit dnsmasq[1]: bad option at line 43 of /var/etc/dnsmasq.conf.cfg01411c
Mon Mar 17 23:18:39 2025 daemon.crit dnsmasq[1]: FAILED to start up
Mon Mar 17 23:18:44 2025 daemon.crit dnsmasq[1]: bad option at line 43 of /var/etc/dnsmasq.conf.cfg01411c
Mon Mar 17 23:18:44 2025 daemon.crit dnsmasq[1]: FAILED to start up
Mon Mar 17 23:18:49 2025 daemon.crit dnsmasq[1]: bad option at line 43 of /var/etc/dnsmasq.conf.cfg01411c
Mon Mar 17 23:18:49 2025 daemon.crit dnsmasq[1]: FAILED to start up
Mon Mar 17 23:18:49 2025 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
```

`/etc/config/dhcp`:
[dhcp.txt](https://github.com/user-attachments/files/19302116/dhcp.txt)

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2025-03-18:
The tipping point is when the total length of the nftset directive in dnsmasq.conf exceeds 1025 characters.

See https://github.com/openwrt/openwrt/issues/17548

TheSainEyereg, 2025-03-18:
Oh, I can see it now. Is there any workaround for this limitation? As I see the string can be split into several nftsets, can this be automated?