openwrt/openwrt #18287: OpenWrt Fails to Update IPv6 Address Immediately After ONT Redials, Causing Temporary Disruptions

daifeilail, 2025-03-18:
### Describe the bug

Issue Description
In a setup where OpenWrt acts as a router and connects via PPPoE through an Optical Network Terminal (ONT), the ONT periodically redials, resulting in a change of the assigned IPv6 address. However, OpenWrt does not immediately update its IPv6 address until the DHCP lease expires, leading to a temporary loss of IPv6 connectivity.

Since applications automatically fall back to IPv4 when IPv6 is unavailable, this issue manifests as degraded network performance, especially in scenarios where IPv6 is used for optimized routing (e.g., certain CDN resolutions or international connections).

Steps to Reproduce
The ONT periodically redials via PPPoE, receiving a new IPv6 address.
OpenWrt retains the old IPv6 address until the DHCP lease expires.
During this period, IPv6 connectivity is lost, forcing traffic to fallback to IPv4, resulting in slower network performance.
Expected Behavior
When the ONT redials and receives a new IPv6 address, OpenWrt should immediately detect the change and update its IPv6 configuration without waiting for the DHCP lease to expire.

Actual Behavior
OpenWrt does not refresh its IPv6 address until the DHCP lease timeout, causing a temporary IPv6 outage.

Possible Solutions
Monitor PPPoE redial events and trigger the DHCPv6 client to renew the address.
Detect IPv6 unreachability and force an immediate DHCPv6 renewal.
Implement a more intelligent IPv6 address change detection mechanism to prevent waiting for the lease timeout.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

r23809-234f1a2efa

### OpenWrt target/subtarget

r23809-234f1a2efa

### Device

cat /tmp/sysinfo/model

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-03-21:
How does ont signal openwrt about routing change?

lylus, 2025-03-21:
I have been able to reproduce this issue in all versions of OpenWrt from 19.x to 24.x, but I found that it has rarely been mentioned online. When the ISP disconnects a user's PPPoE connection and the user attempts to reconnect via PPPoE, if the connection is established too quickly, the IPv6 interface does not have enough time to close before the PPPoE dial-up succeeds. As a result, the system fails to obtain an IPv6 address, and the newly acquired IPv4 gateway is not added to OpenWrt's default routing table. This ultimately causes the entire local network to lose internet connectivity. The specific reasons and solutions are detailed in the following link:  
[GitHub Issue](https://github.com/hanwckf/immortalwrt-mt798x/issues/57)

GoetzGoerisch, 2025-03-21:
Could you provide a PR with the fix described in the issue towards an OpenWrt fork?

kaedeair, 2025-05-10:
> Please report the correct version and target/subtarget you are using.
> 
> Furthermore post the output of and detail who is establishing the PPPoE session?`/etc/config/network`
Hope to helpful
version: 21.02.7 arch: x86_64
```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option packet_steering '1'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'
	list ports 'eth2'
	list ports 'eth3'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '172.16.0.1'
	option netmask '255.255.128.0'
	option ip6assign '64'

config interface 'wan'
	option device 'eth1'
	option proto 'pppoe'
	option username '<user>'
	option password '<password>'
	option ipv6 'auto'

```

```
Sep 20 02:31:08 OpenWrt pppd[29425]: rcvd [LCP TermReq id=0x3]
Sep 20 02:31:08 OpenWrt pppd[29425]: LCP terminated by peer
Sep 20 02:31:08 OpenWrt pppd[29425]: Connect time 10080.2 minutes.
Sep 20 02:31:08 OpenWrt pppd[29425]: Sent 2883250556 bytes, received 950250707 bytes.
Sep 20 02:31:08 OpenWrt pppd[29425]: Script /lib/netifd/ppp-down started (pid 13431)
Sep 20 02:31:08 OpenWrt odhcp6c[29669]: Failed to send RS (Permission denied)
Sep 20 02:31:08 OpenWrt odhcp6c[29669]: Failed to send RELEASE message to ff02::1:2 (Permission denied)
Sep 20 02:31:08 OpenWrt netifd: Network device 'pppoe-wan' link is down
Sep 20 02:31:08 OpenWrt netifd: Network alias 'pppoe-wan' link is down
Sep 20 02:31:08 OpenWrt netifd: Interface 'wan_6' has link connectivity loss
Sep 20 02:31:08 OpenWrt pppd[29425]: Script /lib/netifd/ppp-down started (pid 13433)
Sep 20 02:31:08 OpenWrt pppd[29425]: sent [LCP TermAck id=0x3]
Sep 20 02:31:08 OpenWrt pppd[29425]: Modem hangup
Sep 20 02:31:08 OpenWrt pppd[29425]: Connection terminated.
Sep 20 02:31:08 OpenWrt pppd[29425]: Connect time 10080.2 minutes.
Sep 20 02:31:08 OpenWrt pppd[29425]: Sent 2883250556 bytes, received 950250707 bytes.
Sep 20 02:31:08 OpenWrt netifd: Interface 'wan' has lost the connection
Sep 20 02:31:08 OpenWrt netifd: wan_6 (29669): Command failed: Permission denied
Sep 20 02:31:08 OpenWrt pppd[29425]: Send PPPOE Discovery V1T1 PADT session 0x9c21 length 8
Sep 20 02:31:08 OpenWrt pppd[29425]:  dst ${MAC-BRAS}  src ${MAC-ROUTER}
Sep 20 02:31:08 OpenWrt pppd[29425]:  [host-uniq  f1 72 00 00]
Sep 20 02:31:08 OpenWrt pppd[29425]: Sent PADT
Sep 20 02:31:08 OpenWrt pppd[29425]: Script /lib/netifd/ppp-down finished (pid 13431), status = 0x1
Sep 20 02:31:08 OpenWrt pppd[29425]: Script /lib/netifd/ppp-down finished (pid 13433), status = 0x1
Sep 20 02:31:08 OpenWrt pppd[29425]: Exit.
Sep 20 02:31:08 OpenWrt netifd: Interface 'wan' is now down
Sep 20 02:31:08 OpenWrt netifd: Interface 'wan' is setting up now
Sep 20 02:31:08 OpenWrt insmod: module is already loaded - slhc
Sep 20 02:31:08 OpenWrt insmod: module is already loaded - ppp_generic
Sep 20 02:31:08 OpenWrt insmod: module is already loaded - pppox
Sep 20 02:31:08 OpenWrt insmod: module is already loaded - pppoe
[... 43 lines cut ...]
Sep 20 02:31:08 OpenWrt pppd[13560]: rcvd [IPCP ConfNak id=0x1 <addr ${IP-ADDR}> <ms-dns1 ${DNS-SERVER}> <ms-dns2 ${DNS-SERVER}>]
Sep 20 02:31:08 OpenWrt pppd[13560]: sent [IPCP ConfReq id=0x2 <addr ${IP-ADDR}> <ms-dns1 ${DNS-SERVER}> <ms-dns2 ${DNS-SERVER}>]
Sep 20 02:31:08 OpenWrt pppd[13560]: rcvd [IPV6CP ConfAck id=0x1 <addr ${LOCAL-LL-IPV6}>]
Sep 20 02:31:08 OpenWrt pppd[13560]: local  LL address ${LOCAL-LL-IPV6}
Sep 20 02:31:08 OpenWrt pppd[13560]: remote LL address ${Remote-LL-IPV6}
Sep 20 02:31:08 OpenWrt pppd[13560]: Script /lib/netifd/ppp6-up started (pid 13709)
Sep 20 02:31:08 OpenWrt netifd: Network device 'pppoe-wan' link is up
Sep 20 02:31:08 OpenWrt netifd: Network alias 'pppoe-wan' link is up
Sep 20 02:31:08 OpenWrt netifd: Interface 'wan_6' has link connectivity 
Sep 20 02:31:08 OpenWrt netifd: Interface 'wan' is now up
Sep 20 02:31:08 OpenWrt pppd[13560]: rcvd [IPCP ConfAck id=0x2 <addr ${IP-ADDR}> <ms-dns1 ${DNS-SERVER}> <ms-dns2 ${DNS-SERVER}>]
Sep 20 02:31:08 OpenWrt pppd[13560]: local  IP address ${IP-ADDR}
Sep 20 02:31:08 OpenWrt pppd[13560]: remote IP address ${GateWay}
Sep 20 02:31:08 OpenWrt pppd[13560]: primary   DNS address ${DNS-SERVER}
Sep 20 02:31:08 OpenWrt pppd[13560]: secondary DNS address ${DNS-SERVER}
Sep 20 02:31:08 OpenWrt pppd[13560]: Script /lib/netifd/ppp-up started (pid 13716)
Sep 20 02:31:08 OpenWrt pppd[13560]: Script /lib/netifd/ppp-up finished (pid 13716), status = 0x1
Sep 20 02:31:08 OpenWrt pppd[13560]: Script /lib/netifd/ppp6-up finished (pid 13709), status = 0x0
Sep 20 02:31:08 OpenWrt firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sep 20 02:31:08 OpenWrt odhcp6c[29669]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Sep 20 02:31:08 OpenWrt netifd: Interface 'wan_6' is now down
Sep 20 02:31:08 OpenWrt netifd: Interface 'wan_6' is disabled
Sep 20 02:31:08 OpenWrt netifd: Interface 'wan_6' has link connectivity loss
```
[image]