openwrt/openwrt #18293: openwrt x86 24.10 board.json detection issue

Nit3H8wk, 2025-03-18:
### Describe the bug

I have a CWWK x86 router with 8x intel 226-V nic's. Only 4 of them appear in port status but all of them appear in interfaces/devices. I did have to replace the expansion card with 4 of the 226-v nics as the pre-installed one was defective. I was able to manually edit the board.json and all 8 ports are working now. The pre-installed expansion card was rev 1.0 and the replacement rev 1.2.

[image]

### OpenWrt version

24.10

### OpenWrt release

24.10

### OpenWrt target/subtarget

x86/64

### Device

CWWK micro firewall appliance

### Image kind

Official downloaded image

### Steps to reproduce

Stock openwrt 24.10 x86 install with expansion card rev 1.2 which is 4x intel 226-v nics.

### Actual behaviour

Only 4x ethernet ports show up in port status but all are listed in devices.

### Expected behaviour

eth 0 to 7 in port status under luci overview.

### Additional info

The board.json I edited to make all 8 ports show correctly.

[board.json](https://github.com/user-attachments/files/19325107/board.json)

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Nit3H8wk, 2025-03-18:
> Invalid Release reported. `24.10` Is this from a clean repository.
Yes it was directly from the openwrt website. OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.076.76243~9c000ff

brada4, 2025-03-19:
Edit post to correct please