openwrt/openwrt #18308: Cannot create a new AP for a usb dongle

heeae, 2025-03-21:
I have got a strange problem on adding a new AP for a usb dongle 

the device did support AP mode, confirmed from iw list

I have a few years old FriendlyElec NanoPi R5C

While i want to create another AP network but got strange issue, hostapd cannot start 

I tried to plug in the dongle, go to Network => Wireless, follow the normal procedure, Click "Add", set certain the security setup and save then apply changes, it cannot start, 
I tried to plug in again, there have a auto AP created, i just enabled, it also cannot start. 

checking the log, the key returned is "hostapd: Could not read interface phy2-ap0 flags: No such device" and it then failed

My version was 22.X.X, and then flash the new object again with 24.10.0, but staill failed
i tried 3 more dongles still got similar result 

below is the current firmware version
Firmware version: OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI 7b0663a5557118499dc3b3d44550efc1b6fa3feb branch 25.052.52917~86b595b
Kernel Version: 6.1.99

below is the /var/run/hostapd-phy2.conf write
`driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=CN
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
stationary_ap=1
channel=36
chanlist=36

#num_global_macaddr=1
#macaddr_base=

interface=phy2-ap0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
wpa_passphrase=password
wpa_psk_file=/var/run/hostapd-phy2-ap0.psk
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=FriendlyWrt-f2:69:6e
bridge=br-lan
wds_bridge=
snoop_iface=br-lan
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
dynamic_vlan=0
vlan_naming=1
vlan_no_bridge=1
vlan_file=/var/run/hostapd-phy2-ap0.vlan
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
bssid=00:00:00:00:00:00
#default_macaddr`

below is the server log when try to start the device

`Fri Mar 21 09:33:26 2025 daemon.notice wpa_supplicant[1658]: Set new config for phy phy2
Fri Mar 21 09:33:26 2025 daemon.notice wpa_supplicant[1658]: Set new config for phy phy2
Fri Mar 21 09:33:26 2025 daemon.notice hostapd: Set new config for phy phy2: /var/run/hostapd-phy2.conf
Fri Mar 21 09:33:26 2025 daemon.notice hostapd: Restart interface for phy phy2
Fri Mar 21 09:33:26 2025 daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 country_code=CN ieee80211d=1 ieee80211h=1 hw_mode=a beacon_int=100 stationary_ap=1 chanlist=36 #num_global_macaddr=1 #macaddr_base= channel=36  interface=phy2-ap0 bssid=00:00:00:00:00:00 ctrl_interface=/var/run/hostapd ap_isolate=1 bss_load_update_period=60 chan_util_avg_period=600 disassoc_low_ack=1 skip_inactivity_poll=0 preamble=1 wmm_enabled=1 ignore_broadcast_ssid=0 uapsd_advertisement_enabled=1 utf8_ssid=1 multi_ap=0 wpa_passphrase=password wpa_psk_file=/var/run/hostapd-phy2-ap0.psk auth_algs=1 wpa=2 wpa_pairwise=CCMP ssid=FriendlyWrt-f2:69:6e bridge=br-lan wds_bridge= snoop_iface=br-lan wpa_disable_eapol_key_retries=0 wpa_key_mgmt=WPA-PSK okc=0 disable_pmksa_caching=1 dynamic_vlan=0 vlan_naming=1 vlan_no_bridge=1 vlan_file=/var/run/hostapd-phy2-ap0.vlan qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56 #default_macaddr nas_identifier=0000000
Fri Mar 21 09:33:26 2025 daemon.err hostapd: Could not read interface phy2-ap0 flags: No such device
Fri Mar 21 09:33:26 2025 daemon.notice hostapd: nl80211: Driver does not support authentication/association or connect commands
Fri Mar 21 09:33:26 2025 daemon.notice hostapd: nl80211: deinit ifname=phy2-ap0 disabled_11b_rates=0
Fri Mar 21 09:33:26 2025 daemon.err hostapd: Could not read interface phy2-ap0 flags: No such device
Fri Mar 21 09:33:26 2025 daemon.err hostapd: nl80211 driver initialization failed.
Fri Mar 21 09:33:26 2025 daemon.notice hostapd: phy2-ap0: CTRL-EVENT-TERMINATING
Fri Mar 21 09:33:26 2025 daemon.err hostapd: hostapd_free_hapd_data: Interface phy2-ap0 wasn't started
Fri Mar 21 09:33:26 2025 daemon.notice hostapd: hostapd.add_iface failed for phy phy2 ifname=phy2-ap0
Fri Mar 21 09:33:26 2025 daemon.notice netifd: radio1 (11324): command failed: No such device (-19)
Fri Mar 21 09:33:26 2025 daemon.notice netifd: Wireless device 'radio1' is now up
Fri Mar 21 09:33:28 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Fri Mar 21 09:33:28 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Fri Mar 21 09:33:28 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses`

brada4, 2025-03-21:
Please format your pist, add `ubus call system board` output and lsusb output from the "dongle"

heeae, 2025-03-21:
> Please format your pist, add `ubus call system board` output and lsusb output from the "dongle"

the "Code" seems not listen to me, i try to format again with quote

and also included the lsusb and ubus result

Shine-, 2025-03-21:
`"kernel": "6.1.99",`
[...]
`"version": "24.10.0",`

OpenWrt 24.10.0 is using kernel 6.6, not 6.1. If you are using a modified OpenWrt from your hardware vendor, please contact your hardware vendor instead, or whoever provided the modified OpenWrt to you.

You're not specifying the WiFi dongle model. From your `lsusb` output, USB ID 0bda:c811 should be a RTL8811CU or RTL8812CU, which is not supported out of the box. You would need to install the proper third-party drivers - and again, you would have to contact whoever provided the drivers to you (which is not OpenWrt).

Also, the previous version you specify ("22.X.X") does not exist, so it was likely also a modified version. You best bet would be to approach whoever modified your previous version and ask them to provide a build that supports RTL8811CU/RTL8812CU.