openwrt/openwrt #18311: In the OpenWrt 24.10 version, when configuring IEEE80211r in STA mode, it fails to work.

loveepy, 2025-03-21:
When developing STA using OpenWrt 24.10 version, the configurations of ieee802r and ieee802k for wireless cannot be parsed into wpa_supplicant.conf. After adding the configurations into wpa_supplicant.conf manually, the STA fails to connect to the AP.Here is my configuration.
config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'wan'
	option mode 'sta'
	option ssid 'OpenWrt'
	option encryption 'psk2+ccmp'
	option key 'password'
	option bss_transition '1'
	option disassoc_low_ack '0'
	option time_advertisement '2'
	option wnm_sleep_mode '1'
	option wnm_sleep_mode_no_keys '1'
	option ieee80211r '1'
	option mobility_domain '1234'
	option ft_over_ds '0'
	option rrm_neighbor_report '1'
	option rrm_beacon_report '1'
	option bss_transition '1'
	option wnm_sleep_mode '1'
	option time_advertisement '2'
	option key_mgmt 'FT-PSK'

loveepy, 2025-03-21:
@brada4 Thank you for your answer. After configuring sta mode on luci, there is no option for ieee80211r. Does it mean that ieee80211r in sta mode is not supported for the time being? Thanks again for your help

brada4, 2025-03-21:
Care to fill the template? Hope you know supplicant and hostap do not share same parameter names.

brada4, 2025-03-22:
Cloše this maybe