openwrt/openwrt #18312: Read only on Ubiquiti NanoStation AC Loco

JohnyByk, 2025-03-21:
### Describe the bug

Installed OpenWRT on a Ubiquiti NanoStation AC Loco. I followed this guide: https://openwrt.org/toh/ubiquiti/common, but I compiled my own version with LuCI and Argon.

Procedure:
```
git clone https://github.com/openwrt/openwrt.git
cd openwrt

./scripts/feeds update -a
./scripts/feeds install -a

git clone https://github.com/jerrykuku/luci-theme-argon package/luci-theme-argon
git clone https://github.com/jerrykuku/luci-app-argon-config package/luci-app-argon-config
make menuconfig

make download
make
```

### OpenWrt version

23.05.3 and 24.10.0

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

Ubiquiti Nanostation AC Loco

### Image kind

Self-built image

### Steps to reproduce

Follow this guaide: https://openwrt.org/toh/ubiquiti/common and install any official image, try to save any settings and reboot

### Actual behaviour

I think so system is read only, you cant't update system, change settings - they dissapear after reboot
errors in log:
```
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.190177] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 00786500
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.198825] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 007899a8
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.224935] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 007899a8
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.233526] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 00786500
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.241989] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 007899a8
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.249961] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 007899a8
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.266586] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 007899a8
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.281290] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 00786500
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.289283] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 00786500
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.311251] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 007899a8
Wed Feb 26 22:46:47 2025 kern.warn kernel: [  928.319791] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 007899a8
```

### Expected behaviour

Saving settings should work, and the system update option should function properly.

### Additional info

I used not only the compiled imagesof but also the officials

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

ant-thomas, 2025-03-28:
I am having what looks like the same issue with settings not staying after a reboot.
NanoStation AC Loco
Flashed with 24.10.0 downloaded direct from this page - https://openwrt.org/toh/ubiquiti/nanostation_ac_loco
Not built/compiled by myself.

Many similar errors in dmesg
``jffs2: Node CRC ffffffff != calculated CRC openwrt``

ant-thomas, 2025-03-28:
What's the downgrade method to install a working openwrt version?
I've read you need to go into TFTP mode then install the factory firmware, then do the openwrt install again.
Struggling to get in to TFTP mode.

ant-thomas, 2025-03-28:
Ignore the above - I was accurate enough with the 8 second reset.

marek26340, 2025-05-07:
I am having a bit of trouble myself right now too, on the latest 24.10.1 version. I **really** hope that this pesky bug hasn't made a comeback. 😢

Deach01, 2025-05-25:
I would say this bug (that's been going on and off for 3 years or more) is back!

I too did a sysupgrade to 24.10.1 and killed my Litebeam 5ac ability to write.
I had to do the whole tftp back to factory and installed 23.05.4 and all is good again.

Perhaps a note in the device TOH notes or remove it from the supported list for anything but 23.05.4 until this ongoing issue is resolved permanently and it doesn't sneak back in on new releases.

Teklock, 2025-06-08:
I'm also seeing this same bug (won't save configs after reboot) with both of my NanoStation AC Loco devices flashed to 24.10.0 (I used the link in the Wiki).  Had to tftp back to stock and flashed to 23.05.5.  All is well now..

jmranger, 2025-06-25:
Confirmed present in 24.10.2. I should have searched in the issues before attempTting to upgrade from 23.5.5.

Could someone remove the "invalid" flag?

Duplicate of #17914

ant-thomas, 2025-09-02:
Who is brave enough to try?

JohnyByk, 2025-09-03:
Can I flash OpenWrt on a device with broken write support, or do I need to restore stock firmware first and then install OpenWrt?