openwrt/openwrt #18316: Fetching specific commit from feed fails

eric-j-ason, 2025-03-21:
### Describe the bug

Error while fetching a src-git feed that is locked to a specific commit.

### OpenWrt version

Not applicable. This is a build error.

### OpenWrt release

Not applicable. This is a build error.

### OpenWrt target/subtarget

Not applicable. This is a build error.

### Device

Indepented of device. This is a build error.

### Image kind

Self-built image

### Steps to reproduce

Create a `feeds.conf`, containing a feed locked to a specific commit:

```bash
echo 'src-git packages https://git.openwrt.org/feed/packages.git^9ddc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91' > feeds.conf
```

Update this feed:

```bash
./scripts/feeds clean
./scripts/feeds update packages
```

### Actual behaviour

Fetching the feed fails with an error:

> error: Server does not allow request for unadvertised object 9ddc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91

You end up with the feed being on the wrong commit.

### Expected behaviour

You get the right commit (9ddc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91 in this case) from the feed.

### Additional info

It seems to be caused by 32d0a57dc10811a24816e51f10e55963f40fe462, which aims to optimize feed fetching by not downloading the entire history. I suggest we revert it as well as the accompanying 9ec32cfb2733856a2ab4caee07d9b3297568381d, as a quick fix. They are nice optimizations, though, so I would like to have them back when they are sorted out.

@cedric-chedaleux: Are you able to reproduce this error?

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

eric-j-ason, 2025-03-21:
> Is this from a supported device?

It is for every device, as it pertains to fetching the feeds, before any particular device even enters the picture.

cedric-chedaleux, 2025-03-21:
Yes @eric-j-ason , I do reproduce the bug.

Changing `https://git.openwrt.org/feed/..` to `https://github.com/openwrt/` fixes the issue! It looks like a server limitation of the package feeds git mirror..

cedric-chedaleux, 2025-03-21:
@ynezz as noticed by @eric-j-ason , the shallow clone on a specific commit  https://github.com/openwrt/openwrt/commit/32d0a57dc10811a24816e51f10e55963f40fe462 does not work with `https://git.openwrt.org` but does work with `https://github.com/openwrt/`. Can `git.openwrt.org` configuration restrictions be changed to allow this?

ynezz, 2025-03-22:
> error: Server does not allow request for unadvertised object 9ddc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91

Can you provide complete log of git operations?

> You end up with the feed being on the wrong commit.

What exactly do you mean by that? The feed update operation finished successfully? I would expect it to fail. 

> It seems to be caused by https://github.com/openwrt/openwrt/commit/32d0a57dc10811a24816e51f10e55963f40fe462, which aims to optimize feed fetching by not downloading the entire history.

Interesting, any theory, how exactly? The same repositories and code is being used by buildbots, so we would see the fallout there already, its been 3 weeks.

Following https://git.openwrt.org/?p=feed/packages.git;a=commit;h=9ddc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91 works.

I've just checked directly on git.openwrt.org server:

```
git@git-01:~/repositories/feed/packages.git$ ls -al objects/9d/dc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91
-r--r----- 1 git git 266 Mar 16 05:20 objects/9d/dc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91

git@git-01:~/repositories/feed/packages.git$ git cat-file -p 9ddc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91
tree 4abc13b24e74936744af1a71101e5d7a5b59b43c
parent e416d3f12ee39df1a149a0eef0321ae8e491b948
author George Sapkin <george@sapk.in> 1741958568 +0200
committer Tianling Shen <cnsztl@gmail.com> 1742116732 +0800

adguardhome: bump to 0.107.57

Changelog: https://github.com/AdguardTeam/AdGuardHome/releases/tag/v0.107.57

```

I've just tried following:

```
git clone --depth 1 git://git.openwrt.org/feed/packages.git && \
   cd packages && \
   git fetch --depth=1 origin 9ddc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91 && \
   git -c advice.detachedHead=false checkout 9ddc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91 && \
   git log --pretty=oneline

Cloning into 'packages'...
remote: Enumerating objects: 7322, done.
remote: Counting objects: 100% (7322/7322), done.
remote: Compressing objects: 100% (6005/6005), done.
remote: Total 7322 (delta 247), reused 4793 (delta 175), pack-reused 0
Receiving objects: 100% (7322/7322), 14.23 MiB | 11.03 MiB/s, done.
Resolving deltas: 100% (247/247), done.
remote: Enumerating objects: 151, done.
remote: Counting objects: 100% (151/151), done.
remote: Compressing objects: 100% (76/76), done.
remote: Total 92 (delta 38), reused 21 (delta 0), pack-reused 0
Unpacking objects: 100% (92/92), 16.51 KiB | 768.00 KiB/s, done.
From git://git.openwrt.org/feed/packages
 * branch                      9ddc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91 -> FETCH_HEAD
HEAD is now at 9ddc2c5dcdf3 adguardhome: bump to 0.107.57
9ddc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91 (grafted, HEAD) adguardhome: bump to 0.107.57
```

cedric-chedaleux, 2025-03-22:
Only the HTTPS clone seems concerned by the problem!

```
$ git clone --depth 1 https://git.openwrt.org/feed/packages.git &&    cd packages &&    git fetch --depth=1 origin 9ddc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91 &&    git -c advice.detachedHead=false checkout 9ddc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91 &&    git log --pretty=oneline
Cloning into 'packages'...
remote: Enumerating objects: 7322, done.
remote: Counting objects: 100% (7322/7322), done.
remote: Compressing objects: 100% (6005/6005), done.
remote: Total 7322 (delta 247), reused 4793 (delta 175), pack-reused 0
Receiving objects: 100% (7322/7322), 14.24 MiB | 10.72 MiB/s, done.
Resolving deltas: 100% (247/247), done.
error: Server does not allow request for unadvertised object 9ddc2c5dcdf3f9470ad8ad1ce7d412460b6c5f91
```

eric-j-ason, 2025-03-22:
Thank you for spending time on this problem!

> Can you provide complete log of git operations?

Perhaps what @cedric-chedaleux [just posted](https://github.com/openwrt/openwrt/issues/18316#issuecomment-2745274440) is adequate. If you're looking for something else, please tell me, and I'll produce it.

> > You end up with the feed being on the wrong commit.
> 
> What exactly do you mean by that? The feed update operation finished successfully? I would expect it to fail.

The feed update fails, but leaves the repo with the wrong commit checked out. If you run the feed update again, it now sees that the repo already exists, succeeds, but the repo is still on the wrong commit.

> > It seems to be caused by [32d0a57](https://github.com/openwrt/openwrt/commit/32d0a57dc10811a24816e51f10e55963f40fe462), which aims to optimize feed fetching by not downloading the entire history.
> 
> Interesting, any theory, how exactly? The same repositories and code is being used by buildbots, so we would see the fallout there already, its been 3 weeks.

I have verified that I do not see the problem if I build from the commit just before this one. As for how the buildbots manage, I don't know. Are they perhaps using `git:` instead of `https:`, as @cedric-chedaleux discovered makes a difference?

ynezz, 2025-03-23:
Thanks for the pointers, I was able to reproduce and thus I've enabled `uploadpack.allowReachableSHA1InWant` to make this working.

From https://git-scm.com/docs/git-fetch-pack

> If the remote has enabled the options uploadpack.allowTipSHA1InWant, uploadpack.allowReachableSHA1InWant, or uploadpack.allowAnySHA1InWant, they may alternatively be 40-hex sha1s present on the remote.

closed 2025-03-23 by ynezz

eric-j-ason, 2025-03-23:
> Thanks for the pointers, I was able to reproduce and thus I've enabled `git config --system uploadpack.allowReachableSHA1InWant true` to make this working.

Great, thanks!

Do you know why this already worked for `git:`, but not for `https:`?