openwrt/openwrt #18320: RT-AX53U crashes with 24.10.0

daveastator, 2025-03-22:
### Describe the bug

I had problems with this router and openwrt from the beginning. Tried different RC's. 2 Bugs with latest openwrt 24.10.0:

I. (minor): after fresh 24.10.0 sysupgrade (resetting all configs), i couldnt get wifi working until i deleted /etc/config/wireless and let it regenerate. It looked very different and wifi then worked.

II. (major): when I run speedtest.net with my phone via the AX53U router, the router crashes and reboots after only a few seconds. When I download something with steam via LAN Cable, it does not crash, but caps out at 18MB/s, while my internet offers around 110MB/s, which I achive if I bypass the AX53U and connect the network cable directly to the cable modem.

Any Ideas whats happening? Any guidance on how to get logs even with crash and restart?

Is it possible that my specific router just sucks and I should look for a better one?

/etc/config/wireless:
=============
config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option band '2g'
        option channel '1'
        option htmode 'HE20'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'Love Zone 23'
        option encryption 'sae-mixed'
        option key '********************'
        option ocv '0'
        option wpa_disable_eapol_key_retries '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
        option band '5g'
        option channel '36'
        option htmode 'HE80'
        option cell_density '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'Love Zone 23'
        option encryption 'sae-mixed'
        option key '***********************'
        option ocv '0'
        option wpa_disable_eapol_key_retries '1'

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

I:) Install clean OpenWRT 24.10.0 and you should have a non-working wifi until you delete /etc/config/wireless and reboot

II:) I assume the reset happens for every RT-AX53U as soon as you start a heavy load via WiFi - for me speedtest.net android app does the trick 100%

### Actual behaviour

I:) You config the Wifi in luci like hell, enable everything, but nothing shows up when scanning

II:) Crash/Reboot for unknown reason

### Expected behaviour

I:) Getting WiFi working with luci after clean install WITHOUT having to delete and repopulate /etc/config/wireless once

II:) Not Crashing and around 1000MBit throughput, not barely 100MBit via LAN cable.

### Additional info

my /etc/config/wireless as already posted in the main post:

/etc/config/wireless:
==============
config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option band '2g'
        option channel '1'
        option htmode 'HE20'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'Love Zone 23'
        option encryption 'sae-mixed'
        option key '********************'
        option ocv '0'
        option wpa_disable_eapol_key_retries '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
        option band '5g'
        option channel '36'
        option htmode 'HE80'
        option cell_density '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'Love Zone 23'
        option encryption 'sae-mixed'
        option key '***********************'
        option ocv '0'
        option wpa_disable_eapol_key_retries '1'

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-03-22:
You have to enable hardware offload to near wired gigabit.
Forr wireless you additionally have to enable mt7i6 wed via modprobe.conf to reach half of radio link bw, ie mac 600-some Mbps with 2-antenna ax client displaying 1200 link rate.

brada4, 2025-04-03:
1s‰ poinnt clearly notes you upgraded from v22 ie unsupported upgrade. 

Do a config reset, bring up a basic wifi and check logread after you think it crashed.

neheb, 2025-04-10:
@arinc9 is this related to phy muxing?

arinc9, 2025-04-10:
I don't think so.