openwrt/openwrt #18321: In openwrt-24.10, after the STA mode is configured in wireless and the ieee80211r and ieee80211k parameter is configured, the expected effect cannot be achieved

loveepy, 2025-03-22:
I configured the parameters of ieee80211r in wireless. Configuration method relies on the web site,

>  https://openwrt.org/docs/guide-user/network/wifi/basic#fast_bss_transition_options_80211r.

This is my wireless configuration：
```
config wifi-iface 'default_radio1'                                                                   
        option device 'radio1'                                                                       
        option network 'wan'                                                                         
        option mode 'sta'                                                                            
        option ssid 'OpenWrt'                                                                        
        option encryption 'psk2'                                                                     
        option key 'password'                                                                        
        option ieee80211r '1'                                                                        
        option mobility_domain '1aa1'                                                                
        option reassociation_deadline '20000'                                                        
        option ft_over_ds '0'   
```
After I run the /etc/init.d/network restart command, I go to /tmp/run/wpa_supplicant-phy1-sta0.conf to view the content
```
network={
scan_ssid=1
ssid="OpenWrt"
key_mgmt=WPA-PSK FT-PSK
psk="password"
proto=RSN
beacon_int=100

}
```
I found that the configuration was not resolved in the.conf file, so I had an idea to modify the hostapd.sh file. However, my configuration based on the Internet and AI answers resulted in the inability to connect to the AP network in sta mode

I would like to know how to configure ieee80211r to be usable in STA mode.
I would like to thank the people who helped me in advance, thank you very much

brada4, 2025-03-22:
what parameters of these you want to "resolve" from your obvious access point bits?
https://w1.fi/cgit/hostap/tree/wpa_supplicant/wpa_supplicant.conf

castillofrancodamian, 2025-03-22:
@brada4 Where is it clarified that these parameters are ONLY for AP mode? Why does LuCI offer these options on a STA interface?

brada4, 2025-03-22:
Dont you recall couple of  days ago admitting stuffing them in config file?
DUPLICATE OF https://github.com/openwrt/openwrt/issues/18311

loveepy, 2025-03-24:
In my previous request for helphttps://github.com/openwrt/openwrt/issues/18311, I added the code by modifying the hostpad.sh file directly instead of parsing it through wireless. Moreover, I found that in  https://w1.fi/cgit/hostap/tree/wpa_supplicant/wpa_supplicant.conf, there is only one parsing method for ieee80211r, which is FT-PSK, and there are no parsing methods for the companion protocols ieee80211K and ieee80211V. I hope you can provide more detailed information. I would be very grateful for your assistance. @brada4

loveepy, 2025-03-24:
Thank you for your help. I've resolved the IEEE 802.11r issue—you were right, the configuration parameters were indeed different. Thanks again! However, I'm still not entirely clear on how to configure and use IEEE 802.11k and IEEE 802.11v. @brada4

brada4, 2025-03-24:
How thats related to initial "bug"?