openwrt/openwrt #18327: uboot-tools compile failed without openssl

msdos03, 2025-03-23:
### Describe the bug

After recent update, uboot-envtools is moved to uboot-tools and I recieve errors when compiling with default configuration.
```
make[4]: Entering directory '/home/user/openwrt/build_dir/target-arm_xscale_musl_eabi/u-boot-2025.01'
  HOSTLD  tools/dumpimage
/home/user/openwrt/staging_dir/toolchain-arm_xscale_gcc-13.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: tools/generated/lib/rsa/rsa-sign.o: in function `rsa_engine_get_priv_key':
rsa-sign.c:(.text+0x80): undefined reference to `ENGINE_get_id'
/home/user/openwrt/staging_dir/toolchain-arm_xscale_gcc-13.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: rsa-sign.c:(.text+0x168): undefined reference to `ENGINE_load_private_key'
/home/user/openwrt/staging_dir/toolchain-arm_xscale_gcc-13.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: tools/generated/lib/rsa/rsa-sign.o: in function `rsa_engine_remove':
rsa-sign.c:(.text+0x390): undefined reference to `ENGINE_finish'
/home/user/openwrt/staging_dir/toolchain-arm_xscale_gcc-13.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: rsa-sign.c:(.text+0x39c): undefined reference to `ENGINE_free'
/home/user/openwrt/staging_dir/toolchain-arm_xscale_gcc-13.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: tools/generated/lib/rsa/rsa-sign.o: in function `rsa_engine_init.constprop.0':
rsa-sign.c:(.text+0x3ac): undefined reference to `ENGINE_load_builtin_engines'
/home/user/openwrt/staging_dir/toolchain-arm_xscale_gcc-13.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: rsa-sign.c:(.text+0x3b4): undefined reference to `ENGINE_by_id'
/home/user/openwrt/staging_dir/toolchain-arm_xscale_gcc-13.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: rsa-sign.c:(.text+0x3e8): undefined reference to `ENGINE_init'
/home/user/openwrt/staging_dir/toolchain-arm_xscale_gcc-13.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: rsa-sign.c:(.text+0x408): undefined reference to `ENGINE_free'
/home/user/openwrt/staging_dir/toolchain-arm_xscale_gcc-13.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: rsa-sign.c:(.text+0x414): undefined reference to `ENGINE_set_default_RSA'
/home/user/openwrt/staging_dir/toolchain-arm_xscale_gcc-13.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: rsa-sign.c:(.text+0x434): undefined reference to `ENGINE_finish'
/home/user/openwrt/staging_dir/toolchain-arm_xscale_gcc-13.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: rsa-sign.c:(.text+0x458): undefined reference to `ENGINE_ctrl_cmd_string'
/home/user/openwrt/staging_dir/toolchain-arm_xscale_gcc-13.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: tools/generated/lib/rsa/rsa-sign.o: in function `rsa_add_verify_data':
rsa-sign.c:(.text+0xb48): undefined reference to `ENGINE_get_id'
/home/user/openwrt/staging_dir/toolchain-arm_xscale_gcc-13.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/13.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld.bfd: rsa-sign.c:(.text+0xbf8): undefined reference to `ENGINE_load_public_key'
collect2: error: ld returned 1 exit status
make[5]: *** [scripts/Makefile.host:104: tools/dumpimage] Error 1
make[4]: *** [Makefile:1915: tools] Error 2
make[4]: Leaving directory '/home/user/openwrt/build_dir/target-arm_xscale_musl_eabi/u-boot-2025.01'
make[3]: *** [Makefile:121: /home/user/openwrt/build_dir/target-arm_xscale_musl_eabi/u-boot-2025.01/.built] Error 2
```
This happens when only mbedtls is selected.
If I select openssl too, it will gone.

@TheRootED24 

### OpenWrt version

Snapshot

### OpenWrt release

Snapshot

### OpenWrt target/subtarget

kirkwood

### Device

all devices

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2025-03-23:
@TheRootED24 The way you setup the package it will build all of the tools which will fail since some of them require OpenSSL  even if we only want envtools

TheRootED24, 2025-03-23:
I never considered the dependency.  Thank you for correcting it properly.

borkra, 2025-03-24:
Well, I do build OpenSSL and still getting the failure...

robimarko, 2025-03-24:
@borkra Do you have libopenssl in your .config?

commit 293d5f13 (2025-03-24): uboot-tools: dont build tools unconditionally

Currently, both envtools and the rest of U-Boot tools are being built
regardless if the dumpimage package has been selected.

This will fail if only envtools are selected since the rest of tools
require OpenSSL while envtools do not require them.

So, only build tools if dumpimage is selected.

Fixes: 46e376c93514 ("uboot-tools: migrate uboot-envtools to uboot-tools")
Fixes: #18327
Tested-by: Martin Blumenstingl <martin.blumenstingl@googlemail.com>
Link: https://github.com/openwrt/openwrt/pull/18329
closed 2025-03-24 by openwrt-bot