openwrt/openwrt #18328: No connection as client when Wi-Fi key contains special characters (" and #)

2dann, 2025-03-23:
### Describe the bug

I want to connect to a Wi-Fi network as a client.

The Wi-Fi key of the network contains special characters, including double quotation marks (") and a hash symbol (#).

A connection is then not possible.
The system logs contain the following message regarding this:
```
Sat Mar 22 18:40:30 2025 daemon.notice wpa_supplicant\[1691\]: Set new config for phy phy1  
Sat Mar 22 18:40:30 2025 daemon.err wpa_supplicant\[1691\]: Line 9: failed to parse sae_password '\[KEY DATA REMOVED\]'.  
Sat Mar 22 18:40:30 2025 daemon.err wpa_supplicant\[1691\]: Line 13: failed to parse network block.  
Sat Mar 22 18:40:30 2025 daemon.err wpa_supplicant\[1691\]: Failed to read or parse configuration '/var/run/wpa_supplicant-phy1-sta0.conf'.
```

The file '/etc/config/wireless' contains the key correctly in single quotation marks (').
In the file '/var/run/wpa_supplicant-phy1-sta0.conf', the key is also specified in full, but in double quotation marks (").

I therefore suspect that there is initially an interaction with the double quotation marks and then with the hash symbol.

### OpenWrt version

24.10.0 

### OpenWrt release

r28427-6df0e3d02a

### OpenWrt target/subtarget

???

### Device

Cudy TR3000 v1

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

2dann, 2025-03-23:
I have corrected the following information:
- OpenWrt version
- OpenWrt release
- OpenWrt target/subtarget

robimarko, 2025-03-28:
I guess that its due to:
https://github.com/openwrt/openwrt/blob/main/package/network/config/wifi-scripts/files/lib/netifd/hostapd.sh#L726

2dann, 2025-03-30:
Thank you for the tip, @robimarko.

As a test, I modified the line [726](https://github.com/openwrt/openwrt/blob/main/package/network/config/wifi-scripts/files/lib/netifd/hostapd.sh#L726) in '/lib/netifd/hostapd.sh' as follows:
`append bss_conf 'wpa_passphrase=$key' "$N"`

This did not result in any change.

Then I searched a bit further and discovered line [1451](https://github.com/openwrt/openwrt/blob/main/package/network/config/wifi-scripts/files/lib/netifd/hostapd.sh#L1451). I then modified it as follows:
`passphrase="sae_password='${key}'"`

This resulted in the Wi-Fi key now being enclosed in single quotation marks (') in '/var/run/wpa_supplicant-phy1-sta0.conf'.

However, the error message in the logs unfortunately remains identical:
```
Sun Mar 30 20:04:56 2025 daemon.err wpa_supplicant[1698]: Line 9: failed to parse sae_password '[KEY DATA REMOVED]'.
Sun Mar 30 20:04:56 2025 daemon.err wpa_supplicant[1698]: Line 13: failed to parse network block.
Sun Mar 30 20:04:56 2025 daemon.err wpa_supplicant[1698]: Failed to read or parse configuration'/var/run/wpa_supplicant-phy1-sta0.conf'.
```

The key shown in the logs cuts off exactly at the hash symbol (#).

2dann, 2025-03-31:
Additionally, I tried the following:

Escaping the special characters using a backslash by inserting the following line before line [1451](https://github.com/openwrt/openwrt/blob/main/package/network/config/wifi-scripts/files/lib/netifd/hostapd.sh#L1451) in '/lib/netifd/hostapd.sh':
- For double quotation mark ("):
`key=$(echo "$key" | sed 's/"/\\"/g')`
- For the hash symbol (#):
`key=$(echo "$key" | sed 's/#/\\#/g')`

The respective special character was escaped with a backslash (\) and the backslash was visible both in the file '/var/run/wpa_supplicant-phy1-sta0.conf' and in the logs.

I tested the following variations: only double quotation mark escaped, only hash escaped, both double quotation mark and hash escaped, each with the key enclosed in double (") and single quotation marks (').

Unfortunately, the error message in the logs (the key still breaks at the hash) and the connection issue remain unchanged.