openwrt/openwrt #18332: what is the luci-i18n-opkg-zh-cn package in 24.10?

Sworder2025, 2025-03-23:
### Describe the bug

Not found it. Is the opkg package is obsolete?

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

x86/64

### Device

esxi

### Image kind

Official downloaded image

### Steps to reproduce

Not found `luci-i18n-opkg-zh-cn` package.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

obsy, 2025-03-23:
Install luci-i18n-package-manager-zh-cn. luci-app-opkg was replaced by luci-app-package-manager https://github.com/openwrt/luci/commit/bcd13b918e2f30b8d19027a06e3d773a1b0ec4c0

brada4, 2025-03-24:
Try
luci-i18n-zh-cn