openwrt/openwrt #18333: dnsmasq-full error `netlink: Error: cache initialization failed: Protocol error` when using PBR to split tunnel

LoV432, 2025-03-23:
### Describe the bug

I'm encountering an issue with dnsmasq where it starts spamming error `netlink: Error: cache initialization failed: Protocol error` when using [PBR](https://openwrt.org/docs/guide-user/network/routing/pbr_app) for split VPN tunneling based on specific domains. The setup works initially, but after some time, the errors start appearing, and nftset stops populating.
```
Sun Mar 23 17:52:24 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:52:33 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg046ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:52:58 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:54:08 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:54:22 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg076ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:54:22 2025 daemon.err dnsmasq[4877]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg076ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:54:53 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:54:53 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:54:53 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:54:53 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:54:54 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:55:10 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:55:11 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:55:16 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:55:26 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg076ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:56:46 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:56:54 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 netlink: Error: cache initialization failed: Protocol error
Sun Mar 23 17:56:54 2025 daemon.err dnsmasq[4]: nftset inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 netlink: Error: cache initialization failed: Protocol error
``` 
Originally i posted this on the forum PBR package post 
https://forum.openwrt.org/t/policy-based-routing-pbr-package-discussion/140639/1961?u=lov432
https://forum.openwrt.org/t/policy-based-routing-pbr-package-discussion/140639/1987?u=lov432

but it seems like the issue might be with dnsmasq? It also seems like this error was also encounter before here:
https://oftc.irclog.whitequark.org/openwrt-devel/2023-12-19

I was able to reproduce this issue while running strace. The attached files contain the syslogs and the strace output of roughly same period. The PBR configuration file is also attached.

[logs.txt](https://github.com/user-attachments/files/19410808/logs.txt)

[strace.log.tar.gz](https://github.com/user-attachments/files/19410818/strace.log.tar.gz)

[/etc/conf/pbr](https://github.com/user-attachments/files/19410851/pbr.txt)

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

Linksys EA8100

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2025-03-24:
If you run `nft monitor sets` or `nft monitor elements` on the router and re-create the issue, is there any interesting output that suggests either where it stopped working (i.e. were any successful before the errors). The strace output shows an EBUSY return before each dnsmasq log message.

If it’s related to running dnsmasq in the procd jail, the obvious test would be to run it directly without the jail parameters and try to recreate.

LoV432, 2025-03-24:
I just tried running dnsmasq without the procd jail, and the same issue is occurring.

As far as I can see, the outputs of `nft monitor elements` and `nft monitor sets` seem pretty normal. The output of `nft monitor elements` stops when the error starts.

nft monitor sets:
```
// Restarted PBR here
delete set inet fw4 pbr_wg_xray_4_dst_ip_cfg046ff5
delete set inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5
delete set inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5
delete set inet fw4 pbr_wg_xray_4_dst_ip_cfg076ff5
add set inet fw4 pbr_wg_xray_4_dst_ip_cfg046ff5 { type ipv4_addr; flags interval,timeout; auto-merge; timeout 1d; comment "pbr_domains_1"; }
add set inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5 { type ipv4_addr; flags interval,timeout; auto-merge; timeout 1d; comment "pbr_domains_2"; }
add set inet fw4 pbr_wg_xray_4_dst_ip_cfg066ff5 { type ipv4_addr; flags interval,timeout; auto-merge; timeout 1d; comment "pbr_domains_3"; }
add set inet fw4 pbr_wg_xray_4_dst_ip_cfg076ff5 { type ipv4_addr; flags interval,timeout; auto-merge; timeout 1d; comment "pbr_domains_4"; }
```

nft monitor elements:
```
add element inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5 { 18.214.244.240 expires 1d }
add element inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5 { 54.152.39.45 expires 1d }
add element inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5 { 54.166.74.52 expires 1d }
add element inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5 { 35.174.230.103 expires 1d }
add element inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5 { 23.23.38.99 expires 1d }
add element inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5 { 3.223.139.217 expires 1d }
add element inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5 { 54.92.206.154 expires 1d }
add element inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5 { 54.84.148.5 expires 1d }
add element inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5 { 54.197.81.95 expires 1d }
add element inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5 { 35.174.92.242 expires 1d }
add element inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5 { 35.171.54.46 expires 1d }
add element inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5 { 54.210.224.84 expires 1d }
add element inet fw4 pbr_wg_xray_4_dst_ip_cfg056ff5 { 98.85.50.152 expires 1d }
// Error starts happening here and after that no output happens
```

fixsssator, 2025-03-31:
I've got tons errors like this too:
"netlink: Error: cache initialization failed: Protocol error"

LoV432, 2025-04-30:
I hate to be pushy, but is there anything I can do to help move this along? This issue is being a major pain for me :(

fixsssator, 2025-05-05:
Hi again,

Just wanted to follow up on this issue. I'm still encountering the same error, and it’s been quite frustrating. I’ve tried restarting and verifying the configuration, but no luck so far. The error "netlink: Error: cache initialization failed: Protocol error." still persists after adding several rules to nft (as mentioned before).

Any suggestions or advice from those who’ve dealt with this issue? I'd really appreciate any help or insights to move forward with this.

Thanks for your time and assistance!

Hit360D, 2025-05-05:
Hi, looking forward to resolution to this issue as mentioned above.

dave14305, 2025-05-05:
The nftables code commit says this error should never happen, so I would suggest sending an email to the netfilter-devel mailing list and/or the dnsmasq-discuss mailing list. I’d start with netfilter to see what could cause this, then try the dnsmasq developer, depending on the feedback.

https://git.netfilter.org/nftables/commit/src/cache.c?id=a2ddb38f7eb818312c50be78028bc35145c039ae

https://netfilter.org/mailinglists.html

dave14305, 2025-05-08:
Good job following through. Now there’s a dnsmasq patch to test. 

https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=98189ff988d01d48929057037060d8cb2b4a22a6;hp=e86d53c438571e24d0491d03d553f48a262def91

LoV432, 2025-05-10:
After 2 days of testing, I haven't been able to reproduce the issue!

@fixsssator What router do you have? I can try to build a version for you too. I have a Linksys EA8100 and I've attached the version I'm using below:

[dnsmasq-full_2.92-r4_mipsel_24kc.zip](https://github.com/user-attachments/files/20139224/dnsmasq-full_2.92-r4_mipsel_24kc.zip)

LoV432, 2025-05-10:
Please take backups and keep a local copy of the original dnsmasq-full before you install this, just in case :D

[dnsmasq-full_2.92-r4_aarch64_cortex-a53.zip](https://github.com/user-attachments/files/20143312/dnsmasq-full_2.92-r4_aarch64_cortex-a53.zip)

fixsssator, 2025-05-11:
@LoV432 Thanks a lot once again — I really appreciate your effort and support!

I’ve already installed the version you provided and everything went smoothly. I’ll be doing full testing later this week and will share the results with you.

Thanks again — your help means a lot! 😊

fixsssator, 2025-05-12:
@LoV432

Thank you very much!

So far so good — I've had it running for almost the entire day without a single log message or issue.
Everything seems stable and smooth.

I really appreciate your recommendation to keep a backup of the original dnsmasq-full — I made sure to save a local copy before installing the new one, just to be safe.

Thanks again for your help and guidance — much appreciated!

KorbenDallas, 2025-05-14:
> [dnsmasq-full_2.92-r4_aarch64_cortex-a53.zip](https://github.com/user-attachments/files/20143312/dnsmasq-full_2.92-r4_aarch64_cortex-a53.zip)

Installed this version and 3 days pass without errors. Thank you.

LoV432, 2025-05-14:
I suppose its safe to close the issue now :)

Thanks dave for pointing me in the correct direction!

closed 2025-05-14 by LoV432

HomidWay, 2025-05-19:
> I suppose its safe to close the issue now :)
> 
> Thanks dave for pointing me in the correct direction!

Sorry for bothering, but may you also build package for x86 or link me to some kind of instruction on how to build it myself? Would be really grateful

LoV432, 2025-05-20:
I am assuming that you want x86/64 and not generic or legacy

[dnsmasq-full_2.92-r4_x86_64.zip](https://github.com/user-attachments/files/20325560/dnsmasq-full_2.92-r4_x86_64.zip)

HomidWay, 2025-05-20:
Yes exactly. Thank you, can't tell for how long I had this error, had it for more than year probably.

chazikai24, 2025-06-14:
Hi,can you build package(version 2.92-test13) for x86? 
It seems fix  an nftset error in test6.I want to has some test.Thank you!

fixss, 2025-10-03:
Sorry for bothering, but may you also build package for ramips/mt7621 (Xiaomi Mi Router 4A Gigabit Edition) ? 
Would be really grateful

LoV432, 2025-10-04:
My router EA8100 is also rampis/mt7621 so you should be able to use the same 1 i linked originally. I am still using this version too

https://github.com/user-attachments/files/20139224/dnsmasq-full_2.92-r4_mipsel_24kc.zip

AddRemover, 2025-11-23:
having same issue.
```
root@linksys:~# dnsmasq -v
Dnsmasq version 2.90  Copyright (c) 2000-2024 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
```
here and there I can see following messages in syslog:
```
netlink: Error: cache initialization failed: Protocol error
 Error: cache initialization failed: Resource temporarily unavailable
netlink: Error: cache initialization failed: Resource busy
Error: No such file or directory
```
whenever any of those appear, `nft monitor elements` shows no output.

The effect of this error is that nft set is not being populated with new IP addresses resolved, thus no proper routing table selected.

Please provide 2.92 test version for arm_cortex-a9_vfpv3-d16 platform

Thanks!

LoV432, 2025-11-24:
As always, Please take backups of everything before trying this :D

[dnsmasq-full_2.92-r13_arm_cortex-a9_vfpv3-d16.ipk.zip](https://github.com/user-attachments/files/23724456/dnsmasq-full_2.92-r13_arm_cortex-a9_vfpv3-d16.ipk.zip)

LGA1150, 2025-12-03:
Do not close the issue if the problem still exists.

stavultras, 2025-12-07:
> [dnsmasq-full_2.92-r4_aarch64_cortex-a53.zip](https://github.com/user-attachments/files/20143312/dnsmasq-full_2.92-r4_aarch64_cortex-a53.zip)

Works like a charm on Cudy WR3000P. That's for the fix and built!

Ilya87, 2025-12-28:
[dnsmasq-full_2.92~rc3-r1_aarch64_cortex-a53.zip](https://github.com/user-attachments/files/24360409/dnsmasq-full_2.92.rc3-r1_aarch64_cortex-a53.zip)
Latest dnsmasq 2.92rc3 build for aarch64_cortex-a53

vjagaro, 2025-12-29:
Here some more precompiled builds of dnsmasq v2.92rc3 for those who are waiting for this to be fixed:

https://github.com/sasanarakkha/build-dnsmasq-openwrt/releases/latest