openwrt/openwrt #18337: The size of nftables used in the new release seems to be much higher than that of iptables used in the old release.

rbqvq, 2025-03-24:
```shell
root@binwalk ~ # du -hd 1 _openwrt-24.10.0-ramips-mt7620-phicomm_k2-v22.4-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib
76K     _openwrt-24.10.0-ramips-mt7620-phicomm_k2-v22.4-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib/pppd
816K    _openwrt-24.10.0-ramips-mt7620-phicomm_k2-v22.4-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib/ucode
8.0K    _openwrt-24.10.0-ramips-mt7620-phicomm_k2-v22.4-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib/dnsmasq
1.8M    _openwrt-24.10.0-ramips-mt7620-phicomm_k2-v22.4-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib/opkg
344K    _openwrt-24.10.0-ramips-mt7620-phicomm_k2-v22.4-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib/rpcd
5.4M    _openwrt-24.10.0-ramips-mt7620-phicomm_k2-v22.4-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib
root@binwalk ~ # du -hd  1 _openwrt-19.07.0-ramips-mt7620-psg1218a-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib
568K    _openwrt-19.07.0-ramips-mt7620-psg1218a-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib/lua
32K     _openwrt-19.07.0-ramips-mt7620-psg1218a-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib/pppd
8.0K    _openwrt-19.07.0-ramips-mt7620-psg1218a-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib/dnsmasq
1.4M    _openwrt-19.07.0-ramips-mt7620-psg1218a-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib/opkg
76K     _openwrt-19.07.0-ramips-mt7620-psg1218a-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib/rpcd
4.0K    _openwrt-19.07.0-ramips-mt7620-psg1218a-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib/iptables
2.6M    _openwrt-19.07.0-ramips-mt7620-psg1218a-squashfs-sysupgrade.bin.extracted/squashfs-root/usr/lib
```

rbqvq, 2025-03-24:
> you have to ad ip6tables arptables and ebtables for a fair comparison. It is bigger but with less drama. 3years alrwady

In fact, at the current rate of growth, it may not be long before we end 8M support.

We need an imminent slimming down.

rbqvq, 2025-03-24:
> you have to ad ip6tables arptables and ebtables for a fair comparison. It is bigger but with less drama. 3years alrwady

As far as I can remember, these components are all soft linked to xtables-multi

Another larger size may be that mbedtls is built-in?

I'll check it later.

brada4, 2025-03-24:
This starts looking more of a forum discussion. You can get there with gh federated login.
Yes, wpa3 support took its toll, but gave https at no space extra.
It is like following 5 years in standards development. Yes, those beasts eat sto$age for breakfast and ram for lunch, and dont get smaller.

rbqvq, 2025-03-24:
> This starts looking more of a forum discussion. You can get there with gh federated login. Yes, wpa3 support took its toll, but gave https at no space extra. It is like following 5 years in standards development. Yes, those beasts eat sto$age for breakfast and ram for lunch, and dont get smaller.

Thanks. I will do it.

closed 2025-03-24 by rbqvq