openwrt/openwrt #18338: Cudy WR3000S - Missing leds

pongo20, 2025-03-24:
5 leds are missing, 24.10 and even in snaphots:
WAN + LAN1-LAN4

[image]

RolandoMagico, 2025-03-24:
There is no information about the LEDs in the commit for supporting the device (PR: https://github.com/openwrt/openwrt/pull/16939). Also nothing in the device DTS.

@igieon: Do you have further information regarding support for the WAN an LAN LEDs?

igieon, 2025-03-24:
If you want to investigate more its R59.dts in original firmware:
```
gpio-leds {
		compatible = "gpio-leds";
		internet {
			label = "oem:green:internet";
			gpios = <&pio 11 GPIO_ACTIVE_LOW>;
		};

		status {
			label = "oem:green:status";
			gpios = <&pio 10 GPIO_ACTIVE_LOW>;
		};

		wps {
			label = "oem:green:wps";
			gpios = <&pio 9 GPIO_ACTIVE_LOW>;
		};

		wifi {
			label = "oem:green:wifi";
			gpios = <&pio 6 GPIO_ACTIVE_LOW>;
		};

		wifi5g {
			label = "oem:green:wifi5g";
			gpios = <&pio 7 GPIO_ACTIVE_LOW>;
		};
	}
```
which i use and  default switch:
```
	R53 | R59)
		ucidef_set_interfaces_lan_wan "eth0" "eth1"
		ucidef_add_switch "switch0" \
			"1:lan" "2:lan" "3:lan" "4:lan" "0:wan" "6@eth0" "5@eth1"
		;;
```
SO i don't found direct control of led lan1-lan4 only in original firmware control for led are 
R47 - lan1, lan2 
R63 - lan1 - lan4 (WR3000H) here dts:
```
gpio-leds {
		compatible = "gpio-leds";
		internet {
			label = "oem:green:internet";
			gpios = <&pio 11 GPIO_ACTIVE_LOW>;
		};

		status {
			label = "oem:green:status";
			gpios = <&pio 5 GPIO_ACTIVE_LOW>;
		};

		wps {
			label = "oem:green:wps";
			gpios = <&pio 9 GPIO_ACTIVE_LOW>;
		};

		wifi {
			label = "oem:green:wifi";
			gpios = <&pio 6 GPIO_ACTIVE_LOW>;
		};

		wifi5g {
			label = "oem:green:wifi5g";
			gpios = <&pio 7 GPIO_ACTIVE_LOW>;
		};

		lan1 {
			label = "oem:green:lan1";
			gpios = <&pio 8 GPIO_ACTIVE_LOW>;
		};

		lan2 {
			label = "oem:green:lan2";
			gpios = <&pio 10 GPIO_ACTIVE_LOW>;
		};

		lan3 {
			label = "oem:green:lan3";
			gpios = <&pio 12 GPIO_ACTIVE_LOW>;
		};

		lan4 {
			label = "oem:green:lan4";
			gpios = <&pio 13 GPIO_ACTIVE_LOW>;
		};

		wan {
			label = "oem:green:wan";
			gpios = <&pio 35 GPIO_ACTIVE_LOW>;
		};
	};

```
So if you want to check you can use R63 dts and check if its work.

igieon, 2025-03-24:
Or you can follow https://openwrt.org/docs/techref/hardware/port.gpio#script_static

pongo20, 2025-03-24:
thanks! Also my leds are working "correctly" but for example I can't shut them off (I could on WR3000H, not on S) listing of **/sys/class/leds** shows:

mt76-phy0            
mt76-phy1          
white:status
white:wan-online
white:wlan-2ghz
white:wlan-5ghz
white:wps

which is exactly as listed in the code links above.

pongo20, 2025-03-25:
Failed with GPIO blink all leds test -> still cannot identify missing leds WAN, LAN1, LAN2, LAN3, LAN4

pongo20, 2025-03-25:
with the stock fw it's possible to toggle all led on/off. I'm asking Cudy support for relevant info about the missing leds...

igieon, 2025-03-26:
Probably they do it via calling something on switch directly mt7530

here this was removed https://github.com/openwrt/openwrt/commit/fc33c41c21362b7186aa051a2140623943fa3143

igieon, 2025-03-26:
And here is probably script to disable led https://forum.openwrt.org/t/openwrt-support-for-wax206/127523/293?u=djfe

pongo20, 2025-03-26:
confirmed that is the issue.

mdio mt7530-0 0:0x1f raw 0x21 0x0008 does the job of shutting off the switch leds - tested successfully

code for controlling mt7530 leds shall be pulled back in at least for Cudy WR3000S...

Thanks @igieon !!!

pongo20, 2025-06-07:
> OK perfect works and the WR3000s disco machine is quiet :-)
> 
> I will add that info to the Open-WRT forum. One thing I would be interested in: I remember in standard firmware there was a possibility to switch off LEDs but switch on in case of error, or somehow like that. Does somebody know if there is a possibility like that? Also does these LEDs only support white color?

Q1 - YES, the stock fw only allows you ON/OFF but it states that under certain conditions ("special events") leds may turn on again...just to signal such conditions.

[image]

Q2 - don't know actually...never tested

Netlighter, 2026-09-10:
Hey everyone, for anyone who needed instructions on how to configure LAN ports 1–4 and the WAN port

`opkg update && opkg install mdio-tools` or new `apk update && apk add mdio-tools`

Then do:

`mdio mt7530-0 0:0x1f raw 0x21 0x0009` - lights up when a cable is plugged in
`mdio mt7530-0 0:0x1f raw 0x21 0x0008` - always disabled
`mdio mt7530-0 0:0x1f raw 0x21 0x000b` - blinks