openwrt/openwrt #18342: Kernel Panic on GL-MT6000 with OpenWRT 24.10 stable

cryptage21, 2025-03-24:
### Describe the bug

Hi,

I've just bought a GL.inet MT6000 router on which I've installed OpenWRT 24.10 official (sysupgrade without keeping configuration). 
It's working perfectly but it crashes (kernel panic) and reboots automatically once or twice a day.

I don't understand why and I think it could be a bug because I was using exactly the same configuration on a Xiaomi Mi Router 3 Pro and Xiaomi Mi Router 3G without any problems. 

Crash log doesn't help me as no process is identified (only a memory address). 

Thanks for your help. 

`<1>[21917.635938] Unable to handle kernel paging request at virtual address 0004020801394359
<1>[21917.643865] Mem abort info:
<1>[21917.646643]   ESR = 0x0000000096000004
<1>[21917.650384]   EC = 0x25: DABT (current EL), IL = 32 bits
<1>[21917.655678]   SET = 0, FnV = 0
<1>[21917.658722]   EA = 0, S1PTW = 0
<1>[21917.661847]   FSC = 0x04: level 0 translation fault
<1>[21917.666706] Data abort info:
<1>[21917.669575]   ISV = 0, ISS = 0x00000004, ISS2 = 0x00000000
<1>[21917.675041]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
<1>[21917.680077]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
<1>[21917.685369] [0004020801394359] address between user and kernel address ranges
<0>[21917.692487] Internal error: Oops: 0000000096000004 [#1] SMP
<7>[21917.698041] Modules linked in: pppoe ppp_async xt_state xt_conntrack xt_CT pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat nf_flow_table_inet nf_flow_table nf_conntrack_bridge nf_conntrack mt7915e(O) mt76_connac_lib(O) mt76(O) mac80211(O) ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211(O) xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG tcp_bbr slhc nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_quota nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nf_tables nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcrc32c iptable_mangle iptable_filter ip_tables ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 compat(O) crypto_safexcel xt_set x_tables ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net
<7>[21917.698201]  ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink sha512_arm64 sha1_ce sha1_generic seqiv md5 geniv des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug(O) usbcore usb_common aquantia
<7>[21917.823748] CPU: 3 PID: 2084 Comm: zabbix_agentd Tainted: G           O       6.6.73 #0
<7>[21917.831729] Hardware name: GL.iNet GL-MT6000 (DT)
<7>[21917.836414] pstate: a0400005 (NzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
<7>[21917.843354] pc : 0xffffffc0801a2d6c
<7>[21917.846828] lr : 0xffffffc0801a30c4
<7>[21917.850299] sp : ffffffc082b0ba00
<7>[21917.853596] x29: ffffffc082b0ba00 x28: ffffffc080c28000 x27: ffffffffffff8000
<7>[21917.860712] x26: ffffffc080000000 x25: ffffffc080c28930 x24: ffffffc080000000
<7>[21917.867828] x23: 0000000000008000 x22: ffffff801025bdc8 x21: 0000000000007fff
<7>[21917.874944] x20: ffffff8003640828 x19: 0000000000005000 x18: 0000000000000000
<7>[21917.882060] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
<7>[21917.889176] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
<7>[21917.896290] x11: 0000000000000000 x10: 0000000000000000 x9 : 0000000000000000
<7>[21917.903406] x8 : 0000003d6837b000 x7 : 000000000000cfff x6 : 8b04020801394331
<7>[21917.910521] x5 : ffffff8003640838 x4 : 8b04020801394321 x3 : ffffffc082f15001
<7>[21917.917637] x2 : 0000000000000000 x1 : 3760988f84b817f4 x0 : ffffff801025b058
<7>[21917.924753] Call trace:
<7>[21917.927185]  0xffffffc0801a2d6c
<7>[21917.930311]  0xffffffc0801a3524
<7>[21917.933436]  0xffffffc0801a5ea4
<7>[21917.936561]  0xffffffc080038da8
<7>[21917.939685]  0xffffffc0800397c4
<7>[21917.942810]  0xffffffc080039c00
<7>[21917.945935]  0xffffffc080039df0
<7>[21917.949060]  0xffffffc08001cd6c
<7>[21917.952185]  0xffffffc08001ce3c
<7>[21917.955310]  0xffffffc0808d14e0
<7>[21917.958435]  0xffffffc0808d21a8
<7>[21917.961560]  0xffffffc080011530
<0>[21917.964688] Code: b40017c6 f10040c4 d2800002 54000040 (f9401c82)
<4>[21917.970760] ---[ end trace 0000000000000000 ]---
`

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Steps to reproduce

Random crash so not possible. 

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2025-03-24:
zabbix_agentd is mentioned and it’s not part of the base build.

systemcrash, 2025-03-25:
Yeah, are you running zabbix on it? That would be my first guess.

cryptage21, 2025-03-25:
Thanks for your help.

I'm pretty sure Zabbix (available in official repo) has nothing to do with my problem. 

If we look at the latest crash, there's no Zabbix process : 

```
Oops#1 Part1
<7>[59115.497942]  ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink sha512_arm64 sha1_ce sha1_generic seqiv md5 geniv des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug(O) usbcore usb_common aquantia
<7>[59115.623491] CPU: 3 PID: 1060 Comm: napi/phy0-7 Tainted: G           O       6.6.73 #0
<7>[59115.631299] Hardware name: GL.iNet GL-MT6000 (DT)
<7>[59115.635985] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
<7>[59115.642925] pc : 0xffffffc0791aea10 [mt7915e@0000000041e8e65b+0x1a000]
<7>[59115.649436] lr : 0xffffffc0791ae9fc [mt7915e@0000000041e8e65b+0x1a000]
<7>[59115.655946] sp : ffffffc081043c00
<7>[59115.659244] x29: ffffffc081043c00 x28: 0000000000000040 x27: ffffff8002313fe0
<7>[59115.666360] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff8010980050
<7>[59115.673476] x23: 000000000100128b x22: 0000000000040000 x21: ffffff8002311fe0
<7>[59115.680592] x20: 000000000000000a x19: 0000000000000002 x18: ffffffffffffc7b8
<7>[59115.687707] x17: ffffffffffffc800 x16: 0000000000006998 x15: 0000000000003fb8
<7>[59115.694822] x14: 0000000000000004 x13: ffffff8002316228 x12: 0000000000000000
<7>[59115.701938] x11: 0000000000000040 x10: ffffff8002316230 x9 : ffffff8002316228
<7>[59115.709054] x8 : ffffff8000496b80 x7 : 0000000000000000 x6 : 0000000000000000
<7>[59115.716169] x5 : ffffffc081043b18 x4 : ffffff8002316230 x3 : 0000000000000000
<7>[59115.723285] x2 : ffffff80003c0b80 x1 : 0000000000000000 x0 : ffffff8010980050
<7>[59115.730401] Call trace:
<7>[59115.732833]  0xffffffc0791aea10 [mt7915e@0000000041e8e65b+0x1a000]
<7>[59115.738996]  0xffffffc0791aee68 [mt7915e@0000000041e8e65b+0x1a000]
<7>[59115.745159]  0xffffffc07917c600 [mt76@00000000669a4467+0xd000]
<7>[59115.750977]  0xffffffc0806bf424
<7>[59115.754101]  0xffffffc0806bf71c
<7>[59115.757227]  0xffffffc0806bf7b4
<7>[59115.760351]  0xffffffc08005ebe4
<7>[59115.763476]  0xffffffc080013024
<4>[59115.766601] ---[ end trace 0000000000000000 ]---
<1>[59703.474329] Unable to handle kernel paging request at virtual address 82cb0000c0000037
<1>[59703.482242] Mem abort info:
<1>[59703.485042]   ESR = 0x0000000096000004
<1>[59703.488781]   EC = 0x25: DABT (current EL), IL = 32 bits
<1>[59703.494083]   SET = 0, FnV = 0
<1>[59703.497134]   EA = 0, S1PTW = 0
<1>[59703.500259]   FSC = 0x04: level 0 translation fault
<1>[59703.505124] Data abort info:
<1>[59703.507988]   ISV = 0, ISS = 0x00000004, ISS2 = 0x00000000
<1>[59703.513453]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
<1>[59703.518498]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
<1>[59703.523791] [82cb0000c0000037] address between user and kernel address ranges
<0>[59703.530909] Internal error: Oops: 0000000096000004 [#1] SMP
<7>[59703.536463] Modules linked in: pppoe ppp_async xt_state xt_conntrack xt_CT pppox ppp_generic nft_redir nft_nat nft_masq nft_flow_offload nft_fib_inet nft_ct nft_chain_nat nf_nat nf_flow_table_inet nf_flow_table nf_conntrack_bridge nf_conntrack mt7915e(O) mt76_connac_lib(O) mt76(O) mac80211(O) ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211(O) xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG tcp_bbr slhc nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_quota nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fib_ipv6 nft_fib_ipv4 nft_fib nf_tables nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcrc32c iptable_mangle iptable_filter ip_tables ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 compat(O) crypto_safexcel xt_set x_tables ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net
<7>[59703.536619]  ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink sha512_arm64 sha1_ce sha1_generic seqiv md5 geniv des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug(O) usbcore usb_common aquantia
<7>[59703.662168] CPU: 0 PID: 9515 Comm: kworker/u8:0 Tainted: G        W  O       6.6.73 #0
<7>[59703.670062] Hardware name: GL.iNet GL-MT6000 (DT)
<7>[59703.674748] Workqueue: events_power_efficient 0xffffffc0791cab78 [nf_conntrack@00000000b6985b57+0x10000]
<7>[59703.684214] pstate: a0400005 (NzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
<7>[59703.691155] pc : 0xffffffc0791cac58 [nf_conntrack@00000000b6985b57+0x10000]
<7>[59703.698098] lr : 0xffffffc0791cadc4 [nf_conntrack@00000000b6985b57+0x10000]
<7>[59703.705040] sp : ffffffc08b33bd10
<7>[59703.708338] x29: ffffffc08b33bd10 x28: 0000000000000038 x27: 0000000000000005
<7>[59703.715454] x26: ffffffc080b069c0 x25: 0000000000000000 x24: ffffffc078cd9bc8
<7>[59703.722571] x23: 000000000000006e x22: 00000000000017ff x21: ffffffc080b06000
<7>[59703.729686] x20: 82cb0000c0000000 x19: ffffff8006279e58 x18: 00000000ffffffff
[... cut ...]
<7>[59703.788525]  0xffffffc0800570d8
<7>[59703.791650]  0xffffffc080057534
<7>[59703.794775]  0xffffffc08005ebe4
<7>[59703.797900]  0xffffffc080013024
<0>[59703.801028] Code: 9127035a f8607854 37000674 d503201f (3940de93)
<4>[59703.807100] ---[ end trace 0000000000000000 ]---
```

Here we have  a process linked to Wireless interface "napi/phy0-7" and a Kernel process " kworker/u8:" but, more interesting, we have in "pc:" mt7915e and nf_conntrack. 

Not sure what I can do with that, if you have any ideas ?

systemcrash, 2025-03-25:
OK. 

I've read in the forum (and on GL.iNETs forum) that there is a known hardware failure in some batches of the MT6000, so best to contact GL.iNET and arrange for an RMA. They'll send you a new one for free.

You can search through the openwrt forum and you should find various descriptions. You might want to start there to help get this problem confirmed. Good if you can provide a serial number there too, as someone might be able to validate.

systemcrash, 2025-03-25:
Just to be quite sure it's not that, download the firmware again, sha256sum the download an re-flash it.

cryptage21, 2025-03-25:
Do you have a link for GL.iNet topic ? 

Not sure flashing again is necessary, sha256sum was checked. Will try if can't be resolved.

I've disabled WED as it may cause problems for some people. 
Will see if it's better or not.

sfx2000, 2025-03-27:
Not related to MT6000 - but to a few other Mediatek targets - MT2500, x3000, xe3000...

https://forum.gl-inet.com/t/critical-problem-notification-for-gl-mt2500-gl-x3000-gl-xe3000/56261

@systemcrash - is this the post you are referring to?

cryptage21, 2025-03-28:
Looks like WED is the source of crashes :

```
21:43:16 up 2 days, 23:00,  load average: 0.16, 0.04, 0.01
```

Will check the next few days.

Do developpers need more elements to debug ?

cryptage21, 2025-04-03:
```
21:39:14 up 8 days, 21:52,  load average: 0.00, 0.00, 0.00
```
WED is clearly the problem. No reboot since.

systemcrash, 2025-04-04:
I think @pesa1234 figured out a number of fixes for WED which IIRC are only in his tree, presently(?)

@pesa1234 can you comment?

cryptage21, 2025-04-22:
I'll check in the next few days but the problem seems to be solved with OpenWRT 24.10.1.

closed 2025-04-22 by cryptage21