openwrt/openwrt #18358: ca_bundle GlobalSign issues

timur-davletshin, 2025-03-27:
### Describe the bug

Started from here: https://forum.openwrt.org/t/banip-support-thread/16985/2508

Any download tool from OpenWrt fails to get `https://myip.ms/files/blacklist/general/latest_blacklist.txt` while Firefox can download it. If I save GlobalSign certificate manually in PEM format and load it, let's say, with curl then I can download file mentioned above.

`curl --cacert ~/myip-ms.pem -v https://myip.ms/files/blacklist/general/latest_blacklist.txt`

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-03-27:
myip[.]ms serves incomplefe certificate chain
https://www.ssllabs.com/ssltest/analyze.html?d=myip.ms&s=2001%3a41d0%3a303%3ad9d6%3a237a%3a0%3a0%3a1&hideResults=on&ignoreMismatch=on

timur-davletshin, 2025-03-27:
You're right. Thank you for your comment!

closed 2025-03-27 by timur-davletshin