openwrt/openwrt #18359: [dhcpv6.script] Parsed ipv6 router advertisment maleformed

BitFis, 2025-03-27:
### Describe the bug

Hello

I have an issue that the Ipv6 routes inserted into the routes seemed to be maleformed.
Check steps by steps to see advertisment information, following are the;

<details>
<summery>`ip -6 route`</summery>

```bash
$ ip -6 route
default from ddf8:e7c0:c40b::9aa via fe80::9683:c4ff:fe2f:8a75 dev eth0  metric 512
default from ddf8:e7c0:c40b::/64 via fe80::9683:c4ff:fe2f:8a75 dev eth0  metric 512
ddf8:e7c0:c40b::/64 dev eth0  metric 256
unreachable ddf8:e7c0:c40b::/64 dev lo  metric 2147483647  error -113
fd26:cfe2:578d::/64 dev br-lan  metric 1024
unreachable fd26:cfe2:578d::/48 dev lo  metric 2147483647  error -113
fd41:f700:4c32:1::/64 from ddf8:e7c0:c40b::9aa via fe80::9683:c4ff:fe2f:8a75 dev eth0  metric 512
fd41:f700:4c32:1::/64 from ddf8:e7c0:c40b::/64 via fe80::9683:c4ff:fe2f:8a75 dev eth0  metric 512
fe80::/64 dev eth0  metric 256
fe80::/64 dev br-lan  metric 256
fe80::/64 dev wlan0  metric 256
anycast ddf8:e7c0:c40b:: dev eth0  metric 0
anycast fd26:cfe2:578d:: dev br-lan  metric 0
anycast fe80:: dev br-lan  metric 0
anycast fe80:: dev eth0  metric 0
anycast fe80:: dev wlan0  metric 0
ff00::/8 dev br-lan  metric 256
ff00::/8 dev eth0  metric 256
ff00::/8 dev wlan0  metric 256
```

</details>

<details>
<summery>*/lib/netifd/dhcpv6.script* - `_proto_push_route`</summery>
Small dump of dhcpv6.script for `proto_send_update "$INTERFACE"`

```
"routes6" "::/0/fe80::9683:c4ff:fe2f:8a75/512/1799//ddf8:e7c0:c40b::9aa/128 
::/0/fe80::9683:c4ff:fe2f:8a75/512/1799//ddf8:e7c0:c40b:0:2a0:3ff:fe97:4fe/64 
ddf8:e7c0:c40b::/64//256/4294967295// 
fd41:f700:4c32:1::/64/fe80::9683:c4ff:fe2f:8a75/512/1527//ddf8:e7c0:c40b::9aa/128 
fd41:f700:4c32:1::/64/fe80::9683:c4ff:fe2f:8a75/512/1527//ddf8:e7c0:c40b:0:2a0:3ff:fe97:4fe/64" 
_proto_push_route
```

</details>

### OpenWrt version

r11306-c4a6851c72

### OpenWrt release

19.07.7

### OpenWrt target/subtarget

ipq40xx/generic

### Device

CustomDevice

### Image kind

Self-built image

### Steps to reproduce

1. Receive ipv6 routes:
<details><summary>ICMPv6 Route Adv. Prefix</summary>

```
Frame 56134: 150 bytes on wire (1200 bits), 150 bytes captured (1200 bits) on interface \Device\NPF_{372832C4-F589-42C5-A11C-362EB3538936}, id 0
Ethernet II, Src: GLTechnologi_2f:8a:75 (94:83:c4:2f:8a:75), Dst: IPv6mcast_01 (33:33:00:00:00:01)
Internet Protocol Version 6, Src: fe80::9683:c4ff:fe2f:8a75, Dst: ff02::1
    0110 .... = Version: 6
    .... 0000 0000 .... .... .... .... .... = Traffic Class: 0x00 (DSCP: CS0, ECN: Not-ECT)
    .... 0001 1111 0101 0100 0001 = Flow Label: 0x1f541
    Payload Length: 96
    Next Header: ICMPv6 (58)
    Hop Limit: 255
    Source Address: fe80::9683:c4ff:fe2f:8a75
    Destination Address: ff02::1
    [Source SLAAC MAC: GLTechnologi_2f:8a:75 (94:83:c4:2f:8a:75)]
Internet Control Message Protocol v6
    Type: Router Advertisement (134)
    Code: 0
    Checksum: 0xbac2 [correct]
    [Checksum Status: Good]
    Cur hop limit: 64
    Flags: 0xc0, Managed address configuration, Other configuration, Prf (Default Router Preference): Medium
        1... .... = Managed address configuration: Set
        .1.. .... = Other configuration: Set
        ..0. .... = Home Agent: Not set
        ...0 0... = Prf (Default Router Preference): Medium (0)
        .... .0.. = ND Proxy: Not set
        .... ..0. = Reserved: 0
    Router lifetime (s): 1800
    Reachable time (ms): 0
    Retrans timer (ms): 0
    ICMPv6 Option (Source link-layer address : 94:83:c4:2f:8a:75)
        Type: Source link-layer address (1)
        Length: 1 (8 bytes)
        Link-layer address: GLTechnologi_2f:8a:75 (94:83:c4:2f:8a:75)
    ICMPv6 Option (MTU : 1500)
        Type: MTU (5)
        Length: 1 (8 bytes)
        Reserved
        MTU: 1500
    ICMPv6 Option (Prefix information : ddf8:e7c0:c40b::/64)
        Type: Prefix information (3)
        Length: 4 (32 bytes)
        Prefix Length: 64
        Flag: 0xc0, On-link flag(L), Autonomous address-configuration flag(A)
            1... .... = On-link flag(L): Set
            .1.. .... = Autonomous address-configuration flag(A): Set
            ..0. .... = Router address flag(R): Not set
            ...0 0000 = Reserved: 0
        Valid Lifetime: Infinity (4294967295) (49710 days, 6 hours, 28 minutes, 15 seconds)
        Preferred Lifetime: Infinity (4294967295) (49710 days, 6 hours, 28 minutes, 15 seconds)
        Reserved
        Prefix: ddf8:e7c0:c40b::
    ICMPv6 Option (Recursive DNS Server ddf8:e7c0:c40b::1)
        Type: Recursive DNS Server (25)
        Length: 3 (24 bytes)
        Reserved
        Lifetime: 1800 (30 minutes)
        Recursive DNS Servers: ddf8:e7c0:c40b::1
    ICMPv6 Option (Advertisement Interval : 600000)
        Type: Advertisement Interval (7)
        Length: 1 (8 bytes)
        Reserved
        Advertisement Interval: 600000

```

</details>

<details><summary>ICMPv6 Route Adv. Routes</summary>

```
Frame 56701: 94 bytes on wire (752 bits), 94 bytes captured (752 bits) on interface \Device\NPF_{372832C4-F589-42C5-A11C-362EB3538936}, id 0
Ethernet II, Src: GLTechnologi_2f:8a:75 (94:83:c4:2f:8a:75), Dst: IPv6mcast_01 (33:33:00:00:00:01)
Internet Protocol Version 6, Src: fe80::9683:c4ff:fe2f:8a75, Dst: ff02::1
    0110 .... = Version: 6
    .... 0000 0000 .... .... .... .... .... = Traffic Class: 0x00 (DSCP: CS0, ECN: Not-ECT)
    .... 0001 1111 0101 0100 0001 = Flow Label: 0x1f541
    Payload Length: 40
    Next Header: ICMPv6 (58)
    Hop Limit: 255
    Source Address: fe80::9683:c4ff:fe2f:8a75
    Destination Address: ff02::1
    [Source SLAAC MAC: GLTechnologi_2f:8a:75 (94:83:c4:2f:8a:75)]
Internet Control Message Protocol v6
    Type: Router Advertisement (134)
    Code: 0
    Checksum: 0x16a5 [correct]
    [Checksum Status: Good]
    Cur hop limit: 64
    Flags: 0xc0, Managed address configuration, Other configuration, Prf (Default Router Preference): Medium
        1... .... = Managed address configuration: Set
        .1.. .... = Other configuration: Set
        ..0. .... = Home Agent: Not set
        ...0 0... = Prf (Default Router Preference): Medium (0)
        .... .0.. = ND Proxy: Not set
        .... ..0. = Reserved: 0
    Router lifetime (s): 1800
    Reachable time (ms): 0
    Retrans timer (ms): 0
    ICMPv6 Option (RA Flags Extension)
        Type: RA Flags Extension (26)
        Length: 1 (8 bytes)
        Flags Expansion Option: 0x8000, Managed address configuration, Prf (Default Router Preference): Medium
            1... .... .... .... = Managed address configuration: Set
            .0.. .... .... .... = Other configuration: Not set
            ..0. .... .... .... = Home Agent: Not set
            ...0 0... .... .... = Prf (Default Router Preference): Medium (0)
            .... .0.. .... .... = Proxy: Not set
            .... ..0. 0000 0000 = Reserved: 0
        Reserved
    ICMPv6 Option (Route Information : Medium fd41:f700:4c32:1::/64)
        Type: Route Information (24)
        Length: 2 (16 bytes)
        Prefix Length: 64
        Flag: 0x00, Route Preference: Medium
            ...0 0... = Route Preference: Medium (0)
            000. .000 = Reserved: 0
        Route Lifetime: 1800 (30 minutes)
        Prefix: fd41:f700:4c32:1::
```

</details>

### Actual behaviour

Permission denies fails, works when other routes are added manually, can not delete route entry.
```
$ ping fd41:f700:4c32:1::1
PING fd41:f700:4c32:1::1 (fd41:f700:4c32:1::1): 56 data bytes
ping: sendto: Permission denied
```

### Expected behaviour

Timeout or response depending if device is available
```
$ ping fd41:f700:4c32:1::1
PING fd41:f700:4c32:1::1 (fd41:f700:4c32:1::1): 56 data bytes
...
```

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2025-03-27:
> default from ddf8:e7c0:c40b::9aa via fe80::9683:c4ff:fe2f:8a75 dev eth0  metric 512
> default from ddf8:e7c0:c40b::/64 via fe80::9683:c4ff:fe2f:8a75 dev eth0  metric 512
> ddf8:e7c0:c40b::/64 dev eth0  metric 256
> unreachable ddf8:e7c0:c40b::/64 dev lo  metric 2147483647  error -113.

Is this a badly set ULA from an upstream device you manage?

BitFis, 2025-03-27:
To some part, its from following device; https://www.gl-inet.com/products/gl-s200/, the prefix comes from thread. If i disable the script, those entries mentioned are not there. So they are injected by the dhcpv6.script

BitFis, 2025-03-27:
From talking to colleges, the gls200 device is not behave quite correctly. But generally we figured out, sending the ping via `ping -I ddf8:e7c0:c40b::9aa fd41:f700:4c32:1::1` works. But it is unclear why this script inserts the `from` in the first place.

brada4, 2025-03-27:
Ask glinet about their firmware.

BitFis, 2025-03-27:
But would you mind checking the icmpv6 package dump provided above, i do not see the reason for the route inserting `from` or in fact the default route. Any hints on how those are retrieved from the script and why?

brada4, 2025-03-28:
Your report is NOT related to OpenWrt. No action taken here can help your proprietary firmware in any way. You can close it ... yesterday.

BitFis, 2025-03-28:
I understand how you came to the conclusion the issue is not related to OpenWRT, thought I believe there is an issue, thought unclear which OpenWRT component or the behavior is expected.
My core issue is, if I compare the behavior with other linux systems or windows, the inserted routes from the icmpv6 RA

</details>

<details><summary>ICMPv6 RA</summary>

```
Frame 56701: 94 bytes on wire (752 bits), 94 bytes captured (752 bits) on interface \Device\NPF_{372832C4-F589-42C5-A11C-362EB3538936}, id 0
Ethernet II, Src: GLTechnologi_2f:8a:75 (94:83:c4:2f:8a:75), Dst: IPv6mcast_01 (33:33:00:00:00:01)
Internet Protocol Version 6, Src: fe80::9683:c4ff:fe2f:8a75, Dst: ff02::1
    0110 .... = Version: 6
    .... 0000 0000 .... .... .... .... .... = Traffic Class: 0x00 (DSCP: CS0, ECN: Not-ECT)
    .... 0001 1111 0101 0100 0001 = Flow Label: 0x1f541
    Payload Length: 40
    Next Header: ICMPv6 (58)
    Hop Limit: 255
    Source Address: fe80::9683:c4ff:fe2f:8a75
    Destination Address: ff02::1
    [Source SLAAC MAC: GLTechnologi_2f:8a:75 (94:83:c4:2f:8a:75)]
Internet Control Message Protocol v6
    Type: Router Advertisement (134)
    Code: 0
    Checksum: 0x16a5 [correct]
    [Checksum Status: Good]
    Cur hop limit: 64
    Flags: 0xc0, Managed address configuration, Other configuration, Prf (Default Router Preference): Medium
        1... .... = Managed address configuration: Set
        .1.. .... = Other configuration: Set
        ..0. .... = Home Agent: Not set
        ...0 0... = Prf (Default Router Preference): Medium (0)
        .... .0.. = ND Proxy: Not set
        .... ..0. = Reserved: 0
    Router lifetime (s): 1800
    Reachable time (ms): 0
    Retrans timer (ms): 0
    ICMPv6 Option (RA Flags Extension)
        Type: RA Flags Extension (26)
        Length: 1 (8 bytes)
        Flags Expansion Option: 0x8000, Managed address configuration, Prf (Default Router Preference): Medium
            1... .... .... .... = Managed address configuration: Set
            .0.. .... .... .... = Other configuration: Not set
            ..0. .... .... .... = Home Agent: Not set
            ...0 0... .... .... = Prf (Default Router Preference): Medium (0)
            .... .0.. .... .... = Proxy: Not set
            .... ..0. 0000 0000 = Reserved: 0
        Reserved
    ICMPv6 Option (Route Information : Medium fd41:f700:4c32:1::/64)
        Type: Route Information (24)
        Length: 2 (16 bytes)
        Prefix Length: 64
        Flag: 0x00, Route Preference: Medium
            ...0 0... = Route Preference: Medium (0)
            000. .000 = Reserved: 0
        Route Lifetime: 1800 (30 minutes)
        Prefix: fd41:f700:4c32:1::
```

</details>

Result to (without `from ddf8:e7c0:c40b::9aa`):
```
fd41:f700:4c32:1::/64 via fe80::9683:c4ff:fe2f:8a75 dev eth0  metric 512
fd41:f700:4c32:1::/64 via fe80::9683:c4ff:fe2f:8a75 dev eth0  metric 512
```
While OpenWRT inserts those `from ddf8:e7c0:c40b::...` in the route list.

I currently do not have the logs available, so I will need to provide those routes from other OS's on monday, but I do not understand the difference in behavior here. Since (from my understanding) this is coming from the component [`odhcpd`](https://openwrt.org/docs/techref/odhcpd) and the script `dhcpv6.script` which both are OpenWRT components (from my understanding). So would you mind adding some additional information to help me solve this issue as following:

* ICMPv6 packages is malformed due to X (as hinted in the comments above, its likely this is the case, thought I do not understand how it is malformed since it behaves correctly on other OS systems - routes are entered as expected).
* This behavior is expected on OpenWRT, since the `from` is required due to X, thought unclear to me why am I required to bind specifically to the `from` address? Can I disable this behavior?
* This is specifically related to `odhcpd` and I need to move this issue over there.

From my understanding, the issue is a combination between the router advertisment and OpenWRT RA processing, so I can not close this issue just yet.

brada4, 2025-03-28:
Try to reproduce bug in 24.10.0 in a virtual machine, without that you came to a fruit seller telling broccoli not tasty.

BitFis, 2025-03-31:
Currently its not sure I will find the time, but will try to get to it sooner then later. 👍

Neustradamus, 2025-04-22:
To follow this ticket.