openwrt/openwrt #18372: uboot-tools compile fails again

msdos03, 2025-03-29:
I have reported an compilation failure previously https://github.com/openwrt/openwrt/issues/18327
But after https://github.com/openwrt/openwrt/pull/18329 , there is another issue stopping me from compiling.
It is also mentioned there https://github.com/openwrt/openwrt/issues/18329#issuecomment-2746561230
It seems like kirkwood also has this issue.
Maybe it is an ARM32 generic issue.

```
make[4]: Entering directory '/home/user/openwrt/build_dir/target-arm_xscale_musl_eabi/u-boot-2025.01'
  ENVP    include/generated/env.in
<command-line>: fatal error: include/config.h: No such file or directory
compilation terminated.
make[4]: *** [Makefile:1874: include/generated/env.in] Error 1
make[4]: Leaving directory '/home/user/openwrt/build_dir/target-arm_xscale_musl_eabi/u-boot-2025.01'
make[3]: *** [Makefile:115: /home/user/openwrt/build_dir/target-arm_xscale_musl_eabi/u-boot-2025.01/.built] Error 2
make[3]: Leaving directory '/home/user/openwrt/package/boot/uboot-tools'
time: package/boot/uboot-tools/compile#0.32#0.15#0.46
    ERROR: package/boot/uboot-tools failed to build.
make[2]: *** [package/Makefile:189: package/boot/uboot-tools/compile] Error 1
```

@TheRootED24 @robimarko

robimarko, 2025-03-29:
Please DO NOT remove the issue template cause now stuff like diffconfig that is required to reproduce anything is missing

dick902, 2025-03-30:
I had tested and can to prove that this issue was not related to the config.
What I proceeded as:

Use menuconfig to setup what I need to build, save and exit, make clean & make
And the compile can be successfully. uboot-envtools compile normally.

Backup the .config file (for later compare use)
Make a diff file and configure using config diff file to build again (cp diff .config ,make defconfig && make clean && make)
And now the compile is fail with the uboot-envtools fail.

I did compare both .config files and confirm they are 100% identical. 

Same config, use menuconfig compile ok but use diif config compile will be fail.

I test with different target include Rockchip & Mediatek ARM, the result were same.

dick902, 2025-03-30:
mediatek diffconfig for your reference

```
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r4-poe=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-package-manager=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
```

msdos03, 2025-03-30:
@dick902 
It's strange that I can't get this error again after make distclean and build again with latest snapshot and feeds. I will close this for now.

closed 2025-03-30 by msdos03

Betonmischer86, 2025-04-17:
The issue is still there, and it's intermittent. It may or may not happen to me regardless of whether make menuconfig or make defconfig is used (although, the former makes the build fail less often). Creating the build environment from scratch doesn't prevent it from happening. I'm also building for the BPI-R4, by the way.