openwrt/openwrt #18374: fakeroot segfaults in clean builds

msdos03, 2025-03-29:
After distclean and compile for kirkwood, it stuck here.
My host OS is debian 12 bookworm AMD64
```
/home/user/openwrt/staging_dir/host/bin/fakeroot /home/user/openwrt/staging_dir/host/bin/apk mkpkg --info "name:kmod-gpio-button-hotplug" --info "version:6.6.84-r5" --info "description:This is a replacement for the following in-kernel drivers: 1) gpio_keys (KEYBOARD_GPIO) 2) gpio_keys_polled (KEYBOARD_GPIO_POLLED) Instead of generating input events (like in-kernel drivers do) it generates uevent-s and broadcasts them. This allows disabling input subsystem which is an overkill for OpenWrt simple needs." --info "arch:arm_xscale" --info "license:GPL-2.0" --info "origin:feeds/base/kernel/gpio-button-hotplug" --info "url:" --info "maintainer:" --info "provides:"  --script "post-install:/home/user/openwrt/build_dir/target-arm_xscale_musl_eabi/linux-kirkwood_generic/gpio-button-hotplug/apk-arm_xscale/kmod-gpio-button-hotplug/post-install" --script "pre-deinstall:/home/user/openwrt/build_dir/target-arm_xscale_musl_eabi/linux-kirkwood_generic/gpio-button-hotplug/apk-arm_xscale/kmod-gpio-button-hotplug/pre-deinstall" --info "depends:kernel=6.6.84~f25cad1b5ef508e619561c8c0e9410f7-r1" --files "/home/user/openwrt/build_dir/target-arm_xscale_musl_eabi/linux-kirkwood_generic/gpio-button-hotplug/ipkg-arm_xscale/kmod-gpio-button-hotplug" --output "/home/user/openwrt/bin/targets/kirkwood/generic/packages/kmod-gpio-button-hotplug-6.6.84-r5.apk" --sign "/home/user/openwrt/private-key.pem"
/home/user/openwrt/staging_dir/host/bin/fakeroot: line 185: 615677 Segmentation fault      FAKEROOTKEY=$FAKEROOTKEY LD_LIBRARY_PATH="$PATHS" LD_PRELOAD="$FAKEROOT_LIB" "$@"
```
related issue: https://github.com/openwrt/packages/issues/13855

namiltd, 2025-03-29:
Same on Cl: https://github.com/openwrt/openwrt/actions/runs/14133382239/job/39620643719

robimarko, 2025-03-29:
This one seems to affect Debian and Ubuntu, I can only reproduce it inside of Ubuntu Docker while on Fedora it works fine.

Trying to bisect where it broke

robimarko, 2025-03-29:
And it seems that its somehow due to LibreSSL update

namiltd, 2025-03-29:
I suggest you do a revert https://github.com/openwrt/openwrt/commit/21cece29e9eebc397eadee259e60c7fc44828cce and everything will be clear.

robimarko, 2025-03-29:
Ok, so I tracked it down to APK segfaulting (And not fakeroot itself) once LibreSSL 4.0 is used.

I will revert the update for now.

robimarko, 2025-03-29:
It should be fixed via: https://github.com/openwrt/openwrt/commit/6ac09b940fcdef052fff3a2819a8ee7d44f92776

Please close if it now works

namiltd, 2025-03-29:
It looks like revert helped.

msdos03, 2025-03-30:
Now it works, thanks.

closed 2025-03-30 by msdos03