openwrt/openwrt #18376: VLAN not working after update to 24.10.0

Probiersmal, 2025-03-29:
### Describe the bug

After configuration of VLAN, the ports are off. 
With Version 24.10.0, the 4 Network Ports are looking dead.
No communication with the router is possible. Reset is the only way out.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

lantiq/xrx200

### Device

BT Home Hub 5A

### Image kind

Official downloaded image

### Steps to reproduce

If not noted, all steps are done with luci
Configuration is not what i realy need, but an easy minimalized setup to reproduce the failure.
Description syntax: "v -> 0" means option was checked and i unchecked it
"0 -> v" means option was unchecked and i checked it
**Steps to reproduce:**
1. Flash image openwrt-24.10.0-lantiq-xrx200-bt_homehub-v5a-squashfs-sysupgrade.bin
2. LuCi | System
  Hostname = OWrt1_Test
  Timezone = Europe/Berlin
3. Network | DHCP and DNS-Server | General
  Authoritative   v -> 0
4. LuCi | Network | Interfaces | lan
  edit lan
  DHCP Server | General Setup
    Ignore interface 0 -> v
  DHCP Server | IPv6 Settings
    RA-Service = disabled
    DHCPv6-Service = disabled
  General Settings
    IPv4 address 192.168.1.1 (unchanged)
    IPv4 gateway 192.168.1.252
  Advanced Settings
    Use custom DNS servers 192.168.1.252
    Delegate IPv6 prefixes v -> 0
    IPv6 assignment length = disabled

So far Network works as expected.
e.g. Network | Diagnostics
ping openwrt.org works
System | Software | Update lists works fine

Now comes the critical configuration of VLAN
5. Network | Interfaces | Devices | br-lan | Configure...
  General device options
    Bridge ports add    eth0
  Bridge VLAN filtering
    Add 1 line for VLAN
    VLAN ID = 6    Local=v   eth0=T   lan1=U   lan2=U  lan3=U  lan4=T
  Network | Interfaces | lan | Edit
     General Settings
     Device br-lan.6
  Save & Apply
Network Ports are now switched off. No connection to router possible.
Reset ist the only way to come out here

### Actual behaviour

Network Ports are now switched off. No connection to router possible.

### Expected behaviour

Network Ports 1, 2, 3 should work without VLAN Tags
Network Port 4 should work with VLAN Tag

### Additional info

I use VLANs since 4 years with the router. Last working Version 23.05.2
Exactly the same configuration works fine before 24.10.0

The device is connected to a router (192.168.1.252) which has the wan connection via DSL

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Probiersmal, 2025-03-29:
Uploading file network did not work.
I renamed the file to network.txt and now something happended and i got a sort of a link. I don't know if the link goes to my uploaded file.

[network.txt](https://github.com/user-attachments/files/19523123/network.txt)

brada4, 2025-03-29:
Because yor platform is migrated from swconfig to dsa.
dsa master is not ethernet with vlans.

Probiersmal, 2025-03-29:
Is there no way to use VLAN's any more with dsa? (whatever dsa is ... i will search and dig into the documentation later)
Or is VLAN configured in a different way with dsa?

brada4, 2025-03-30:
Ask in forum, no bug there. You restored backup when sysupgrade told you not to.
Uou enablee vlans on br-lan

Probiersmal, 2025-03-30:
Your comment: 
> You restored backup when sysupgrade told you not to.

I did not restore a backup. I don't know why you expect that.

I do not know what your following line tells me. (i did not find the word nables in dictionary leo.org):
> One nables vlans on br-lan

Anyhow. I read the documentation of OpenWRT about dsa and vlans.
It looks like the only difference for me as the admin of the router is:
"Do not add eth0 to the bridge br-lan" (see step 5 in my steps to reproduce)

I did a new try without adding eth0 to the bridge br-lan and anything worked fine as expected.

Probiersmal, 2025-03-30:
I thank you very much for the hint about dsa. 
I am impressed about your quick answer
:-)        :-)        :-)

closed 2025-03-30 by Probiersmal

brada4, 2025-03-30:
On different platform i locked myself out dozen times years ago ;-)

mauron85, 2025-10-16:
Sorry ignore my comment. VLANs are working. I just misconfigured router.