openwrt/openwrt #18387: DNS hijack rule does not apply DNAT reliably

mark8223, 2025-03-31:
### Describe the bug

Have a DNS hijack rule for IPv6 that DNAT's packets to Google's DNS server as follows:
```
config redirect
	option dest 'wan'
	option target 'DNAT'
	option name 'Hijack-DNS-IPv6'
	option src 'lan'
	option src_dport '53'
	option dest_ip '2001:4860:4860::8888'
	option family 'ipv6'
	list src_mac '$clientMAC'
```
Then from an iOS client with iSH shell app, fire off a DNS query to a fictional server that should get hijacked by the rule and DNAT'ed:
```
nslookup google.com fd::1
```
This will fire off an A query and an AAAA query very close to each other. Listening with tcpdump on the wan interface of the router you'll see the A quer being DNAT'ed, but the AAAA query goes out to the internet still with the original destination (obviously wrong and undesired behavior):
```
11:53:01.900537 Out ethertype IPv6 (0x86dd), length 92: $clientIPv6.61581 > 2001:4860:4860::8888.53: 27025+ A? google.com. (28)
11:53:01.900584 Out ethertype IPv6 (0x86dd), length 92: $clientIPv6.61581 > fd::1.53: 53025+ AAAA? google.com. (28)
```

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer C6U v1

### Image kind

Official downloaded image

### Steps to reproduce

Install rule as in the example.
Get an iOS phone and install iSH app then fire off an `nslookup` as above.
I don't know what's special about this client, but it's able to generate the second packet that is not DNAT'ed correctly by the rule. Other Linux-based nslookup clients I have seem to not be able to reproduce this.

### Actual behaviour

AAAA query shortly after A query is not DNAT'ed.

### Expected behaviour

Both A and AAAA queries should be DNAT'ed.

### Additional info

You should have a working IPv6 WAN setup to reproduce this.
This bug reproduces 100% of the time on 24.10 and 0% of the time on 23.05.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-03-31:
Remove dst wan line

mark8223, 2025-03-31:
> Remove dst wan line

Made no difference.
```
13:06:38.748218 Out ethertype IPv6 (0x86dd), length 92: $clientIPv6.51902 > 2001:4860:4860::8888.53: 42310+ A? google.com. (28)
13:06:38.748263 Out ethertype IPv6 (0x86dd), length 92: $clientIPv6.51902 > fd::1.53: 2774+ AAAA? google.com. (28)
```

brada4, 2025-03-31:
Did you reload firewall?
Could you share generated fragment from `nft list ruleset` ?

egc112, 2025-04-01:
In addition to @brada4 's advice make sure your iphone has a fixed MAC address and has a working IPv6 GUA address , check with `https://test-ipv6.com/`

For the record the rules I use are (so without dest 'wan'):
```
/etc/config/firewall:  
config redirect
	option target 'DNAT'
	option name 'DNSto8888ipv4'
	option src 'lan'
	option src_dport '53'
	option dest_ip '8.8.8.8'
	list src_mac 'D0:AB:D5:92:CC:CC'
	option reflection '0'

config redirect
	option target 'DNAT'
	option name 'DNSto8888ipv6'
	option src 'lan'
	list src_mac 'D0:AB:D5:92:CC:CC'
	option src_dport '53'
	option reflection '0'
	option dest_ip '2001:4860:4860::8888'
	option family 'ipv6'
```

brada4, 2025-04-06:
Please add a something.nft into /etc/nftables.d/
```
chain raw_prerouting {
 type filter hook prerouting priority raw; policy accept;
 # maybe prepend .. iif "br-lan" .. instead
 iif != "lo" th sport . th dport { 53 . 0-65535 , 0-65535 . 53 ) meta nftrace set 1
}
```
chech with fw4 check

Then run `nft monitor trace | tee -a /tmp/nftrace.txt`

Then check how f::: target bypasses the rule.

REF: https://wiki.nftables.org/wiki-nftables/index.php/Ruleset_debug/tracing
 - i add trace bit to packet at first rule of our earliest hook, if you have custom hooks like ingress/offload you need to make custom lower priority cf examples in page.
 - it is pre-conntrack, so i manually match both directions

brada4, 2025-04-06:
If it never enters nftables it cannot be transformed in there...
You can leave nftrace long term, no adverse effect when not tracing.

Not my pond at dsa bridge tracing,  ip-bridge shows dsa shortcuts bypassing soc cpu unlike brctl, also (wrong and) otherwise invisible vlan ~~flops~~ mappings

brada4, 2025-04-06:
dnat rule establishes conntrack state (status (not state) eq dnat) and rule itself does not process each packet. But how comes it now works?

brada4, 2025-04-06:
ok cool, note down dsa `bridge fdb` and compare, does not seem like any firewall innovation at fault .

tmcqueen-materials, 2025-04-06:
@mark8223 : The behavior you observe on 23.05 is as expected -- the first time you hit the rule a conntrack entry is created, and then the second packets matches that connection and so get handled in the same way without propagating through the entire nft rule stack.

I suspect iSH triggers this because it uses the same source port for both the A and AAAA queries (meaning that the src-ip, src-mac, src-port, dst-ip, dst-port are the same for both packets). 

@brada4 is probably correct in that it is some problem with a hardware shortcut forwarding engine seeing the conntrack entry and forwarding the second packet, but not modifying the destination ip.

tmcqueen-materials, 2025-04-06:
@brada4 : yes, exactly ... as you would expect with a hardware forwarding engine that gets an entry configured everytime a conntrack entry is added (the usual sfe thing)

brada4, 2025-04-06:
No no, that is other kind of quickpath, out of reach of firewall conntrack running on soc cpu.

tmcqueen-materials, 2025-04-06:
@mark8223 : That is properly weird! I do not have an `Archer C6U v1 (EU)` to play with, but I have been unable to reproduce this packet mangling on completely different hardware (`ipq807x`). I always see things like (LAN side):
```
15:05:19.800745 IP6 (flowlabel 0xb0205, hlim 64, next-header UDP (17) payload length: 40) ${clientIP}.18880 > fd::1.53: [udp sum ok] 17957+ A? www.google.com. (32)
15:05:19.801383 IP6 (flowlabel 0xb0205, hlim 64, next-header UDP (17) payload length: 40) ${clientIP}.18880 > fd::1.53: [udp sum ok] 35956+ AAAA? www.google.com. (32)
```
And appropriately transformed packets thereafter. Even when run over a million iterations.

Do you have another piece of hardware to see if you also see this issue on it? 

Can you also try doing this from a LAN-side system:
`dig -b ${clientIP}#18880 +nocookie +bufsize=0 @fd::1 www.google.com A @fd::1 www.google.com AAAA`
and confirm that you *do* see the issue (this was how I simulated the iSH queries for the million iteration trials, after not seeing it in a handful from iSH on iOS).

brada4, 2025-04-06:
Ok, so mtk dsa but not ipq...
Is nftrace saying same as OP described vs v23?

brada4, 2025-04-06:
2 packets stuffed together could be gro superpacket, but that is supposedly reaching kernel as multiple packets after optimal DMA from the device.

tmcqueen-materials, 2025-04-06:
@mark8223 : You could disable gro and see if it fixes the issue. Something like:

1. `ethtool -K iface rx-gro-list off`
2. Run tests
3. `ethtool -K iface gro off``
4. Run tests

Where `iface` are the MT7621 switch / things attached to the lan bridge.

brada4, 2025-04-08:
Hope @hauke agrees packets flying past firewall is a wild leak problem.
(besides rx-gro-list being a menace to qos like sqm already)

brada4, 2025-04-08:
> one more regression

Dont be shy, let it out.

tmcqueen-materials, 2025-04-08:
@mark8223 : can you please clarify whether disabling `rx-gro-list` fixes both problems you observed, namely:
   1. Sometimes getting a single packet of length 118 instead of two packets of length ~90 on the input side
   2. iSH simultaneous A and AAAA lookups only having DNAT applied to the first query.

And a couple quick followups:
   1. Does the iSH issue happen only when tcpdump shows a single packet on the input side, of longer length, instead of the expected pair?
   2. Does the same behavior happen with IPv4 sources, or is it limited to IPv6?

Thank you!

brada4, 2025-04-08:
Also from forum discussion
If  ` cat /sys/kernel/debug/ppe0/entries` is not empty power router off and on, check if problem goes away without ethtool.

brada4, 2025-04-09:
@nbd168 udp6 seems to be having similar or even worse issue (related to patches linked 2 posts ago)

tmcqueen-materials, 2025-04-09:
@mark8223 : thank you!

brada4, 2025-04-09:
Seems we are not alone
https://github.com/sipwise/rtpengine/commit/06eced5b28d648257f562ce32bad8df0916c8140

Shine-, 2025-04-09:
> @Shine- Don't worry, I tried with 24.10-SNAPSHOT first, It's the same behavior. I only posted the bug tied to a versioned build so we can get a reproducible environment. You never know when you get other regressions in an evermoving nightly build and now you're dealing with more unexpected problems that can skew the investigation.
> 
> The issue is unrelated to offloading settings. It behaves the same with offloading off, software offloading, or hardware offloading. That commit does not fix the issue.

Thanks for elaborating, in my opinion this is important information that should've already been part of the original problem description!

tmcqueen-materials, 2025-04-10:
@mark8223 @brada4 : Alright, I am now able to reproduce this on other hardware, e.g. DL-WRX36 (ipq807x) and x86_64. The key is to make sure multiple packets with the same 5-tuple (udp, src-ip, src-port, dest-ip, dest-port) arrive at the "same time", combined with `rx-gro-list` being enabled. 

This can be achieved by:
1. Running `dig -b ${clientIP}#18880 +nocookie +bufsize=0 @fd::1 www.google.com A @fd::1 www.google.com AAAA` twice while capturing packets (`tcpdump -w file.pcap src port 18880 and dst port 53`) *on a single* client system. This produces a pcap with 4 packets (2 A and 2 AAAA queries).
2. Editing `file.pcap` in wireshark to make all four packets have identical timestamps.
3. Replaying `edited.pcap` from the client system.

With such a file, on the router, you will almost always see, with `rx-gro-list` enabled, only a single packet of an incorrect long length on the input interface instead of the expected four of shorter length:
```
12:38:01.420691 IP6 (flowlabel 0x6bbd4, hlim 64, next-header UDP (17) payload length: 136) ${clientIP}.18880 > fd::1.53: [bad udp cksum 0x1849 -> 0x473b!] 26036+ A? www.google.com. (128)

```
While with `rx-gro-list` disabled, you will correctly see four packets of appropriate length on the input interface:
```
12:41:38.464702 IP6 (flowlabel 0x6bbd4, hlim 64, next-header UDP (17) payload length: 40) ${clientIP}.18880 > fd::1.53: [udp sum ok] 26036+ A? www.google.com. (32)
12:41:38.464702 IP6 (flowlabel 0x6bbd4, hlim 64, next-header UDP (17) payload length: 40) ${clientIP}.18880 > fd::1.53: [udp sum ok] 2227+ AAAA? www.google.com. (32)
12:41:38.464702 IP6 (flowlabel 0x6bbd4, hlim 64, next-header UDP (17) payload length: 40) ${clientIP}.18880 > fd::1.53: [udp sum ok] 62609+ A? www.google.com. (32)
12:41:38.464703 IP6 (flowlabel 0x6bbd4, hlim 64, next-header UDP (17) payload length: 40) ${clientIP}.18880 > fd::1.53: [udp sum ok] 9852+ AAAA? www.google.com. (32)
```

If these packets match a DNAT rule, then in both cases, all four packets transit the output interface. with `rx-gro-list` enabled, only the first packet has the destination IP rewritten, while with it disabled, all four packets are correctly rewritten.

With the above, I am able to reproduce this on both 23.05 (various versions) and 24.10 (.0 and SNAPSHOT)... and also on non-WRT linux when `rx-gro-list` is enabled, and using kernels 5.15 or 6.6.

This only happens with IPv6. With IPv4 and `rx-gro-list` enabled, I see two larger packets (instead of four smaller) [look to be gro combinations of packets 1&2 and 3&4], but all four are correctly DNATed.

So it appears this has surfaced a long-standing bug in linux's gro-list implementation as applied to IPv6.

I do not have ability to test this on kernel-next (i.e. if it was fixed sometime after 6.6), but maybe someone else here can?

In either case, I suspect the fix should be backported to LTS kernels -- there are security implications here because it means that carefully written client software can avoid linux firewall routing rules.

brada4, 2025-04-10:
Here raspberry
https://github.com/openwrt/openwrt/issues/18122
Lets wait for devs to chime in.

brada4, 2025-04-14:
Even older https://github.com/openwrt/openwrt/issues/16261

tmcqueen-materials, 2025-04-19:
Just confirming this issue is still present both in 24.10.1, and in 24.10-SNAPSHOT (as of 13:29 April 19th 2025 UTC).

brada4, 2025-04-20:
rx-gro-list is enabled by default by OpenWrt patch after 23.05.x

tmcqueen-materials, 2025-04-21:
@brada4 : yes, and that is a difference from upstream, which is presumably why issues like this don't have quite the same visibility for broader linux. 

If I were OpenWRT, I would remove that patch for now, at least for 24.10 branch.

brada4, 2025-04-21:
Kinda tagged author, lots of performances in bypassing firewall

https://github.com/openwrt/openwrt/commits/493b0cd8176ac53ba6377c09d8eb591afe6e0cd3/target/linux/generic/hack-6.6/600-net-enable-fraglist-GRO-by-default.patch

Neustradamus, 2025-04-22:
To follow this ticket.

tmcqueen-materials, 2025-04-26:
@brada4 @mark8223 : The patch posted by @nbd168 over in https://github.com/openwrt/openwrt/issues/18608#issuecomment-2832234620 resolves this issue in my tests. Can you also test and report back?

closed 2025-04-26 by openwrt-bot