openwrt/openwrt #18388: 24.10.0 - OpenMesh OM2P-HS v2 - jff2

sanosis, 2025-03-31:
### Describe the bug

Update from previous openwrt versions leads to effectively locked device - serial connection is necessary to do anything: no network, no overlay… firstboot results in clean openwrt, but with temporary overlay (jffs2 not ready yet, using temporary tmpfs overlay).

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ath79/generic

### Device

OpenMesh OM2P-HS v2

### Image kind

Official downloaded image

### Steps to reproduce

Update from previous releases of openwrt to 24.10.0

### Actual behaviour

Update commences, but the device is unresponsive - serial login shows, that the system boots but no network is configured, previous configs are gone and no overlay is mounted. Firstboot results in working device but with temporary overlay (jffs2 not ready yet, using temporary tmpfs overlay)

### Additional info

Firstboot & update to previous version (23.x) results in working device.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

sanosis, 2025-03-31:
For anyone facing same issue: serial pins (from Δ/rj45 and power sockets…):
1. ?(maybe VCC)
2. GND
3. TX
4. RX

DragonBluep, 2025-04-01:
Please attach the kernel log here.

sanosis, 2025-04-01:
same problem on the latest snapshot form firmware-selector:

Model
OpenMesh OM2P-HS v2
Platform
ath79/generic
Version
SNAPSHOT (r29126-7728f66606)

user.notice kernel: [   14.264228] mount_root: jffs2 not ready yet, using temporary tmpfs overlay
mount_root: jffs2 not ready yet, using temporary tmpfs overlay
[  582.148784] Too few erase blocks (3)

[snapshot_logread.txt](https://github.com/user-attachments/files/19545853/snapshot_logread.txt)