openwrt/openwrt #18391: Unable to build with authorized_keys in static files

EPinci, 2025-04-01:
### Describe the bug

I'm building my own image with a non-root user.
When building with a public key in `/files/etc/dropbear/authorized_keys` the folder is set with permissions that are missing the "x" bit resulting in the same user not being able to chdir and thus failing the build during `package/install`.

[image]

[image]

### OpenWrt version

r0-6418628

### OpenWrt release

24.10-SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

bpi-r4

### Image kind

Self-built image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2025-04-01:
authorized_keys should normally be a file, not a directory.

EPinci, 2025-04-01:
Dumb me, you're right, closing this one.