openwrt/openwrt #18394: 24.10 SIGSEGV - dont know cause - its danger?

tojestzart, 2025-04-01:
### Describe the bug

`Tue Apr  1 17:46:36 2025 kern.info kernel: [   88.917197] do_page_fault(): sending SIGSEGV to writer#1 for invalid read access from 0008a200
Tue Apr  1 17:46:36 2025 kern.info kernel: [   88.926144] epc = 77d9b0b0 in libc.so[77d44000+b2000]
Tue Apr  1 17:46:36 2025 kern.info kernel: [   88.931298] ra  = 77d9b570 in libc.so[77d44000+b2000]`

i updated my firmware to use banip 1.5.5 from 1.5.3 and i seen this on second restart, do i need to worry?

### OpenWrt version

24.10

### OpenWrt release

24.10

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

DragonBluep, 2025-04-01:
It looks like a null pointer issue happened on application `writer`. Are you using some unofficial packages?

tojestzart, 2025-04-01:
OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.090.70923~aade99d

repository is on https://firmware-selector.openwrt.org/ to this date 01/04/2025 builded by buildbots.

brada4, 2025-04-01:
Likely deficienvy in a package.
```
opkg search `which wtiter`
```

PS please edit your post recovering "invalid" fields from `ubus call system boaed`

tojestzart, 2025-04-01:
```
root@OpenWrt:~# opkg search 'which writer'
root@OpenWrt:~#

```
done nothing found.
I edited per your request, hopefully is correct.
Im closing this issue since i dont think this error will appear again

closed 2025-04-01 by tojestzart