openwrt/openwrt #18400: sysupgrade runned by LuCI killed if it runs for too long

remittor, 2025-04-02:
### Describe the bug

BUG can be reproduced by adding `sleep 15` to this place:

https://github.com/openwrt/openwrt/blob/65b8a978deac05894d1fd20dfe63494f4b43848d/package/base-files/files/sbin/sysupgrade#L381-L385

But if you run this script (`sysupgrade`) through the terminal, then everything works fine.

The bug only occurs when launched from LuCI:

https://github.com/openwrt/luci/blob/4921920821b31e467d381073db3b7282a663d5e9/modules/luci-mod-system/htdocs/luci-static/resources/view/system/flash.js#L332-L340

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/filogic

### Device

ASUS TUF-AX4200

### Image kind

Self-built image

### Steps to reproduce

BUG can be reproduced by adding `sleep 15` to this place:

https://github.com/openwrt/openwrt/blob/65b8a978deac05894d1fd20dfe63494f4b43848d/package/base-files/files/sbin/sysupgrade#L381-L385

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-04-03:
You mean sha256sum would take 15s?

remittor, 2025-04-04:
The source of the problem was localized:

https://github.com/openwrt/luci/blob/4921920821b31e467d381073db3b7282a663d5e9/modules/luci-base/htdocs/luci-static/resources/rpc.js#L37-L41

In 2019, this problem was more obvious. And maintainers had to increase the timeout to 20 sec:

https://github.com/openwrt/luci/commit/21d1266c53d07f02c3b24314c62e86f2fa82fe9b

remittor, 2025-04-04:
> You mean sha256sum would take 15s?

Are you 100% sure that this function always runs quickly?

https://github.com/openwrt/openwrt/blob/65b8a978deac05894d1fd20dfe63494f4b43848d/package/base-files/files/lib/upgrade/stage2#L159

https://github.com/openwrt/openwrt/blob/65b8a978deac05894d1fd20dfe63494f4b43848d/package/base-files/files/lib/upgrade/stage2#L121-L124

remittor, 2025-04-05:
@systemcrash , I just thought, maybe it's only a problem with LuCI... What do you think?

remittor, 2025-04-09:
> Are you 100% sure that this function always completes fast?

I just searched the OpenWrt forum and found that the process of killing extra processes (and creating the ramdisk) usually takes between 4 and 12 seconds.

Example for `OpenWrt One` (12 sec):
```
Thu Oct 24 20:49:29 UTC 2024 upgrade: Commencing upgrade. Closing all shell sessions.
root@OpenWrt:/tmp# Watchdog handover: fd=3
- watchdog -
Watchdog does not have CARDRESET support
Thu Oct 24 20:49:29 UTC 2024 upgrade: Sending TERM to remaining processes ...
Thu Oct 24 20:49:33 UTC 2024 upgrade: Sending KILL to remaining processes ...
[  216.901428] stage2 (3534): drop_caches: 3
Thu Oct 24 20:49:40 UTC 2024 upgrade: Switching to ramdisk...
Thu Oct 24 20:49:41 UTC 2024 upgrade: Performing system upgrade...
```

Therefore, it is foolish to hope that in all cases this process will fit into the allotted 20 seconds.

systemcrash, 2025-04-09:
If the process is spawned by rpcd it's possible that it times out. Weird.