openwrt/openwrt #18404: (packages) build docker image broke today: halts at GPG key check

systemcrash, 2025-04-03:
Is this known/expected?

```
ImageID
Digest
Metadata
Reference
Check build summary support
Run docker run --rm \
...
2025-04-03 17:51:38 (5.93 MB/s) - ‘sha256sums.asc’ saved [299/299]
gpg: directory '/builder/.gnupg' created
gpg: keybox '/builder/.gnupg/pubring.kbx' created
gpg: key 1D53D1877742E911: 3 signatures not checked due to missing keys
gpg: /builder/.gnupg/trustdb.gpg: trustdb created
gpg: key 1D53D1877742E911: public key "OpenWrt Build System (Nitrokey3) <contact@openwrt.org>" imported
gpg: key 28A39BC32074BE7A: 2 signatures not checked due to missing keys
gpg: key 28A39BC32074BE7A: public key "OpenWrt Build System (PGP key for 19.07 release builds) <pgpsign-19.07@openwrt.org>" imported
gpg: key CD84BCED626471F1: 2 signatures not checked due to missing keys
gpg: key CD84BCED626471F1: public key "OpenWrt Build System (PGP key for unattended snapshot builds) <pgpsign-snapshots@openwrt.org>" imported
gpg: key 88CA59E88F681580: public key "OpenWrt Build System (PGP key for 21.02 release builds) <pgpsign-21.02@openwrt.org>" imported
gpg: key CD54E82DADB3684D: public key "OpenWrt Build System (GnuPGP key for 22.03 release builds) <pgpsign-22.03@openwrt.org>" imported
gpg: Total number processed: 5
gpg:               imported: 5
gpg: no ultimately trusted keys found
rm: cannot remove '/builder/keys/88CA59E8.asc': Permission denied
rm: cannot remove '/builder/keys/CD54E82DADB3684D.asc': Permission denied
rm: cannot remove '/builder/keys/2074BE7A.asc': Permission denied
rm: cannot remove '/builder/keys/0x1D53D1877742E911.asc': Permission denied
rm: cannot remove '/builder/keys/626471F1.asc': Permission denied
Error: Process completed with exit code 1.
```

systemcrash, 2025-04-03:
Or is this just some github weather?

ynezz, 2025-04-03:
Duplicate of https://github.com/openwrt/docker/pull/165

closed 2025-04-03 by ynezz