openwrt/openwrt #18405: All Ethernet Ports died after installing 24.10

barcell1, 2025-04-03:
### Describe the bug

I had a fully functional Archer Tplink  C6 V3.2 with stock firmware, and then in the process of installing 24.10 , something goes wrong because router got fixed led just on the power (others led become disabled ) also all ethernets ports stopped to work

I used the serial to reach uboot, that also didnt recognize ethernet ports, just the switch eth0, also downgrade to 23.05, and same problem , both 24.10, 23.05 and uboot just recognize eth0

I dont expect to this problem got fixed because looks like a hardware failed, but im opening this bug report because its very possible that the installation of this firmware (24.10) lead to hardware fail, since it was working well just before the installation

also is important  to mention, i have others exactly similar router that i installed 23.05 without a problem on all of them

### OpenWrt version

24.10

### OpenWrt release

24.10

### OpenWrt target/subtarget

24.10

### Device

Archer c6 V3.2

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-04-03:
Remove dsa master eth0 from the br-lan

barcell1, 2025-04-04:
> Remove dsa master eth0 from the br-lan

sorry how i do that ? and do you think it would make any difference if even in uboot , that do not contain openwrt, the ethernets do not show up ?

brada4, 2025-04-04:
> eth0: configuring for fixed/rgmii 

That is a port to the dsa switch which is wrongly assigned to br-lan aka misconfiguration.
Could you boot device in failsafe from poweron to see what system without your config can do.

brada4, 2025-04-05:
what do you see in `ip link` (last half of macs not needed) in failsafe vs normal boot.