openwrt/openwrt #18410: lantiq/xrx200: ImageBuilder: No rule to make target openwrt-lantiq-xrx200-arcadyan_arv7519rw22-squashfs-sysupgrade.bin

ynezz, 2025-04-04:
### Describe the bug

https://github.com/openwrt/docker/actions/runs/14251313226/job/39944529774#step:7:1224

```shell
[mktplinkfw2] firmware file "/builder/build_dir/target-mips_24kc_musl/linux-lantiq_xrx200/tmp/openwrt-lantiq-xrx200-tplink_vr200v-squashfs-sysupgrade.bin.new" completed
make[3]: *** No rule to make target '/builder/build_dir/target-mips_24kc_musl/linux-lantiq_xrx200/arcadyan_arv7519rw22-kernel.bin', needed by '/builder/build_dir/target-mips_24kc_musl/linux-lantiq_xrx200/tmp/openwrt-lantiq-xrx200-arcadyan_arv7519rw22-squashfs-sysupgrade.bin'.  Stop.
make[2]: *** [Makefile:263: build_image] Error 2
make[1]: *** [Makefile:159: _call_image] Error 2
make: *** [Makefile:332: image] Error 2
```

### OpenWrt version

Snapshot

### OpenWrt release

Snapshot

### OpenWrt target/subtarget

lantiq/xrx200

### Device

Default profile

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-04-04:
The device has `DEFAULT := n` in the device recipe - is that related?

robimarko, 2025-04-14:
This is expected since `arcadyan_arv7519rw22` is set to `DEFAULT := n` so nothing is being built for it on buildbots, hence no kernel for it and thus IB will fail for it.

efahl, 2025-04-29:
This breaks the creation of the docker image used by ASU (https://github.com/openwrt/docker/actions/runs/14438695035/job/40484177868), so none of the subtargets for the lantiq/xr200 are able to be upgraded using the ASU tools.  How can we get this resolved so all the non-arcadyan_arv7519rw22 device users don't suffer?  (Here's one of several reports https://forum.openwrt.org/t/bt-home-hub-5a-snapshot-24-10-1-firmware-build-fail/230656/6)

robimarko, 2025-04-29:
Well, basically, we need to add support for IB to honor `DEFAULT := n` as well and not build images by default for profiles that have it set.

Or avoid opting out of smoke testing for certain subtargets, though there are probably more that have the same issue

efahl, 2025-04-30:
Does this look like it's on the right track?  I remove those subtargets from `PROFILE_NAMES` in `profiles.mk` by filtering out those whose `DEFAULT` explicitly starts with `n`...  This results in `make info` not listing the offending targets, and all subsequent info is also purged.

```
diff --git a/scripts/target-metadata.pl b/scripts/target-metadata.pl
index ce96d1e7d538..83a3a5e9e6ed 100755
--- a/scripts/target-metadata.pl
+++ b/scripts/target-metadata.pl
@@ -441,9 +441,10 @@ sub gen_profile_mk() {
        my @targets = parse_target_metadata($file);
        foreach my $cur (@targets) {
                next unless $cur->{id} eq $target;
-               my @profile_ids_unique =  do { my %seen; grep { !$seen{$_}++} map { $_->{id} } @{$cur->{profiles}}};
+               my @profile_ids_unique =  do { my %seen; grep { !$seen{$_}++} map { $_->{id} } grep { $_->{default} !~ /^n/ } @{$cur->{profiles}}};
                print "PROFILE_NAMES = ".join(" ", @profile_ids_unique)."\n";
                foreach my $profile (@{$cur->{profiles}}) {
                        print $profile->{id}.'_NAME:='.$profile->{name}."\n";
                        print $profile->{id}.'_HAS_IMAGE_METADATA:='.$profile->{has_image_metadata}."\n";
                        if (defined($profile->{supported_devices}) and @{$profile->{supported_devices}} > 0) {
```

Over in the imagebuilder
```
$ make info
Current Target: "lantiq/xrx200"
Current Architecture: "mips"
Current Revision: "r28597-0425664679"
Default Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-mbedtls logd mtd netifd uci uclient-fetch urandom-seed urngd kmod-leds-gpio kmod-gpio-button-hotplug ltq-vdsl-vr9-vectoring-fw-installer kmod-ltq-vdsl-vr9-mei kmod-ltq-vdsl-vr9 kmod-ltq-atm-vr9 kmod-ltq-deu-vr9 kmod-ltq-ptm-vr9 ltq-vdsl-vr9-app dsl-vrx200-firmware-xdsl-a dsl-vrx200-firmware-xdsl-b-patch ppp-mod-pppoa procd-ujail dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe opkg
Available Profiles:

Default:
    Default Profile
    Packages:
    hasImageMetadata: 0
avm_fritz3370-rev2-hynix:
    AVM FRITZ!Box 3370 Rev. 2 (Hynix NAND)
    Packages: kmod-ath9k wpad-basic-mbedtls kmod-usb-dwc2 fritz-tffs iwinfo
    hasImageMetadata: 1
    SupportedDevices: avm,fritz3370-rev2-hynix
...
```
where both of 
  - `arcadyan_arv7519rw22` 
  - `zyxel_p-2812hnu-f3`

have been removed from the list.

andybjackson, 2025-05-07:
Gentle nudge.

robimarko, 2025-05-07:
@efahl I guess that would work as I see no point in a device that we dont build by default being visible, so please make a PR

closed 2025-05-09 by openwrt-bot

efahl, 2025-05-09:
@ynezz Does the PR need backport to 24.10?  If so, do we also need to manually force imagebuilder rebuilds on the affected targets?

A rough guess at what's broken right now shows maybe a half dozen targets...
```
$ git grep 'DEFAULT *:= *n' | sort -u
target/linux/at91/image/sam9x.mk:  DEFAULT := n
target/linux/ath79/image/common-tp-link.mk:  DEFAULT := n
target/linux/ath79/image/generic.mk:  DEFAULT := n
target/linux/ath79/image/generic-tp-link.mk:  DEFAULT := n
target/linux/ath79/image/tiny.mk:  DEFAULT := n
target/linux/ath79/image/tiny-netgear.mk:  DEFAULT := n
target/linux/ath79/image/tiny-tp-link.mk:  DEFAULT := n
target/linux/ath79/image/tiny-ubnt.mk:  DEFAULT := n
target/linux/bcm47xx/image/generic.mk:  DEFAULT := n
target/linux/bcm47xx/image/legacy.mk:  DEFAULT := n
target/linux/bcm47xx/image/mips74k.mk:  DEFAULT := n
target/linux/bcm53xx/image/Makefile:  DEFAULT := n
target/linux/ipq40xx/image/generic.mk:  DEFAULT := n
target/linux/ixp4xx/image/Makefile:     DEFAULT := n
target/linux/lantiq/image/amazonse.mk:  DEFAULT := n
target/linux/lantiq/image/ar9.mk:  DEFAULT := n
target/linux/lantiq/image/danube.mk:  DEFAULT := n
target/linux/lantiq/image/falcon.mk:  DEFAULT := n
target/linux/lantiq/image/vr9_legacy.mk:  DEFAULT := n
target/linux/lantiq/image/vr9.mk:  DEFAULT := n
target/linux/lantiq/image/xway_legacy.mk:  DEFAULT := n
target/linux/ramips/image/mt7620.mk:  DEFAULT := n
target/linux/ramips/image/mt76x8.mk:  DEFAULT := n
target/linux/ramips/image/rt288x.mk:  DEFAULT := n
target/linux/ramips/image/rt305x.mk:  DEFAULT := n
target/linux/ramips/image/rt3883.mk:  DEFAULT := n
```

robimarko, 2025-05-09:
Most likely backport to 24.10 is needed as well, nothing needs to be forced, IB is built when targets are built, it will take a day or two for buildbots to catch up

efahl, 2025-05-09:
I was under the impression that the point releases were never rebuilt, except by manual intervention.  For example, looking at https://downloads.openwrt.org/releases/24.10.1/targets/lantiq/xrx200/ all the artifacts are dated Apr 14, the day before the official release.

I was hoping to get at least the image builder for lantiq/xrx200 on 24.10.1, as there are a handful of ASU users not able to proceed because the docker image fails its smoke test, thus affecting all devices, not just the couple that have `DEFAULT:=n`...

(I'll do a backport PR for 24.10, so at least 24.10-snapshot will get the fix.)

andybjackson, 2025-05-14:
SNAPSHOT still has GPG issue. In my case for lantiq/xrx200 British Telecom (BT) Home Hub 5A. Suspect that without further manual intervention, this will never see correction. I have waited through two or three release updates.

robimarko, 2025-05-14:
> SNAPSHOT still has GPG issue. In my case for lantiq/xrx200 British Telecom (BT) Home Hub 5A. Suspect that without further manual intervention, this will never see correction. I have waited through two or three release updates.

That's cause for some reason last IB container was pushed for main/master a month ago, I see a push is queued up in actions once runner is available

efahl, 2025-05-14:
> Well, its still failing on smoke-testing, now its missing:
> 
> ```
> make[3]: *** No rule to make target '/builder/build_dir/target-mips_24kc_musl/linux-lantiq_xrx200/arcadyan_arv7519rw22-kernel.bin', needed by '/
> ```

Grr, ok so when you do `make image` without an explicit `PROFILE`, the imagebuilder chooses `tplink_vr200` as its target, but somehow that implicit profile wants the `arcadyan_arv7519rw22` kernel?  Truly WTF.  

If I explicitly specify `make image PROFILE=tplink_vr200`, everything works fine.  I'm trying to figure out how those target makefiles are selecting their dependencies and mixing the profiles, it's somewhere inside wherever this ends up: https://github.com/openwrt/openwrt/blob/main/target/imagebuilder/files/Makefile#L264

Or maybe that makefile should just do a substitution when `PROFILE=<nothing>`, and pluck the first one from the profile list and make it explicit???

Or a dirtier workaround would be to change the docker smoke test to just use an explicit profile...

```
$ profile=$(awk '/PROFILE_NAME/ {sub(/DEVICE_/, "", $4); print $4}' .profiles.mk)
$ make info PROFILE=$profile
...
```

robimarko, 2025-05-14:
No, it does not select `tplink_vr200`, there is a "Default" profile in `xrx200` so that ends up being used but since that recipe lacks anything to generate an image somehow the first recipe from `vr9.mk` ends up being used (And that is our `arcadyan_arv7519rw22`).

I tested other targets that have "Default" profile and they all behave the same, I really have no clue what is even the purpose of the stupid "Default" profile

efahl, 2025-05-14:
Right, it's internally "Default", but that makes a weird combo of the tplink and arcadyan profiles:
```
[mktplinkfw2] firmware file "build_dir/target-mips_24kc_musl/linux-lantiq_xrx200/tmp/openwrt-lantiq-xrx200-tplink_vr200v-squashfs-sysupgrade.bin.new" completed
make[3]: *** No rule to make target 'build_dir/target-mips_24kc_musl/linux-lantiq_xrx200/arcadyan_arv7519rw22-kernel.bin', needed by 'build_dir/target-mips_24kc_musl/linux-lantiq_xrx200/tmp/openwrt-lantiq-xrx200-arcadyan_arv7519rw22-squashfs-sysupgrade.bin'.  Stop.
```

So, how about I supplement my recent PR to also eliminate `Default` from `PROFILE_NAMES` in `.profiles.mk`?  That seems to work fine, `make image` then uses `DEVICE_avm_fritz3370-rev2-hynix` as the value for `USER_PROFILE` and it builds just fine, no other side effects seen with explicit targets (except of course, you can no longer say `make image PROFILE=Default`).

robimarko, 2025-05-14:
Well, the thing is that I lack the understanding of why the `Default` profiles even exist and if something would break.

Cause I dont see a purpose for them at all, and it makes sense to just remove their definitions from the targets instead of filtering them out for IB.

So, I would go ahead with your PR and we can try to tag some people to see if anybody has an idea about their purpose

efahl, 2025-05-14:
The odd thing is that "Default" isn't in all targets.  The x86/64 imagebuilder just has a single entry, `DEVICE_generic`, in its profile names...  Likewise, ath79/generic doesn't have one either, just its massive list of profiles.

I've got something ready, still testing it a bit, but should submit it soon.

robimarko, 2025-05-14:
> The odd thing is that "Default" isn't in all targets. The x86/64 imagebuilder just has a single entry, `DEVICE_generic`, in its profile names... Likewise, ath79/generic doesn't have one either, just [...]

Check out IRC log:
https://oftc.catirclogs.org/openwrt-devel/2025-05-14

The thing that I understand is how `PROFILES := Default` expands

efahl, 2025-05-14:
So based on your discussion, I should amend my PR with "Generic", too, it seems.

```
$ scripts/target-metadata.pl profile_mk tmp/.targetinfo bcm53xx/generic | head -1
PROFILE_NAMES = Generic DEVICE_asus_rt-ac3100 DEVICE_asus_rt-ac56u ...
```

efahl, 2025-05-14:
Did we get them all?
```
$ git grep 'PROFILES *='
target/linux/airoha/image/Makefile:  PROFILES = Default $$(DEVICE_NAME)
target/linux/ath79/image/Makefile:  PROFILES = Default
target/linux/bcm53xx/image/Makefile:  PROFILES = Generic $$(DEVICE_NAME)
target/linux/bmips/image/Makefile:  PROFILES = Default $$(DEVICE_NAME)
target/linux/lantiq/image/Makefile:  PROFILES = Default
target/linux/mediatek/image/Makefile:  PROFILES = Default $$(DEVICE_NAME)
target/linux/octeon/image/Makefile:  PROFILES = Default $$(DEVICE_NAME)
target/linux/ramips/image/Makefile:  PROFILES = Default
target/linux/realtek/image/Makefile:  PROFILES = Default
target/linux/siflower/image/sf21.mk:  PROFILES = Default $$(DEVICE_NAME)
```

ynezz, 2025-05-15:
Thanks for looking into this!

> Or a dirtier workaround would be to change the docker smoke test to just use an explicit profile...

If it helps and works, then why not. The idea of the smoke test is to just check, that the IB container will be somehow usable. Additionally it might be safer for 24.10 handling, I'm not sure, that we want to backport such changes into the stable release.

efahl, 2025-05-15:
> The idea of the smoke test is to just check, that the IB container will be somehow usable. Additionally it might be safer for 24.10 handling, I'm not sure, that we want to backport such changes into the stable release.

@ynezz If you do implement this, make sure to grab the _second_ (or last?) target in the `PROFILE_NAMES` as the first one is occasionally `Default`, which results in the same "missing kernel"...  Hmm, some targets only have one profile, so "second" doesn't exist.

Yeah, the last profile seems to work on all the targets I just tested (ath79/generic, lantiq/xrx200, x86/64).  Both of these seem to do the job, but the first one requires that one run `make` first, to create the `.profiles.mk`.
```
$ profile=$(awk '/PROFILE_NAME/ {sub(/DEVICE_/, "", $NF); print $NF}' .profiles.mk)

or

$ make info | awk -F: '/^[^ ]/ {p =$1} END {print p}'
```

robimarko, 2025-05-17:
@andybjackson XRX200 containers were finally built and pushed so now firmware selector works for it again

andybjackson, 2025-05-18:
Thank you. Especially to @robimarko & @efahl. Sadly the ASU has some issues right now that mean I can't build my firmware of choice, but I'm sure that will resolve itself with time.

commit f060615a (2025-05-16): image: respect DEFAULT and BROKEN when Default profile is selected

Currently, when you select the Default profile it does not honor DEFAULT:=n
nor BROKEN:=y in device profiles but rather just tries to build all of them.

This may work when building directly, but when using Image Builder it will
always fail since no kernel or anything else is present for devices that
have DEFAULT:=n or BROKEN:=Y set since those are skipped during build.

So, lets look for DEFAULT being set to "n" or BROKEN being set to "y" and
then remove clear _PROFILE_SET so they dont end up being marked for
installation.

F [... cut]
PR #18814 (2025-05-16): image: respect DEFAULT:=n when Default profile is selected
Currently, when you select the Default profile it does not honor DEFAULT:=n in device profiles but rather just tries to build all of them.

This may work when building directly, but when using Image Builder it will always fail since no kernel or anything else is present for devices that have DEFAULT:=n set since those are skipped during build.

So, lets look for DEFAULT variable being set and then mark the target for those devices as install-disabled as to not try and install them.

Fixes: #18410
commit 6d455442 (2025-05-18): image: respect DEFAULT and BROKEN when Default profile is selected

Currently, when you select the Default profile it does not honor DEFAULT:=n
nor BROKEN:=y in device profiles but rather just tries to build all of them.

This may work when building directly, but when using Image Builder it will
always fail since no kernel or anything else is present for devices that
have DEFAULT:=n or BROKEN:=Y set since those are skipped during build.

So, lets look for DEFAULT being set to "n" or BROKEN being set to "y" and
then remove clear _PROFILE_SET so they dont end up being marked for
installation.

F [... cut]