openwrt/openwrt #18420: [MT7621] EA6350 v4: PCIe init fails sporadically, system hangs before init

DoWenzl94, 2025-04-04:
### Describe the bug

🧠 Summary:

After rebooting, my EA6350 v4 occasionally fails to boot properly. The system hangs early during the boot process. It does not respond to ping, SSH, or enter failsafe mode. Power cycling the device several times eventually allows a successful boot.

⸻

🔍 Observed behavior (in bootlogs):
	•	During failed boots, the last visible line in dmesg ends right after the UART/console is enabled.
	•	Crucially, PCIE0 enabled does not appear in the log.
	•	In successful boots, I consistently see:

mt7621-pci 1e140000.pcie: PCIE0 enabled
mt7621-pci 1e140000.pcie: PCIE1 enabled

This matches known issues with MT7621-based devices where PCIe initialization silently fails, and the system hangs before init or procd is launched.

💬 Related reports:

Users in Japanese OpenWrt communities (e.g., [Thread from 2025-03](https://mao-5ch-net.translate.goog/test/read.cgi/network/1721470536/l50?_x_tr_sl=auto&_x_tr_tl=de&_x_tr_hl=de&_x_tr_pto=wapp)) describe identical behavior on other MT7621 routers like the WN-DX1167R and WN-AX1167GR2.

According to them, this issue began after OpenWrt switched to the mainline PCIe driver for MT7621 (post kernel 5.4).
It’s suspected that the driver shuts down PHYs if certain PCIe slots are “empty”, which causes the PCIe bus to hang.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

Linksys EA6350 v4

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-04-05:
Does it fail on cold boot too?

DoWenzl94, 2025-04-05:
> Does it fail on cold boot too?

Yes

paraka, 2025-04-10:
With the description provided I cannot be sure it is a PCIe issue. Regarding "the driver shuts down PHYs if certain PCIe slots are “empty”, which causes the PCIe bus to hang", this is not true at all, I have different boards that have slots empty and the pci just works properly. And also in your case phy0 and phy1 are used so phy2 is the only one empty so it is just not enabled at all.... Can you provide a complete pci boot trace so I can check if anything looks weird?

DoWenzl94, 2025-04-10:
> With the description provided I cannot be sure it is a PCIe issue. Regarding "the driver shuts down PHYs if certain PCIe slots are “empty”, which causes the PCIe bus to hang", this is not true at [...]

Thanks for your quick reply!
You’re absolutely right – I don’t have proof that this is a PCIe issue, it just appeared to be the most consistent symptom across multiple logs.

However, I’ve now recovered both the dmesg and logread logs from both a failed and a successful boot. I’ll upload them here for reference.

1743799812 is successful
1743799743 not

[dmesg_1743799812.txt](https://github.com/user-attachments/files/19693708/dmesg_1743799812.txt)
[logread_1743799743.txt](https://github.com/user-attachments/files/19693707/logread_1743799743.txt)
[dmesg_1743799743.txt](https://github.com/user-attachments/files/19693709/dmesg_1743799743.txt)
[logread_1743799812.txt](https://github.com/user-attachments/files/19693710/logread_1743799812.txt)

neheb, 2025-04-11:
diff is

```
colordiff 1 2
23,24c23,24
<  Writing ErrCtl register=0004286c
<  Readback ErrCtl register=0004286c
---
>  Writing ErrCtl register=00042860
>  Readback ErrCtl register=00042860
101c101
<  pci 0000:00:00.0: reg 0x14: [mem 0x00000000-0x0000ffff
---
>  pci 0000:00:00.0: reg 0x14: [mem 0x60500000-0x6050ffff
106c106
<  pci 0000:00:01.0: reg 0x14: [mem 0x00000000-0x0000ffff
---
>  pci 0000:00:01.0: reg 0x14: [mem 0x60510000-0x6051ffff
109,110d108
<  pci 0000:00:00.0: bridge configuration invalid ([bus 00-00
<  pci 0000:00:01.0: bridge configuration invalid ([bus 00-00
116,117c114,115
<  pci 0000:00:00.0:   bridge window [mem 0x00000000-0x000fffff
<  pci 0000:00:00.0:   bridge window [mem 0x00000000-0x000fffff pref
---
>  pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff
>  pci 0000:00:00.0:   bridge window [mem 0x60100000-0x601fffff pref
128,129c126,127
<  pci 0000:00:01.0:   bridge window [mem 0x00000000-0x000fffff
<  pci 0000:00:01.0:   bridge window [mem 0x00000000-0x000fffff pref
---
>  pci 0000:00:01.0:   bridge window [mem 0x60200000-0x602fffff
>  pci 0000:00:01.0:   bridge window [mem 0x60300000-0x604fffff pref
217c215
<  ubi0: max/mean erase counter: 8/3, WL threshold: 4096, image sequence number: 1738624177
---
>  ubi0: max/mean erase counter: 7/3, WL threshold: 4096, image sequence number: 1738624177
219c217
<  ubi0: background thread "ubi_bgt0d" started, PID 474
---
>  ubi0: background thread "ubi_bgt0d" started, PID 471
260a259
>  mt7530-mdio mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control off
262c261
<  UBIFS (ubi0:1): background thread "ubifs_bgt0_1" started, PID 665
---
>  UBIFS (ubi0:1): background thread "ubifs_bgt0_1" started, PID 661
271a271
>  mt7530-mdio mdio-bus:1f lan1: Link is Down
282c282
<  pci 0000:00:00.0: enabling device (0000 -> 0003)
---
>  pci 0000:00:00.0: enabling device (0006 -> 0007)
289d288
<  urngd: v1.0.2 started.
291c290
<  pci 0000:00:01.0: enabling device (0000 -> 0003)
---
>  pci 0000:00:01.0: enabling device (0006 -> 0007)
295a295
>  urngd: v1.0.2 started.
```

weird.

paraka, 2025-04-11:
Can you share your pci related device tree nodes and explain what you have plugged in them? 

There is one weird message in both traces that I have never see before in my board boot traces:

[    1.436302] pci_bus 0000:00: Some PCI device resources are unassigned, try booting with pci=realloc

paraka, 2025-04-11:
> Can you share your pci related device tree nodes and explain what you have plugged in them?

I guess is this as it is: https://github.com/openwrt/openwrt/blob/main/target/linux/ramips/dts/mt7621_linksys_ea7xxx.dtsi

paraka, 2025-04-11:
> <  pci 0000:00:00.0: reg 0x14: [mem 0x00000000-0x0000ffff
> ---
> >  pci 0000:00:00.0: reg 0x14: [mem 0x60500000-0x6050ffff <- NO SENSE
> 106c106
> <  pci 0000:00:01.0: reg 0x14: [mem 0x00000000-0x0000ffff
> ---
> >  pci 0000:00:01.0: reg 0x14: [mem 0x60510000-0x6051ffff <- NO SENSE
> 109,110d108
> <  pci 0000:00:00.0: bridge configuration invalid ([bus 00-00
> <  pci 0000:00:01.0: bridge configuration invalid ([bus 00-00
> 116,117c114,115
> <  pci 0000:00:00.0:   bridge window [mem 0x00000000-0x000fffff
> <  pci 0000:00:00.0:   bridge window [mem 0x00000000-0x000fffff pref
> ---
> >  pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff <- NO SENSE
> >  pci 0000:00:00.0:   bridge window [mem 0x60100000-0x601fffff pref <- NO SENSE
> 128,129c126,127
> <  pci 0000:00:01.0:   bridge window [mem 0x00000000-0x000fffff
> <  pci 0000:00:01.0:   bridge window [mem 0x00000000-0x000fffff pref
> ---
> >  pci 0000:00:01.0:   bridge window [mem 0x60200000-0x602fffff <- NO SENSE 
> >  pci 0000:00:01.0:   bridge window [mem 0x60300000-0x604fffff pref <- NO SENSE
> 217c215
>

All of this are the bridge windows of "PCI bridge to [bus 00]" which should have only sizes that will be applied after, when you see the following in normal enumeration trace:

[   21.921801] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[   21.937654] pci 0000:00:01.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[   21.953536] pci 0000:00:02.0: bridge configuration invalid ([bus 00-00]), reconfiguring

I don't have any idea of why are you getting already assigned addresses here... Might be like something is not properly reseted on your board between soft and cold resets?

paraka, 2025-04-11:
> Can you share your pci related device tree nodes and explain what you have plugged in them?
> 
> There is one weird message in both traces that I have never see before in my board boot traces:
> 
> [ 1.436302] pci_bus 0000:00: Some PCI device resources are unassigned, try booting with pci=realloc

Not true, I also have this from 6.x kernel.

DoWenzl94, 2025-04-11:
Thanks again for the input!

I’ve found the full live device tree node at `/proc/device-tree/pcie@1e140000/`, including the three subnodes:
- `pcie@0,0` (used – likely MT7603)
- `pcie@1,0` (used – likely MT7615)
- `pcie@2,0` (unused)

To make analysis easier, I’ve attached a `.tar.gz` dump of the full PCIe node from `/proc/device-tree`.

[pcie-dump.tar.gz](https://github.com/user-attachments/files/19706532/pcie-dump.tar.gz)

Let me know if you’d prefer a flat-text dump of specific fields (like `reg`, `ranges`, `interrupts`, etc.) or a decompiled `.dtb`.

Also, for clarity:  

The boot behavior is the same regardless of whether I do a **soft reboot** or a **cold power cycle** – the issue appears randomly in both cases.

Thanks again!

paraka, 2025-04-11:
Thanks,

Nothing looks weird to my eyes regarding the device tree you uploaded ...

Can you confirm that you have experimented the same erratic boot behaviour on a cold power cycle after the board has been power off for a while?

Can you provide a "lspci -vvv" command output of a good and bad boots?

DoWenzl94, 2025-04-13:
Thanks for the follow-up!

Yes, I’ve tested with cold boots where I removed power completely and waited for ~30 seconds before powering the device back on. Unfortunately, this doesn’t change the behavior – the issue is still sporadic and can happen regardless of how long the device has been off.

Regarding the lspci -vvv output from a failed boot – unfortunately, that’s not possible to capture:
When the system enters this “bad” state, it becomes completely unreachable:

	•	No web interface (LuCI)
	•	No SSH
	•	No ping

It hangs early enough that user space never comes up, and I can’t interact with the system at all.

Let me know if you’d like me to test with early boot flags like pci=realloc, earlyprintk, or provide other logs from a successful boot.

DoWenzl94, 2025-05-03:
> [@DoWenzl94](https://github.com/DoWenzl94), would you be able to confirm that (1) it reproduces with 23.05.5, but (2) not with 23.05.4? The differences between them are few enough that they could be relatively easy to study and identify what's wrong.

Thanks again for the guidance – here’s an update:

I went ahead and tested multiple OpenWrt versions, including 23.05.5 and 23.05.4.
Surprisingly, I could not reproduce the issue with either of them – everything worked fine after several reboots.

I then reinstalled my custom firmware and restored the backup configuration.
The first two reboots worked perfectly, and for a moment, I really thought the issue was resolved.

However, after the third reboot, the exact same failure occurred again: the system froze early in the boot process and became completely unresponsive (no serial, no SSH, no web UI, no ping).

⸻

🧩 Current situation:

At this point, I’m starting to suspect a possible hardware issue.
	•	I can’t identify any consistent pattern — not related to firmware version, configuration, or specific changes.
	•	The only thing I’ve noticed is that the problem seems to get worse the more I reboot.
	•	In most cases, the only reliable recovery method is to disconnect all cables (including power and Ethernet) for several minutes — and even that doesn’t always guarantee a clean boot right away.

⸻

Let me know if there’s anything else I can try to further isolate the root cause.
I’m still happy to help test things – but I’m honestly at a bit of a loss at this point.