openwrt/openwrt #18423: For Linksys MR7350 router, file etc/fw_env.config does exit and etc/config/ubootenv has zero content

iv7777, 2025-04-05:
### Describe the bug

After flashing the latest snapshot build for Linksys MR7350 router, file etc/fw_env.config does exit and etc/config/ubootenv has zero content. The problem makes the router stuck at its current firmware. Flashing a new one and it still boot into its current one.

### OpenWrt version

r29144-a947be41b7

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq60xx

### Device

Linksys MR7350

### Image kind

Official downloaded image

### Steps to reproduce

Just flash the SNAPSHOT image

### Actual behaviour

After flashing the SNAPSHOT image, the router will get stuck at this firmware forever. Flashing a different image and reboot, the router is still running the same firmware, not switching to the one just flashed.

### Expected behaviour

The router will switch to the new image that's just flashed.

### Additional info

Re-run the compilation of an older version in my own build environment based on a previous successful build, it will create an image with the exact content but introduce this new problem. My guess is the upsteam content of uboot-tools caused this. It happened after switching to uboot-tools from uboot-envtools in the OpenWrt repository.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2025-04-11:
Can you post your device partition table?

I have tried to reproduce this on my board, but `fw_env.config` is properly populated:
```
cat /etc/fw_env.config 
/dev/mtd11 0x0 0x40000 0x20000
```

iv7777, 2025-05-05:
It's been resolved in the recent build. Didn't know what went wrong and what has been resolved. At least it's working now.

closed 2025-05-05 by iv7777