openwrt/openwrt #18436: ED25519 Dropbear not available on version 25.10.0

M0rPH3U53, 2025-04-07:
### Describe the bug

Unable to connect with an ed25519 key, will the problem be resolved in a future version of Dropbear?

### OpenWrt version

25.10.0

### OpenWrt release

nothing

### OpenWrt target/subtarget

nothing

### Device

neatgear r6020

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

dave14305, 2025-04-07:
Same as https://github.com/openwrt/openwrt/issues/17901?

brada4, 2025-04-08:
Curve crypto is disabled for lowmem ie 8/64 devices in 24.10

M0rPH3U53, 2025-04-08:
how do I activate it?

M0rPH3U53, 2025-04-08:
No, I want ED25519. It's lighter and more secure for me.
Do you know why they removed this crypto?

aimacintyre, 2025-04-08:
> how do I activate it?

Compile a custom firmware for your device with the Dropbear ED25519 option enabled.

> Do you know why they removed this crypto?

to reduce flash space usage on 8M flash devices.

Neustradamus, 2025-04-08:
@M0rPH3U53: Can you rename the title and your description?
It is 24.10...