openwrt/openwrt #18441: dns-rr entry does not appear to be available from dnsmasq

systemdlete, 2025-04-08:
### Describe the bug

I created a Resource Record for a TXT type (16) in openwrt.  

When I run nslookup for test-chimaera from the openwrt command line, it only returns the IP address, but not the Resource Record (13 or 16; both fail to show).  Similar outcome when calling nslookup or host from another system whose DNS is pointed to the same openwrt router in question.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

x86/64

### Device

innotek GmbH VirtualBox

### Image kind

Official downloaded image

### Steps to reproduce

I created a resource record using the LuCI->Network->DHCP and DNS->DNS Records->DNS-RR, clicked Add, entered (field-by-field) the Resource Record Name, Resource Record Number, and the (optional) data value.  

For the Resource Record Name, I used the name of a system which already has a DNS record for its IP address (Hostnames tab, next to DNS-RR).   

For the Resource Record Number, which I assume (possibly incorrectly) is the Resource Type, I first tried 16 (TXT type), and then 13 (HINFO).  I understand from various docs on the web that these Resource Types have been co-opted for other purposes than their original intentions, but for my own local use here, they would work (and I could always try to find a type that is not and never has been in use).  

At any rate, LuCI did not have an error, and even running "service dnsmasq restart" from the command line did not error, either.

The entry in the /etc/config/dhcp file shows:
```
config dnsrr
	option dnsrr 'test-chimaera'
	option rrnumber '16'
	option hexdata '2f66'
```
Again, "test-chimaera" is the name of a valid DNS Hostnames entry.

### Actual behaviour

LuCI does not give any indication of a problem when I create the record and save it.  And launching dnsmasq from the command line does not, either.  Further more, the following commands only emit the (correct) IP address, but not the txt resource record:
```
nslookup -type txt test-chimaera
host -t txt test-chimaera
```

### Expected behaviour

I expected to see the resource records in the output of either "nslookup" or "host" along with the IP address and/or other information (depending on the options given to those programs).

### Additional info

Having spent a few hours researching why the resource record does not seem to be invoked by dnsmasq.  I examined the contents of the generated config file and the arguments passed to dnsmasq (by "ps www" |grep dnsmasq) and I do not see any sign that the resource record is acknowledged.

I added some debug statements to the dnsmasq init script and its overlays to see what it was doing.  I note that when the script tries to do a "config_get rrname," it fails (not set, apparently).  The git notes (https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=7c501e0ed2e891069b9fa0eb35a378e2ac330a44) state "dnsmasq dns-rr are not affected by filter-rr" but I wonder if that might be the problem; I haven't looked into this any further.

It is also possible I have done something incorrectly elsewhere in my configuration.  I can forward the entire config directory upon request.  However, please note that, since adding the RR, I have not experienced any problems with the openwrt router.

I hope you will consider this for the upcoming release, which I understand to be 24.10.1.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

Habbie, 2025-04-08:
> option hexdata '2f66'

what is the goal of this data?

dave14305, 2025-04-08:
I can make it work simply by appending a dot to the unqualified short name.
```
config dnsrr
        option rrname 'test-chimaera'
        option rrnumber '16'
        option hexdata '2f66' 

root@router:~# nslookup -type=txt test-chimaera.
Server:         127.0.0.1
Address:        127.0.0.1:53

test-chimaera   text = "f66"
```

systemdlete, 2025-04-08:
@Habbie The data field is optional, per the spec, so it should not matter what the content is, so long as it is hex.

@dave14305 Thank you for your response.
I've tried appending the dot also, and tried multiple variants of the option (-type/-query).  
What version of openwrt are you using, and is it a repo version or your own?
Also, I am curious:  In the output from nslookup, it shows only the last 3 hex digits, whereas the configuration has 4 hex digits?  Did you copy-paste the output, or tried to faithfully copy it manually?

dave14305, 2025-04-08:
I’m on a slightly custom 24.10 SNAPSHOT build (just an extra WiFi mac80211 patch added). I do notice that the first character is missing. And I also noted that dig command wouldn’t return the same record, plus it complained of an extra 20 bytes on the  message. My output above is a copy/paste from the terminal. That’s what it looked like.

I also ran tcpdump to see if it was an nslookup issue or not, and you can assume dnsmasq is chomping the first byte.
```
config dnsrr
        option rrname 'test-dave14305'
        option rrnumber '16'
        option hexdata '646176653134333035'

root@router:~# nslookup -type=txt test-dave14305.
Server:         127.0.0.1
Address:        127.0.0.1:53

test-dave14305  text = "ave14305"

root@router:~# tcpdump -nvpi lo udp and port 53
tcpdump: listening on lo, link-type EN10MB (Ethernet), snapshot length 262144 bytes
18:48:56.213030 IP (tos 0x0, ttl 64, id 13750, offset 0, flags [DF], proto UDP (17), length 60)
    127.0.0.1.51998 > 127.0.0.1.53: 37485+ TXT? test-dave14305. (32)
18:48:56.213199 IP (tos 0x0, ttl 64, id 61338, offset 0, flags [DF], proto UDP (17), length 81)
    127.0.0.1.53 > 127.0.0.1.51998: 37485* 1/0/0 test-dave14305. TXT "ave14305 [|domain]

root@router:~# dig test-dave14305. -t txt
;; Warning: Message parser reports malformed message packet.

; <<>> DiG 9.20.4 <<>> test-dave14305. -t txt
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 2194
;; flags: qr aa rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1
;; WARNING: Message has 20 extra bytes at end

;; QUESTION SECTION:
;test-dave14305.                        IN      TXT

;; Query time: 0 msec
;; SERVER: 127.0.0.1#53(127.0.0.1) (UDP)
;; WHEN: Tue Apr 08 18:52:29 EDT 2025
;; MSG SIZE  rcvd: 64

root@router:~# tcpdump -nvpi lo udp and port 53
tcpdump: listening on lo, link-type EN10MB (Ethernet), snapshot length 262144 bytes
18:53:02.150657 IP (tos 0x0, ttl 64, id 9689, offset 0, flags [none], proto UDP (17), length 83)
    127.0.0.1.57895 > 127.0.0.1.53: 36293+ [1au] TXT? test-dave14305. (55)
18:53:02.150843 IP (tos 0x0, ttl 64, id 64799, offset 0, flags [DF], proto UDP (17), length 92)
    127.0.0.1.53 > 127.0.0.1.57895: 36293* 1/0/1 test-dave14305. TXT "ave14305^@^@)^DM-P^@^@^@^@^@^@ [|domain]

root@router:~# ubus call system board
{
        "kernel": "6.6.85",
        "hostname": "router",
        "system": "ARMv8 Processor rev 4",
        "model": "GL.iNet GL-MT6000",
        "board_name": "glinet,gl-mt6000",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10-SNAPSHOT",
                "revision": "r27986+595-fc236f3b3c",
                "target": "mediatek/filogic",
                "description": "OpenWrt 24.10-SNAPSHOT r27986+595-fc236f3b3c",
                "builddate": "1743877101"
        }
}
```

Habbie, 2025-04-09:
> so it should not matter what the content is

it does matter what the content is - clients need to be able to parse it.

> option hexdata '646176653134333035'

can you try `option hexdata '09646176653134333035'`?

dave14305, 2025-04-09:
> can you try `option hexdata '09646176653134333035'`?

That works better for both nslookup and dig. But the LuCI interface would make it difficult to prepend an entry with a tab character, since it encodes the string to hex for you.

Habbie, 2025-04-09:
> I see the 09 is the length of the string.

came here to mention that! The longer the string, the easier to type in LuCi then ;)

dave14305, 2025-04-09:
> The longer the string, the easier to type in LuCi then ;)

dnsmasq already natively supports a txt-record option (OpenWrt does not expose it), so perhaps it’s a bad example for dns-rr.

systemdlete, 2025-04-10:
I tried using the most recent version of dnsmasq.init from the github repo, but it still doesn't work.  Either the storage or the retrieval of the resource records is failing.  I might look into this deeper a bit later.

The fact that neither of you appear to have this issue might be due to different hardware or some other environmental difference.   @dave14305 is apparently using a more recent openwrt with some changes, maybe local to their own sources, so that might not be a good comparison.  

I wish that someone would try to reproduce the problem in a similar environment using openwrt 24.10.0 from the repos with as few changes to that configuration as possible.

Habbie, 2025-04-10:
> The fact that neither of you appear to have this issue

I need to clarify that I did not actually try anything. I just noticed the missing length prefix on the hex data.

systemdlete, 2025-04-10:
> > The fact that neither of you appear to have this issue
> 
> I need to clarify that I did not actually try anything. I just noticed the missing length prefix on the hex data.

It's still appreciated.  I added the length byte by editing the dhcp config file directly, of course.   Obviously, something needs to be done about the LuCI interface, but for the moment, I think figuring out why the whole mechanism isn't working is the main issue.

systemdlete, 2025-04-12:
After many hours of hacking the init code for dnsmasq, I have discovered the source of the problem:  The dnsmasq script wants to get a value for a field named "rrname" which does not exist, at least not from the LuCI-generated record for the DNS-RR record.  This points to another issue with LuCI, along with the length byte problem.

(Sidenote:  None of us noticed that @dave14305 had thet field name rrname in their dhcp config file whereas mine was dnsrr, generated by LuCI, not me.  That would have spared much of this grief.   I just assumed that their config file was more or less the same as mine.)

After modifying my dhcp config file, I am seeing evidence of the RR in the data.  However, I am still having issues with nslookup, host, and dig:  none of them are showing the txt record.   I will post back with additional info.

systemdlete, 2025-04-12:
So there are 2 problems, both stemming from the LuCI interface which generated a DNS-RR entry that made all of the noise in this report from the beginning:

1.  The field name is "rrname," NOT "dnsrr" as generated by the LuCI interface.
2.  The LuCI is not handling the length byte properly.

For the time, I can manually add these entries.  I think this bug warrants inclusion in the upcoming update to prevent other DNS neophytes from stumbling over this issue.

dave14305, 2025-04-12:
Luci fix is already in 24.10, and you might see it in an opkg update for luci-mod-network.
https://github.com/openwrt/luci/commit/0c6f5468f9e647753a41e1dbe294eb249b9df980

LuCI makes no attempt to calculate the length. Do all DNS RRs require one?

systemdlete, 2025-04-13:
> Luci fix is already in 24.10, and you might see it in an opkg update for luci-mod-network. [openwrt/luci@0c6f546](https://github.com/openwrt/luci/commit/0c6f5468f9e647753a41e1dbe294eb249b9df980)
> 
> LuCI makes no attempt to calculate the length. Do all DNS RRs require one?

Yes, the spec (RFC1035, Sec 3.3) requires that ALL DNS RRs require the first byte of a <character string> be a byte containing the length of the said string:
```
[3.3](https://www.rfc-editor.org/rfc/rfc1035#section-3.3). Standard RRs

The following RR definitions are expected to occur, at least
potentially, in all classes.  In particular, NS, SOA, CNAME, and PTR
will be used in all classes, and have the same format in all classes.
Because their RDATA format is known, all domain names in the RDATA
section of these RRs may be compressed.

<domain-name> is a domain name represented as a series of labels, and
terminated by a label with zero length.  <character-string> is a single
length octet followed by that number of characters.  <character-string>
is treated as binary information, and can be up to 256 characters in
length (including the length octet).
```

Which fix is in the (upcoming branch?) 24.10?   The correction for the field name or something else?

systemdlete, 2025-04-18:
I just installed 24.10.1 and now LuCI is generating a proper record in the /etc/config/dhcp file.  However, it is NOT generating and saving the length byte for the hex string in the file, and so, as before, calls to retrieve the RR TXT record fail.  host(1) complains about a bad packet.  I can see in its data dump that there really is a length byte before my string, yet it fails.  So if my string is entered in LuCI as "msg=hello", I see something like

```096d73673d68656c6c6f```

I added the length byte manually by editing the file and now the TXT records do show up in dig, host, nslookup output.  The only issue is that by examining the actual response in a tool like wireshark, I can see that the length byte I added is preceded by yet another length byte.  Now I see:

```0a096d73673d68656c6c6f```

This double length byte business might be an issue with dnsmasq itself, Idk.  But LuCI is still not calculating the length properly.   It means a user still has to do manual edits to the file.

twisteroidambassador, 2025-04-24:
> Yes, the spec (RFC1035, Sec 3.3) requires that Standard DNS RRs require the first byte of a be a byte containing the length of the said string:
> 
> {EDIT: Not literally ALL; actually it is the standard RRs. Some of the others, like the machine type, etc, are a different format. However, from what I've read--and one would suspect--is that those are not used much anymore and it is unlikely that sysadmins are maintaining them anyway. One workaround is, obviously, to only permit certain RR types.]

This is unfortunately incomplete, as RFC1035 is a really old document, and DNS is an ever-evolving standard. Many modern, widely-in-use rdatatypes are defined with hex data structures and not a length byte + text string, such as the HTTPS record, where the wire format for `1 . alpn="h3,h2"` is hex `00010000010006026833026832`.

This, in turn, means that to be truly useful, Luci not only must not add a length byte to data entered by uses, but in fact must be able to accept hex user input directly somehow.

Habbie, 2025-04-24:
> the spec (RFC1035, Sec 3.3) requires that Standard DNS RRs require the first byte of a be a byte containing the length of the said string

besides the note above, about a lot that has happened after 1035, the claim is not true for 1035 either. TXT requires a length prefix *per string in the TXT*, not just one at the start. Other types have their own requirements, even in 1035.