openwrt/openwrt #18442: Unhandled kernel unaligned access on Asus WL-500GP-V1

p-m, 2025-04-08:
### Describe the bug

Hi,
I'm using the latest git version from the openwrt-23.05 branch on my Asus WL-500gP-V1. Unfortunately there is a kernel panic after some hours of uptime, especially when using the WLAN.

I know, that you don’t support such old devices, but I’ve really good experience with openwrt-23.05 on other WL-500gP-V1 without WLAN, so it would be nice to fix this bug to be able to use openwrt-23.05 with WLAN.

It’s not a memory problem: when the kernel panics, there are 9MB of free RAM.

TIA for any hints, Peter

### OpenWrt version

r24167-dec59db8fb

### OpenWrt release

23.05-SNAPSHOT

### OpenWrt target/subtarget

bcm47xx/legacy

### Device

Asus WL500GP V1

### Image kind

Self-built image

### Actual behaviour

kernel panic after a while

### Expected behaviour

no kernel panic

### Additional info

Console output:

```text
[11871.161896] Unhandled kernel unaligned access[#1]:
[11871.166833] CPU: 0 PID: 1213 Comm: irq/6-b43 Not tainted 5.15.173 #0
[11871.173297] $ 0   : 00000000 1000dc00 817bba40 81412118
[11871.178715] $ 4   : d59a8059 80bd9ce8 80bd9ce8 00000000
[11871.184126] $ 8   : 80bd9ce8 801f7fbc 00000000 00000001
[11871.189537] $12   : 00000008 00000018 000000fc 00000003
[11871.194947] $16   : 80b460c0 81410d80 00000001 00000000
[11871.200359] $20   : 817bb426 00000000 00000000 000066fb
[11871.205769] $24   : 00000004 801ca97c                  
[11871.211181] $28   : 80bd8000 80bd9cb0 40000210 802a2214
[11871.216600] Hi    : 00000000
[11871.219532] Lo    : 00000024
[11871.222465] epc   : 802a2210 kfree_skb_list+0x20/0x3c
[11871.227668] ra    : 802a2214 kfree_skb_list+0x24/0x3c
[11871.232844] Status: 1000dc03 KERNEL EXL IE 
[11871.237162] Cause : 80800010 (ExcCode 04)
[11871.241237] BadVA : d59a8059
[11871.244169] PrId  : 00029006 (Broadcom BMIPS3300)
[11871.248944] Modules linked in: b43 mac80211 cfg80211 libcrc32c cordic compat sha512_generic seqiv jitterentropy_rng drbg hmac cmac ohci_platform ohci_hcd fsc
[11871.273607] Process irq/6-b43 (pid: 1213, threadinfo=b45a9cee, task=6fc8bec4, tls=00000000)
[11871.282100] Stack : 80b460c0 81410d80 00000001 00000000 80b460c0 8150264c 8167ff00 00000004
[11871.290765]         1000dc01 80039fbc 80bd9d58 00000000 806092e0 80b46240 00000032 800f6a90
[11871.299421]         00000000 80043768 8143c840 805f0000 8143c840 80b46240 80bd9da0 8168c000
[11871.308079]         81410d80 80bd9da0 00000000 80b460c0 00000001 00000000 80b460d8 000066fb
[11871.316738]         40000210 81503c78 81410d80 817bb426 00000002 00000000 00000000 00000000
[11871.325394]         ...
[11871.327928] Call Trace:
[11871.330420] [<802a2210>] kfree_skb_list+0x20/0x3c
[11871.335269] [<802a2214>] kfree_skb_list+0x24/0x3c
[11871.340114] 
[11871.341655] Code: afbf0014  afb00010  0c0a887a <8c900000> 1600fffd  02002025  8fbf0014  8fb00010  03e00008 
[11871.351768] 
[11871.353488] ---[ end trace b5af45d64a491a74 ]---
[11871.358311] Kernel panic - not syncing: Fatal exception
[11871.363643] Rebooting in 3 seconds..
```

### Diffconfig

```text
CONFIG_TARGET_bcm47xx=y
CONFIG_TARGET_bcm47xx_legacy=y
CONFIG_TARGET_bcm47xx_legacy_DEVICE_asus_wl-500gp-v1=y
CONFIG_BUSYBOX_CUSTOM=y
# CONFIG_BUSYBOX_CONFIG_FEATURE_IPV6 is not set
# CONFIG_BUSYBOX_CONFIG_FEATURE_PASSWD_WEAK_CHECK is not set
# CONFIG_BUSYBOX_CONFIG_INCLUDE_SUSv2 is not set
# CONFIG_BUSYBOX_CONFIG_ZCAT is not set
# CONFIG_KERNEL_IPV6_MROUTE is not set
# CONFIG_LUCI_CSSTIDY is not set
# CONFIG_LUCI_JSMIN is not set
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_SMALL_FOOTPRINT=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
# CONFIG_PACKAGE_dnsmasq is not set
# CONFIG_PACKAGE_dropbear is not set
# CONFIG_PACKAGE_firewall4 is not set
# CONFIG_PACKAGE_jansson is not set
# CONFIG_PACKAGE_kmod-leds-gpio is not set
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
# CONFIG_PACKAGE_kmod-nf-conntrack is not set
# CONFIG_PACKAGE_kmod-nf-conntrack6 is not set
# CONFIG_PACKAGE_kmod-nf-flow is not set
# CONFIG_PACKAGE_kmod-nf-log is not set
# CONFIG_PACKAGE_kmod-nf-log6 is not set
# CONFIG_PACKAGE_kmod-nf-nat is not set
# CONFIG_PACKAGE_kmod-nf-reject is not set
# CONFIG_PACKAGE_kmod-nf-reject6 is not set
# CONFIG_PACKAGE_kmod-nfnetlink is not set
# CONFIG_PACKAGE_kmod-nft-core is not set
# CONFIG_PACKAGE_kmod-nft-fib is not set
# CONFIG_PACKAGE_kmod-nft-nat is not set
# CONFIG_PACKAGE_kmod-nft-offload is not set
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_libatomic=y
# CONFIG_PACKAGE_libmnl is not set
CONFIG_PACKAGE_libncurses=y
# CONFIG_PACKAGE_libnftnl is not set
CONFIG_PACKAGE_libopenssl=y
# CONFIG_PACKAGE_libpthread is not set
# CONFIG_PACKAGE_libustream-mbedtls is not set
# CONFIG_PACKAGE_nftables-json is not set
# CONFIG_PACKAGE_odhcp6c is not set
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
CONFIG_PACKAGE_openssh-client=y
CONFIG_PACKAGE_openssh-keygen=y
CONFIG_PACKAGE_openssh-server=y
CONFIG_PACKAGE_openssh-sftp-server=y
# CONFIG_PACKAGE_opkg is not set
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_screen=y
CONFIG_PACKAGE_terminfo=y
# CONFIG_PACKAGE_ucode-mod-uci is not set
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-mini=y
CONFIG_PACKAGE_zlib=y
CONFIG_MBEDTLS_AES_C=y
# CONFIG_MBEDTLS_CAMELLIA_C is not set
# CONFIG_MBEDTLS_CCM_C is not set
# CONFIG_MBEDTLS_CERTS_C is not set
# CONFIG_MBEDTLS_CIPHER_MODE_OFB is not set
# CONFIG_MBEDTLS_CIPHER_MODE_XTS is not set
CONFIG_MBEDTLS_CMAC_C=y
# CONFIG_MBEDTLS_DEBUG_C is not set
CONFIG_MBEDTLS_DES_C=y
# CONFIG_MBEDTLS_ECP_DP_BP256R1_ENABLED is not set
# CONFIG_MBEDTLS_ECP_DP_BP384R1_ENABLED is not set
# CONFIG_MBEDTLS_ECP_DP_BP512R1_ENABLED is not set
CONFIG_MBEDTLS_ECP_DP_CURVE25519_ENABLED=y
# CONFIG_MBEDTLS_ECP_DP_CURVE448_ENABLED is not set
# CONFIG_MBEDTLS_ECP_DP_SECP192K1_ENABLED is not set
# CONFIG_MBEDTLS_ECP_DP_SECP192R1_ENABLED is not set
# CONFIG_MBEDTLS_ECP_DP_SECP224K1_ENABLED is not set
# CONFIG_MBEDTLS_ECP_DP_SECP224R1_ENABLED is not set
CONFIG_MBEDTLS_ECP_DP_SECP256K1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP256R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP384R1_ENABLED=y
CONFIG_MBEDTLS_ECP_DP_SECP521R1_ENABLED=y
CONFIG_MBEDTLS_ENTROPY_FORCE_SHA256=y
CONFIG_MBEDTLS_GCM_C=y
# CONFIG_MBEDTLS_HKDF_C is not set
# CONFIG_MBEDTLS_KEY_EXCHANGE_DHE_PSK_ENABLED is not set
# CONFIG_MBEDTLS_KEY_EXCHANGE_DHE_RSA_ENABLED is not set
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_PSK_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_RSA_ENABLED=y
# CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_ECDSA_ENABLED is not set
# CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_RSA_ENABLED is not set
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK_ENABLED=y
# CONFIG_MBEDTLS_KEY_EXCHANGE_RSA_ENABLED is not set
# CONFIG_MBEDTLS_KEY_EXCHANGE_RSA_PSK_ENABLED is not set
CONFIG_MBEDTLS_NIST_KW_C=y
# CONFIG_MBEDTLS_PLATFORM_C is not set
# CONFIG_MBEDTLS_RIPEMD160_C is not set
CONFIG_MBEDTLS_RSA_NO_CRT=y
# CONFIG_MBEDTLS_SELF_TEST is not set
# CONFIG_MBEDTLS_SSL_RENEGOTIATION is not set
# CONFIG_MBEDTLS_SSL_TRUNCATED_HMAC is not set
# CONFIG_MBEDTLS_VERSION_C is not set
# CONFIG_MBEDTLS_VERSION_FEATURES is not set
# CONFIG_MBEDTLS_XTEA_C is not set
# CONFIG_OPENSSH_LIBFIDO2 is not set
CONFIG_PACKAGE_kmod-crypto-crc32c=y
CONFIG_PACKAGE_kmod-lib-crc32c=y
# CONFIG_PACKAGE_libcbor is not set
# CONFIG_PACKAGE_libevdev is not set
# CONFIG_PACKAGE_libfido2 is not set
CONFIG_PACKAGE_libmbedtls=y
# CONFIG_PACKAGE_libudev-zero is not set
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2025-04-08:
Duplicate of
https://github.com/openwrt/openwrt/issues/18417
First explore compiler options specific to architecture. Is there any 16/64 or 8/64 device using same SoC?

p-m, 2025-04-08:
> Duplicate of [#18417](https://github.com/openwrt/openwrt/issues/18417)

Why please?

> First explore compiler options specific to architecture.

Where and how please?

> Is there any 16/64 or 8/64 device using same SoC?

I don’t know.

brada4, 2025-04-08:
1. sigbus is valid instruction with wrong parameters like unaligned access or pointer where only register is allowed.
2. `info gcc`
3.  Thus no visible bug in codebase.

p-m, 2025-04-08:
Ok, then it’s too difficult for me and I cannot help here.

But I can test patches of course!

brada4, 2025-04-08:
Did i mention to seek archaeological excavation help in forum? The device id desupporyed 5 years ago, and no wonder after 5 years of no testing something fails.

Neustradamus, 2025-04-08:
To follow this ticket.

brada4, 2025-04-09:
Nobofy cares about sub-par platforms.

p-m, 2025-04-09:
> Nobofy cares about sub-par platforms.

Please don’t speak for others. I understand, that *you* don’t care.
But there are people, who care. Example: https://github.com/openwrt/openwrt/issues/13789

brada4, 2025-04-09:
Device is desupported since? Get over it.