openwrt/openwrt #18444: dir-825 b1 LAN ports reversed

brianjmurrell, 2025-04-08:
### Describe the bug

On my DIR-825 rev. B1, I have the port labeled #4 on the router plugged in (i.e. with link) and configured to (i.e. VLAN) tag packets.  No other ports are plugged in:

```
# swconfig dev switch0 show
Global attributes:
	enable_learning: 1
	enable_vlan: 1
	enable_vlan4k: 0
	blinkrate: 0
	max_length: 1
	green_mode: 0
Port 0:
	mib: Port 0 MIB counters
IfInOctets                          : 782650
EtherStatsOctets                    : 782650
EtherStatsUnderSizePkts             : 0
EtherFragments                      : 0
EtherStatsPkts64Octets              : 3
EtherStatsPkts65to127Octets         : 3765
EtherStatsPkts128to255Octets        : 420
EtherStatsPkts256to511Octets        : 635
EtherStatsPkts512to1023Octets       : 178
EtherStatsPkts1024to1518Octets      : 16
EtherOversizeStats                  : 0
EtherStatsJabbers                   : 0
IfInUcastPkts                       : 1262
EtherStatsMulticastPkts             : 2416
EtherStatsBroadcastPkts             : 1339
EtherStatsDropEvents                : 0
Dot3StatsFCSErrors                  : 0
Dot3StatsSymbolErrors               : 0
Dot3InPauseFrames                   : 0
Dot3ControlInUnknownOpcodes         : 0
IfOutOctets                         : 672786
Dot3StatsSingleCollisionFrames      : 0
Dot3StatMultipleCollisionFrames     : 0
Dot3sDeferredTransmissions          : 147
Dot3StatsLateCollisions             : 0
EtherStatsCollisions                : 0
Dot3StatsExcessiveCollisions        : 0
Dot3OutPauseFrames                  : 0
Dot1dBasePortDelayExceededDiscards  : 0
Dot1dTpPortInDiscards               : 0
IfOutUcastPkts                      : 1319
IfOutMulticastPkts                  : 22
IfOutBroadcastPkts                  : 2

	led: 2
	green_port: 0
	pvid: 1
	link: port:0 link:up speed:100baseT full-duplex txflow rxflow 
Port 1:
	mib: Port 1 MIB counters
IfInOctets                          : 0
EtherStatsOctets                    : 0
EtherStatsUnderSizePkts             : 0
EtherFragments                      : 0
EtherStatsPkts64Octets              : 0
EtherStatsPkts65to127Octets         : 0
EtherStatsPkts128to255Octets        : 0
EtherStatsPkts256to511Octets        : 0
EtherStatsPkts512to1023Octets       : 0
EtherStatsPkts1024to1518Octets      : 0
EtherOversizeStats                  : 0
EtherStatsJabbers                   : 0
IfInUcastPkts                       : 0
EtherStatsMulticastPkts             : 0
EtherStatsBroadcastPkts             : 0
[... 205 lines cut ...]
### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ath79/generic

### Device

D-Link DIR825B1

### Image kind

Self-built image

### Diffconfig

```text
Mine is an ImageBuilder built image, not built from source.
```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2025-04-08:
Edit file etc/board.d/02_network line 243

`			"0:lan" "1:lan" "2:lan" "3:lan" "5@eth0"`
to
`			"0:lan:4" "1:lan:3" "2:lan:2" "3:lan:1" "5@eth0"`

Reboot and test

brianjmurrell, 2025-04-08:
Did not seem to fix it:

[image]

[I do have a couple more devices plugged into it now, obviously]

```
# sed -ne '/dlink,dir-825-b1/,/;;/p'  /etc/board.d/02_network
	dlink,dir-825-b1|\
	trendnet,tew-673gru)
		ucidef_set_interface_wan "eth1"
		ucidef_add_switch "switch0" \
			"0:lan:4" "1:lan:3" "2:lan:2" "3:lan:1" "5@eth0"
		;;
```

brianjmurrell, 2025-04-09:
I'm afraid still not correct:

[image]

```
# sed -ne '/dlink,dir-825-b1/,/;;/p'  /etc/board.d/02_network
	dlink,dir-825-b1|\
	trendnet,tew-673gru)
		ucidef_set_interface_wan "eth1"
		ucidef_add_switch "switch0" \
			"3:lan" "2:lan" "1:lan" "0:lan" "5@eth0"
		;;
```

brianjmurrell, 2025-04-16:
Anything more to try here?

S-h-a-d-e-r, 2025-09-13:
Tryed to remove some ports - "5@eth0" "3:lan:1" "0:lan:4", and "0:lan" "3:lan" "5@eth0".

Tryed to shorten "ath79_setup_interfaces()" in "02_network" - without using "case" construction, only the required "ucidef_set_interface_wan" and "ucidef_add_switch" commands with arguments.

Nothing changes. Like it uses some default config, or is configured elsewhere.