openwrt/openwrt #18446: The MediaTek MT7986 network card driver fails to start the network due to a failure in allocating 4MB of contiguous DMA memory during initialization.

ENJINEIN, 2025-04-09:
`In` the OpenWRT23.05.1 system (kernel 5.15.134), the MT7986 wireless card driver experiences a failure in allocating 4MB of continuous DMA memory for netifd. The reproduction scenario is still unknown.
The complete error message is as follows：
`pr  1 22:15:11 1024 netifd: You have delegated IPv6-prefixes but haven't assigned them to any interface. Did you forget to set option ip6assign on your lan-interfaces?                                                                                        
Apr  2 06:15:11 1024 kernel: [3410198.733855] netifd: page allocation failure: order:10, mode:0x40dc0(GFP_KERNEL|__GFP_COMP|__GFP_ZERO), nodemask=(null),cpuset=/,mems_allowed=0                                                                                
Apr  2 06:15:11 1024 kernel: [3410198.746878] CPU: 2 PID: 15087 Comm: netifd Not tainted 5.15.134 #0                                                                                                                                                            
Apr  2 06:15:11 1024 kernel: [3410198.753232] Hardware name: UniElec u7986-06 (DT)                                                                                                                                                                              
Apr  2 06:15:11 1024 kernel: [3410198.758009] Call trace:                                                                                                                                                                                                       
Apr  2 06:15:11 1024 kernel: [3410198.760618]  dump_backtrace+0x0/0x170                                                                                                                                                                                         
Apr  2 06:15:11 1024 kernel: [3410198.764450]  show_stack+0x14/0x20                                                                                                                                                                                             
Apr  2 06:15:11 1024 kernel: [3410198.767925]  dump_stack_lvl+0x60/0x78                                                                                                                                                                                         
Apr  2 06:15:11 1024 kernel: [3410198.771749]  dump_stack+0x14/0x2c                                                                                                                                                                                             
Apr  2 06:15:11 1024 kernel: [3410198.775222]  warn_alloc+0xd8/0x16c                                                                                                                                                                                            
Apr  2 06:15:11 1024 kernel: [3410198.778784]  __alloc_pages_slowpath.constprop.0+0xaf0/0xb00                                                                                                                                                                   
Apr  2 06:15:11 1024 kernel: [3410198.784514]  __alloc_pages+0x24c/0x2b0                                                                                                                                                                                        
Apr  2 06:15:11 1024 kernel: [3410198.788420]  kmalloc_order+0x30/0xb4                                                                                                                                                                                          
Apr  2 06:15:11 1024 kernel: [3410198.792155]  kmalloc_order_trace+0x28/0xfc                                                                                                                                                                                    
Apr  2 06:15:11 1024 kernel: [3410198.796408]  mtk_open+0x57c/0x9bc                                                                                                                                                                                             
Apr  2 06:15:11 1024 kernel: [3410198.799883]  __dev_open+0x104/0x1a0                                                                                                                                                                                           
Apr  2 06:15:11 1024 kernel: [3410198.803532]  __dev_change_flags+0x138/0x190                                                                                                                                                                                   
Apr  2 06:15:11 1024 kernel: [3410198.807872]  dev_change_flags+0x20/0x60                                                                                                                                                                                       
Apr  2 06:15:11 1024 kernel: [3410198.811867]  dev_ifsioc+0x3ec/0x43c                                                                                                                                                                                           
Apr  2 06:15:11 1024 kernel: [3410198.815516]  dev_ioctl+0x144/0x4c0                                                                                                                                                                                            
Apr  2 06:15:11 1024 kernel: [3410198.819076]  sock_ioctl+0x3ac/0x490                                                                                                                                                                                           
Apr  2 06:15:11 1024 kernel: [3410198.822725]  __arm64_sys_ioctl+0x588/0x106c                                                                                                                                                                                   
Apr  2 06:15:11 1024 kernel: [3410198.827068]  invoke_syscall.constprop.0+0x4c/0xe0                                                                                                                                                                             
Apr  2 06:15:11 1024 kernel: [3410198.831930]  do_el0_svc+0x40/0xd0                                                                                                                                                                                             
Apr  2 06:15:11 1024 kernel: [3410198.835403]  el0_svc+0x14/0x50                                                                                                                                                                                                
Apr  2 06:15:11 1024 kernel: [3410198.838617]  el0t_64_sync_handler+0xe0/0x110                                                                                                                                                                                  
Apr  2 06:15:11 1024 kernel: [3410198.843044]  el0t_64_sync+0x15c/0x160                                                                                                                                                                                         
Apr  2 06:15:11 1024 kernel: [3410198.846894] Mem-Info:                                                                                                                                                                                                         
Apr  2 06:15:11 1024 kernel: [3410198.849350] active_anon:25043 inactive_anon:27682 isolated_anon:0                                                                                                                                                             
Apr  2 06:15:11 1024 kernel: [3410198.849350]  active_file:7965 inactive_file:117043 isolated_file:0                                                                                                                                                            
Apr  2 06:15:11 1024 kernel: [3410198.849350]  unevictable:0 dirty:17 writeback:0                                                                                                                                                                               
Apr  2 06:15:11 1024 kernel: [3410198.849350]  slab_reclaimable:5882 slab_unreclaimable:10175                                                                                                                                                                   
[... cut ...]
in the openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.134/drivers/net/ethernet/mediatek/mtk_eth_soc.c : mtk_init_fq_dma, i found
`	eth->scratch_head = kcalloc(cnt, MTK_QDMA_PAGE_SIZE, GFP_KERNEL);
`
cnt=MTK_QDMA_RING_SIZE=2048 MTK_QDMA_PAGE_SIZE=2048, so kcalloc will calloc 4M.  Is this related to that?

and i also found a patch :  https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/net/ethernet/mediatek?id=c57e558194430d10d5e5f4acd8a8655b68dade13    Is this related to that?

ENJINEIN, 2025-04-09:
pr  1 22:15:11 1024 netifd: You have delegated IPv6-prefixes but haven't assigned them to any interface. Did you forget to set option ip6assign on your lan-interfaces?                                                                                        
Apr  2 06:15:11 1024 kernel: [3410198.733855] netifd: page allocation failure: order:10, mode:0x40dc0(GFP_KERNEL|__GFP_COMP|__GFP_ZERO), nodemask=(null),cpuset=/,mems_allowed=0                                                                                
Apr  2 06:15:11 1024 kernel: [3410198.746878] CPU: 2 PID: 15087 Comm: netifd Not tainted 5.15.134 #0                                                                                                                                                            
Apr  2 06:15:11 1024 kernel: [3410198.753232] Hardware name: UniElec u7986-06 (DT)                                                                                                                                                                              
Apr  2 06:15:11 1024 kernel: [3410198.758009] Call trace:                                                                                                                                                                                                       
Apr  2 06:15:11 1024 kernel: [3410198.760618]  dump_backtrace+0x0/0x170                                                                                                                                                                                         
Apr  2 06:15:11 1024 kernel: [3410198.764450]  show_stack+0x14/0x20                                                                                                                                                                                             
Apr  2 06:15:11 1024 kernel: [3410198.767925]  dump_stack_lvl+0x60/0x78                                                                                                                                                                                         
Apr  2 06:15:11 1024 kernel: [3410198.771749]  dump_stack+0x14/0x2c                                                                                                                                                                                             
Apr  2 06:15:11 1024 kernel: [3410198.775222]  warn_alloc+0xd8/0x16c                                                                                                                                                                                            
Apr  2 06:15:11 1024 kernel: [3410198.778784]  __alloc_pages_slowpath.constprop.0+0xaf0/0xb00                                                                                                                                                                   
Apr  2 06:15:11 1024 kernel: [3410198.784514]  __alloc_pages+0x24c/0x2b0                                                                                                                                                                                        
Apr  2 06:15:11 1024 kernel: [3410198.788420]  kmalloc_order+0x30/0xb4                                                                                                                                                                                          
Apr  2 06:15:11 1024 kernel: [3410198.792155]  kmalloc_order_trace+0x28/0xfc                                                                                                                                                                                    
Apr  2 06:15:11 1024 kernel: [3410198.796408]  mtk_open+0x57c/0x9bc                                                                                                                                                                                             
Apr  2 06:15:11 1024 kernel: [3410198.799883]  __dev_open+0x104/0x1a0                                                                                                                                                                                           
Apr  2 06:15:11 1024 kernel: [3410198.803532]  __dev_change_flags+0x138/0x190                                                                                                                                                                                   
Apr  2 06:15:11 1024 kernel: [3410198.807872]  dev_change_flags+0x20/0x60                                                                                                                                                                                       
Apr  2 06:15:11 1024 kernel: [3410198.811867]  dev_ifsioc+0x3ec/0x43c                                                                                                                                                                                           
Apr  2 06:15:11 1024 kernel: [3410198.815516]  dev_ioctl+0x144/0x4c0                                                                                                                                                                                            
Apr  2 06:15:11 1024 kernel: [3410198.819076]  sock_ioctl+0x3ac/0x490                                                                                                                                                                                           
[... cut ...]
Apr  2 06:15:11 1024 kernel: [3410198.983063] 0 pages HighMem/MovableOnly                                                                                                                                                                                       
Apr  2 06:15:11 1024 kernel: [3410198.987068] 9663 pages reserved                                                                                                                                                                                               
Apr  2 06:15:11 1024 kernel: [3410198.999840] mt7530-mdio mdio-bus:1f lan2: failed to open master eth0

ENJINEIN, 2025-04-09:
I also tried to reproduce it, such as filling up the 4M continuous space, and then restarting the network, but the same situation did not occur again.  
```bash
root@1023:~# cat /proc/buddyinfo  
Node 0, zone      DMA    481    578    632    433    327    247    243    112     11      1      0
```

castiel652, 2025-04-10:
Have you tried upgrading to 23.05

castiel652, 2025-04-10:
> > Have you tried upgrading to 23.05
> 
> What do you mean? My current version is OpenWRT 23.05.1. Are you suggesting that I upgrade to version 23.05.5 or 24.10.0? However, due to some special reasons, I can only stay on the 23.05 version.

Yes sorry I meant 23.05.5

ENJINEIN, 2025-04-10:
> > > Have you tried upgrading to 23.05
> > 
> > 
> > What do you mean? My current version is OpenWRT 23.05.1. Are you suggesting that I upgrade to version 23.05.5 or 24.10.0? However, due to some special reasons, I can only stay on the 23.05 version.
> 
> Yes sorry I meant 23.05.5

Okay, thank you. But I still have some questions. Even if I upgrade to 23.05.5, I can't be sure whether it has fixed the issue, as I don't know how to reproduce the problem. Do you have any idea on how to reproduce this issue? Thanks again.

castiel652, 2025-04-10:
> Okay, thank you. But I still have some questions. Even if I upgrade to 23.05.5, I can't be sure whether it has fixed the issue, as I don't know how to reproduce the problem. Do you have any idea on [...]

Sorry I do not know since I don't have the same device and config.

Bartel-C8, 2025-08-20:
Dropping my bug report here. 
https://github.com/openwrt/openwrt/issues/19613

I found a way to reproduce it, thanks to https://github.com/openwrt/mt76/issues/592