openwrt/openwrt #18447: OpenWrt One Keeps crashing after enabling the 5G radio

10code, 2025-04-09:
### Describe the bug

Hello, I have a brand new OpenWrt One board and when I enable the 5G radio from the CLI with the following command it keeps rebooting.

uci set wireless.radio1.disabled='0'

I have attached the log for your reference. Please let me know what is going wrong here!

[wrtOnelog.txt](https://github.com/user-attachments/files/19672950/wrtOnelog.txt)

### OpenWrt version

r29161-f44984f19c

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

OpenWrt One

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

10code, 2025-04-10:
If there is a recommended power adaptor or rating details for openWRT one board, please share

DragonBluep, 2025-04-13:
There are some discussions. OpenWrt one requires 15V power supply.
https://forum.openwrt.org/t/openwrt-one-setup-install-and-discussion/211866/174