openwrt/openwrt #18473: [24.10] IPv6 router advertisements going out wrong interface

cranderson, 2025-04-13:
### Describe the bug

I have a dual-homed host `server` attached to both the WAN and LAN, in parallel with the Linksys WRT1900AC v1 running OpenWrt 24.10.0 also connected to the same WAN and LAN.  The ISP sends IPv6 RA and DHCPv6-PD.  OpenWrt's `wan` interface is able to request a /60 and delegate /64s to downstream hosts on the `br-lan`, `br-guest`, etc. interfaces without issues.  The `server` that is attached to both the ISP WAN and OpenWrt `br-lan` is able to use IPv6 assigned via the ISP's RA/SLAAC to communicate to the Internet directly, and the `server` is able to use IPv6 assigned via OpenWrt's RA/DHCPv6 to communicate to the LAN directly.  This broke since upgrading from 23.05.5 to 24.10.0.

Since 24.10, OpenWrt's `br-lan` IPv6 RA is being seen on devices attached to both the `br-lan` and `wan` interfaces.   The `server` is seeing OpenWrt's `br-lan` RA on its own WAN `enp3s0` interface as well as the expected LAN `enp4s0` interface, which breaks the `server`'s communication to the LAN since the `server` thinks the LAN's IPv6 prefixes are *on-link* on its WAN interface.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT1900AC v1

### Image kind

Official downloaded image

### Steps to reproduce

Here is a network diagram:

```
ISP modem/WAN switch
     |       |
    wan    enp3s0
 OpenWrt   server
  br-lan   enp4s0
     |       |
    LAN switch
          |
Other LAN clients
```

Here are some configuration details.  I verified that the OpenWrt configuration between 23.05.5 and 24.10.0 is identical.

```
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.dhcpv4='server'
dhcp.lan.dhcpv6='server'
dhcp.lan.ra='server'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='7'

network.@device[0]=device
network.@device[0].name='br-lan'
network.@device[0].type='bridge'
network.@device[0].ports='lan1' 'lan2' 'lan3' 'lan4'
network.@device[0].macaddr='96:XX:XX:XX:XX:XX'

network.@device[1]=device
[... 187 lines cut ...]
    192.168.1.1 > 224.0.0.1: ICMP echo request, id 12090, seq 1, length 64

00:04:38.182351 96:XX:XX:XX:XX:XX > ff:ff:ff:ff:ff:ff, ethertype IPv4 (0x0800), length 98: (tos 0x0, ttl 64, id 7938, offset 0, flags [DF], proto ICMP (1), length 84)
    192.168.1.1 > 255.255.255.255: ICMP echo request, id 13085, seq 0, length 64
00:04:39.182627 96:XX:XX:XX:XX:XX > ff:ff:ff:ff:ff:ff, ethertype IPv4 (0x0800), length 98: (tos 0x0, ttl 64, id 7970, offset 0, flags [DF], proto ICMP (1), length 84)
    192.168.1.1 > 255.255.255.255: ICMP echo request, id 13085, seq 1, length 64
```

### Actual behaviour

IPv4 broadcasts and IPv4/IPv6 link-local multicasts sent by OpenWrt on a specific interface such as IPv6 Router Advertisement messages are appearing on other interfaces than the interface they are sent on.  This is breaking IPv6 connectivity for hosts connected to the other interfaces.

### Expected behaviour

IPv4/IPv6 link-local multicasts sent by OpenWrt on a specific interface, especially IPv6 Router Advertisement messages, should not appear on other interfaces.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

tmcqueen-materials, 2025-04-13:
@cranderson : I am unable to reproduce this, either with ipv4 or ipv6 link local multicast, or ipv6 RA messages. It is also unusual in that single packets from the source (openwrt) would result in multiple packets going out on the wire.

What does your complete OpenWRT configuration look like? What do your route tables, nftables ruleset, and live bridge configs look like?

Is it possible there is some configuration on `server` that is causing packets to be forwarded across its two interfaces? What does its nftables ruleset, route tables, etc., look like?

If you do packet captures on both OpenWRT interfaces (lan and wan), what do you see? Does in fact one ping result in two packets, one going out on each interface?

tmcqueen-materials, 2025-04-13:
@cranderson : Few more things as well.

Do you have any multicast mirroring/routing software/rules on either OpenWRT or Server? Do you still see this effect when you physically disconnect server from the LAN interface (i.e. it can only see traffic on WAN)?

cranderson, 2025-04-13:
> [@cranderson](https://github.com/cranderson) : I am unable to reproduce this, either with ipv4 or ipv6 link local multicast, or ipv6 RA messages. It is also unusual in that single packets from the source (openwrt) would result in multiple packets going out on the wire.

My guess at this point is that it is a problem with DSA, where the physical hardware switch is the one sending out the packets on the wan interface when OpenWrt sends it to br-lan.
 
> What does your complete OpenWRT configuration look like? What do your route tables, nftables ruleset, and live bridge configs look like?

Complete (redacted) OpenWrt configuration:

```
BusyBox v1.36.1 (2025-02-03 23:09:37 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 24.10.0, r28427-6df0e3d02a
 -----------------------------------------------------
root@OpenWrt: ~# uci show
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].cachesize='1000'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.dhcpv4='server'
dhcp.lan.dhcpv6='server'
dhcp.lan.ra='server'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='7'
dhcp.@host[0]=host
dhcp.@host[0].name='host0'
dhcp.@host[0].ip='192.168.1.6'
dhcp.@host[0].mac='mac0'
dhcp.@host[0].dns='1'
dhcp.@host[1]=host
dhcp.@host[1].name='host1'
dhcp.@host[1].mac='mac1'
dhcp.@host[1].ip='192.168.1.10'
dhcp.@host[1].dns='1'
dhcp.@host[2]=host
dhcp.@host[2].name='host2'
dhcp.@host[2].mac='mac2'
dhcp.@host[2].ip='192.168.1.24'
dhcp.@host[2].dns='1'
dhcp.@host[3]=host
dhcp.@host[3].name='host3'
dhcp.@host[3].mac='mac3'
[... 878 lines cut ...]
Here are `server` routes with LAN `enp4s0` unplugged.  I also manually flushed all the `enp3s0` routes, but they came back again from the RA messages:

```
>ip -6 route 
20ZZ:ZZZZ:ZZZZ:ZZZZ:ZZZZ:ZZZZ:ZZZZ:ZZZZ dev enp3s0 proto kernel metric 100 pref medium
26XX:XXXX:XXXX:XXX1::/64 dev enp3s0 proto ra metric 100 pref medium
fdd1:XXXX:XXXX:1::/64 dev enp3s0 proto ra metric 100 pref medium
fe80::/64 dev enp3s0 proto kernel metric 1024 pref medium
default proto ra metric 20100 pref medium
	nexthop via fe80::YYYY:YYYY:6:1 dev enp3s0 weight 1 
	nexthop via fe80::YYYY:YYYY:2:1 dev enp3s0 weight 1 
	nexthop via fe80::YYYY:YYYY:1:1 dev enp3s0 weight 1
```

> If you do packet captures on both OpenWRT interfaces (lan and wan), what do you see? Does in fact one ping result in two packets, one going out on each interface?

I see one packet going out only on `br-lan` when capturing from OpenWrt itself.  I never see the ping or RA packets on `wan`.

> Do you have any multicast mirroring/routing software/rules on either OpenWRT or Server?

Nothing that I specifically configured.

> Do you still see this effect when you physically disconnect server from the LAN interface (i.e. it can only see traffic on WAN)?

Yes.

cranderson, 2025-04-13:
Here is a tcpdump of the DSA conduit interface `eth0` while sending a single ping to `ff02::2` on the `br-lan` interface:

```
root@OpenWrt:~# ping -I br-lan -6 ff02::2%br-lan -c1
PING ff02::2%br-lan (ff02::2%16): 56 data bytes
64 bytes from fe80::94XX:XXff:feXX:XXbd: seq=0 ttl=64 time=0.599 ms

--- ff02::2%br-lan ping statistics ---
1 packets transmitted, 1 packets received, 0% packet loss
round-trip min/avg/max = 0.599/0.599/0.599 ms
```

```
root@OpenWrt:/# tcpdump -i eth0 -n -s0 -v -e | grep  -F ' > 33:33:00:00:00:02'
tcpdump: listening on eth0, link-type DSA_TAG_EDSA (Marvell EDSA), snapshot length 262144 bytes
15:55:37.448825 96:XX:XX:XX:XX:bd > 33:33:00:00:00:02, Marvell EDSA ethertype 0xdada (Unknown), rsvd 0 0, mode Forward, dev 1, port 0, untagged, VID 4095, FPri 0, ethertype IPv6 (0x86dd), length 126: (flowlabel 0x2efc7, hlim 1, next-header ICMPv6 (58) payload length: 64) fe80::94XX:XXff:feXX:XXbd > ff02::2: [icmp6 sum ok] ICMP6, echo request, id 9990, seq 0
```

And the same while sending a single ping to `ff02::2` on the `wan` interface:

```
root@OpenWrt:~# ping -I wan -6 ff02::2%wan -c1
PING ff02::2%wan (ff02::2%8): 56 data bytes
64 bytes from fe80::96XX:XXff:feXX:XXbd: seq=0 ttl=64 time=0.422 ms

--- ff02::2%wan ping statistics ---
1 packets transmitted, 1 packets received, 0% packet loss
round-trip min/avg/max = 0.422/0.422/0.422 ms
```

```
root@OpenWrt:/# tcpdump -i eth0 -n -s0 -v -e | grep  -F ' > 33:33:00:00:00:02'
tcpdump: listening on eth0, link-type DSA_TAG_EDSA (Marvell EDSA), snapshot length 262144 bytes
15:53:05.623606 94:XX:XX:XX:XX:bd > 33:33:00:00:00:02, Marvell EDSA ethertype 0xdada (Unknown), rsvd 0 0, mode From CPU, target dev 0, port 4, untagged, VID 0, FPri 0, ethertype IPv6 (0x86dd), length 126: (flowlabel 0xccb2c, hlim 1, next-header ICMPv6 (58) payload length: 64) fe80::96XX:XXff:feXX:XXbd > ff02::2: [icmp6 sum ok] ICMP6, echo request, id 9523, seq 0
```

tmcqueen-materials, 2025-04-13:
@cranderson : You are correct, this sure looks like some artifact from the switch sending it to all ports instead of just those associated with the expected output. Are you sure this wasn't broken on 23.05.5? I don't see any obvious changes to 88E6172 that would explain why this would happen on 24.10 and not 23.05 (there appears to have been similar breakage in previous releases, e.g. #10997, #11877). 

I unfortunately have no hardware to test with to further isolate the cause. Perhaps a dev can chime in.

cranderson, 2025-04-13:
> [@cranderson](https://github.com/cranderson) : You are correct, this sure looks like some artifact from the switch sending it to all ports instead of just those associated with the expected output. [...]

I'm fairly certain my IPv6 wasn't broken on `server` in 23.05.5 :-)  I suppose it is possible that the problem was there, but hidden by a higher `metric` on the bad routes?

> I unfortunately have no hardware to test with to further isolate the cause. Perhaps a dev can chime in.

Thanks for your help so far.  I'll look at those other issues.

cranderson, 2025-04-13:
I'm not sure how to interpret these outputs.  I think the dump shows VID 1 as untagged on` lan1`-`lan4` (switch chip ports 3,2,1,0) and not a member of `wan` (switch chip port 4).  But looking at the tcpdump above, it seems maybe the `br-lan` traffic from the CPU is being sent to `mode Forward, dev 1, port 0, untagged, VID 4095`.  And VID 4095 below shows all ports including switch chip port 4 (`wan`) as `V - a member, egress not modified`.  Maybe that's the problem?

```
root@OpenWrt:~# bridge vlan
port              vlan-id  
lan4              1 PVID Egress Untagged
lan3              1 PVID Egress Untagged
lan2              1 PVID Egress Untagged
lan1              1 PVID Egress Untagged
br-guest          1 PVID Egress Untagged
br-iot            1 PVID Egress Untagged
phy1-ap0          1 PVID Egress Untagged
phy0-ap0          1 PVID Egress Untagged
phy0-ap1          1 PVID Egress Untagged
phy0-ap2          1 PVID Egress Untagged
br-lan            1 PVID Egress Untagged

root@OpenWrt:~# mv88e6xxx_dump --vtu
Using device <mdio_bus/f1072004.mdio-mii:00>
VTU:
	V - a member, egress not modified
	U - a member, egress untagged
	T - a member, egress tagged
	X - not a member, Ingress frames with VID discarded
P  VID 0123456  FID  SID QPrio FPrio VidPolicy
0    0 VVVVVVV    0    0     -     -     1  
0    1 UUUUXVV    2    0     -     -     0  
0 4095 VVVVVVV    1    0     -     -     0  
```

cranderson, 2025-04-14:
```
root@OpenWrt:~# dmesg|grep mv88 | head -20
[    1.629644] mv88e6085 f1072004.mdio-mii:00: switch 0x1720 detected: Marvell 88E6172, revision 1
[    1.962991] mv88e6085 f1072004.mdio-mii:00: switch 0x1720 detected: Marvell 88E6172, revision 1
[    2.835326] mv88e6085 f1072004.mdio-mii:00: configuring for fixed/rgmii-id link mode
[    2.847494] mv88e6085 f1072004.mdio-mii:00: Link is Up - 1Gbps/Full - flow control off
[    2.951893] mv88e6085 f1072004.mdio-mii:00 lan4 (uninitialized): PHY [mv88e6xxx-0:00] driver [Marvell 88E1540] (irq=57)
[    3.061961] mv88e6085 f1072004.mdio-mii:00 lan3 (uninitialized): PHY [mv88e6xxx-0:01] driver [Marvell 88E1540] (irq=58)
[    3.169890] mv88e6085 f1072004.mdio-mii:00 lan2 (uninitialized): PHY [mv88e6xxx-0:02] driver [Marvell 88E1540] (irq=59)
[    3.269723] mv88e6085 f1072004.mdio-mii:00 lan1 (uninitialized): PHY [mv88e6xxx-0:03] driver [Marvell 88E1540] (irq=60)
[    3.379788] mv88e6085 f1072004.mdio-mii:00 wan (uninitialized): PHY [mv88e6xxx-0:04] driver [Marvell 88E1540] (irq=61)
[    9.397535] mv88e6085 f1072004.mdio-mii:00 lan1: configuring for phy/gmii link mode
[    9.463579] mv88e6085 f1072004.mdio-mii:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   14.355073] mv88e6085 f1072004.mdio-mii:00 lan1: Link is Down
[   24.272303] mv88e6085 f1072004.mdio-mii:00 lan1: configuring for phy/gmii link mode
[   24.308336] mv88e6085 f1072004.mdio-mii:00 lan1: entered allmulticast mode
[   24.385755] mv88e6085 f1072004.mdio-mii:00 lan1: entered promiscuous mode
```

```
root@OpenWrt:~# mv88e6xxx_dump --atu
Using device <mdio_bus/f1072004.mdio-mii:00>
ATU:
FID  MAC	       T 0123456 Prio State
...
   1 94:XX:XX:XX:XX:be   00000100000    0 Static
   1 94:XX:XX:XX:XX:bf   00000100000    0 Static
   1 96:XX:XX:XX:XX:bd   00000100000    0 Static
...
   1 ff:ff:ff:ff:ff:ff   11111100000    0 Static
   2 94:XX:XX:XX:XX:be   00000100000    0 Static
   2 94:XX:XX:XX:XX:bf   00000100000    0 Static
   2 96:XX:XX:XX:XX:bd   00000100000    0 Static
   2 ff:ff:ff:ff:ff:ff   11111100000    0 Static
```

```
root@OpenWrt:~# mv88e6xxx_dump --ports
Using device <mdio_bus/f1072004.mdio-mii:00>
			   0    1    2    3    4    5    6 
00 Port status            100f 9e4f 100f 9e0f 1e0f 0e07 0e07 
01 Physical control       0043 0043 0043 0043 0043 c07e c07e 
02 Jamming control        0000 0000 0000 0000 0000 0000 ff00 
03 Switch ID              1721 1721 1721 1721 1721 1721 1721 
04 Port control           043c 043f 043c 043f 043f 333f 007c 
05 Port control 1         0000 0000 0000 0000 0000 0000 0000 
06 Port base VLAN map     002e 002d 002b 0027 0020 005f 8010 
07 Def VLAN ID & Prio     0fff 0fff 0fff 0fff 0000 0000 0000 
08 Port control 2         0080 0080 0080 0080 0000 1c80 2c80 
09 Egress rate control    0001 0001 0001 0001 0001 0001 0001 
0a Egress rate control 2  0000 0000 0000 0000 0000 0000 8000 
0b Port association vec   1001 1002 1004 1008 1000 1020 0000 
0c Port ATU control       0000 0000 0000 0000 0000 0000 0000 
0d Override               0000 0000 0000 0000 0000 0000 0000 
0e Policy control         0000 0000 0000 0000 0000 0000 0000 
0f Port ether type        9100 9100 9100 9100 9100 dada 9100 
10 In discard low         0000 0000 0000 0000 0000 0000 0000 
11 In discard high        0000 0000 0000 0000 0000 0000 0000 
12 In filtered            0000 0000 0000 0000 0000 0000 0000 
13 RX frame count         0000 77e9 0000 a22a a5ff 3384 0000 
14 Reserved               0000 0000 0000 0000 0000 0000 0000 
15 Reserved               0000 0000 0000 0000 0000 0000 0000 
16 LED control            6011 6011 6011 6011 0033 0033 0000 
17 Reserved               0000 0000 0000 0000 0000 0000 0000 
18 Tag remap low          3210 3210 3210 3210 3210 3210 3210 
19 Tag remap high         7654 7654 7654 7654 7654 7654 7654 
1a Reserved               0000 0000 0000 0000 0000 0000 0000 
1b Queue counters         8000 8000 8000 8000 8000 8000 8000 
1c Reserved               0000 0000 0000 0000 0000 0000 0000 
1d Reserved               0000 0000 0000 0000 0000 0000 0000 
[... 54 lines cut ...]
08 Trunk mapping                    7800
09 Ingress rate command             1600
0a Ingress rate data                0000
0b Cross chip port VLAN addr        3010
0c Cross chip port VLAN data        002f
0d Switch MAC/WoL/WoF               05b2
0e ATU Stats                        001c
0f Priority override table          0f00
10 Reserved                         0000
11 Reserved                         0000
12 Reserved                         0000
13 Reserved                         0000
14 EEPROM command                   0400
15 EEPROM addr                      0000
16 AVB command                      0000
17 AVB data                         0000
18 SMI PHY command                  1825
19 SMI PHY data                     df41
1a Scratch & Misc                   7301
1b Watchdog control                 0029
1c QoS Weights                      0000
1d Misc                             8000
1e Reserved                         0000
1f Reserved                         0000
```

Neustradamus, 2025-04-22:
To follow this ticket.

9uyone, 2025-05-01:
The same on bpi r3

9uyone, 2025-06-06:
In my case the problem was caused by Docker, which I had installed on my router. It was automatically creating some IPv6 firewall rule that was blocking traffic. DHCPv6 and RA were working fine, but ping no

fpengziyang, 2025-09-17:
解决了没？我的也是这样。