openwrt/openwrt #18477: rpi 3b boot loop

CodeSpoof, 2025-04-13:
### Describe the bug

After Connecting the device to power, the boot process starts, visible n serial console. After approximatly 20 seconds, the boot process seems to restart. This is before any access to the serial terminal is possible. Booting into failsaf seems to work, but only partially.
Every once in a while the boot process get's far enough to enable access to the (normal) serial console, but most of the time just restarts boot after a few seconds anyway. After a few tries, it is sometimes able to keep from crashing and allows access.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

bcm27xx/bcm2710

### Device

Raspberry Pi 3 Model B Rev 1.2

### Image kind

Official downloaded image

### Steps to reproduce

- Flash SD card
- Boot RPi 3B
- Watch serial console

### Actual behaviour

Repeated boot attempts

### Expected behaviour

Boot without looping

### Additional info

The RPi has its power supplied from the Arduino that I use as a serial adapter.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-04-16:
Please try with the Pi powered by a standard Pi power supply not via the Arduino.

schuettecarsten, 2025-04-22:
> Please try with the Pi powered by a standard Pi power supply not via the Arduino.

This has nothing to do with the power supply. I have the same issue with an rpi4 (bcm2711) and an original power supply. It started some days ago, the last successful build that was on April 8th.

See also here:
[Device 'grafx' does not have a release() function](https://github.com/raspberrypi/linux/issues/6777#top)

And maybe:
https://github.com/torvalds/linux/commit/f44fa354a0715577ca32b085f6f60bcf32c748dd

@Noltari what do you think?

CodeSpoof, 2025-04-22:
I have tried it with the original power supply aswell, but with the same result.

Noltari, 2025-04-22:
> > Please try with the Pi powered by a standard Pi power supply not via the Arduino.
> 
> I doubkt that this is caused by the power supply. I have the same issue with an rpi4 (bcm2711) and an original power supply, the issue also happens on other power supplies. It started some days ago, the last working (snapshot) build was on April 8th.
> 
> See also here: [Device 'grafx' does not have a release() function](https://github.com/raspberrypi/linux/issues/6777#top)
> 
> And maybe: [torvalds/linux@f44fa35](https://github.com/torvalds/linux/commit/f44fa354a0715577ca32b085f6f60bcf32c748dd)
> 
> [@Noltari](https://github.com/Noltari) what do you think?

@schuettecarsten no, that's a different issue and the warning is harmless.

I'll try to investigate this.

Noltari, 2025-04-22:
Update: I can boot a RPi 3B on OpenWrt 24.10.0, 24.10.1 and master images with no issues at all:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.73 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28427-6df0e3d02a) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Mon Feb  3 23:09:37 2025
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000037400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000037400000..0x000000003b3fffff (65536 KiB) map reusable linux,cma
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0, zone DMA: 19456 pages in unavailable ranges
[    0.000000] percpu: Embedded 19 pages/cpu s37160 r8192 d32472 u77824
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 cgroup_disable=memory bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 console=ttyAMA0,115200 root=PARTUUID=952f2311-02 rootfstype=squashfs,ext4 rootwait
[    0.000000] cgroup: Disabling memory control group subsystem
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 238896
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 872580K/970752K available (8704K kernel code, 924K rwdata, 1488K rodata, 448K init, 558K bss, 32636K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: 0xffffffc080010044
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000259] Console: colour dummy device 80x25
[    0.000275] printk: console [tty1] enabled
[    0.000724] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000759] pid_max: default: 32768 minimum: 301
[    0.001154] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.001199] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.003451] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.003758] rcu: Hierarchical SRCU implementation.
[    0.003779] rcu: 	Max phase no-delay instances is 1000.
[    0.004558] smp: Bringing up secondary CPUs ...
[    0.005137] Detected VIPT I-cache on CPU1
[    0.005264] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.006163] Detected VIPT I-cache on CPU2
[    0.006252] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.007049] Detected VIPT I-cache on CPU3
[    0.007132] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.007409] smp: Brought up 1 node, 4 CPUs
[    0.007494] SMP: Total of 4 processors activated.
[    0.007512] CPU features: detected: 32-bit EL0 Support
[    0.007528] CPU features: detected: CRC32 instructions
[    0.007627] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.007653] CPU: All CPU(s) started at EL2
[    0.007667] alternatives: applying system-wide alternatives
[    0.021053] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.021121] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.023876] pinctrl core: initialized pinctrl subsystem
[    0.025242] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.031723] DMA: preallocated 1024 KiB GFP_KERNEL pool for atomic allocations
[    0.032107] DMA: preallocated 1024 KiB GFP_KERNEL|GFP_DMA pool for atomic allocations
[    0.032687] DMA: preallocated 1024 KiB GFP_KERNEL|GFP_DMA32 pool for atomic allocations
[    0.033410] thermal_sys: Registered thermal governor 'step_wise'
[... 234 lines cut ...]
[   14.183023] smsc95xx 1-1.1:1.0 eth0: entered allmulticast mode
[   14.190323] smsc95xx 1-1.1:1.0 eth0: entered promiscuous mode
```

For some reason your RPi gets some smsc95xx and the USB ethernet is disconnected
```
[... cut ...]
[    3.339321] smsc95xx 1-1.1:1.0 (unnamed net_device) (uninitialized): Failed to read reg index 0x00000114: -71
[    3.352402] smsc95xx 1-1.1:1.0 (unnamed net_device) (uninitialized): Error reading MII_ACCESS
[    3.363988] smsc95xx 1-1.1:1.0 (unnamed net_device) (uninitialized): smsc95xx_mdio_read: MII is busy
[    3.377511] usb 1-1.1: USB disconnect, device number 3
[    3.590926] Indeed it is in host mode hprt0 = 00021501
```
And detected again a couple of times...

Can you try the same device with Raspberry Pi OS and see what happens?