openwrt/openwrt #18491: x86 microcode loader is broken in 24.10.1

k-gabor, 2025-04-14:
### Describe the bug

Microcode loading is broken in version 24.10.1, while it worked correctly in 24.10.0.

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

x86/64

### Device

Techvision TVI7309X

### Image kind

Official downloaded image

### Steps to reproduce

Run the following command to trigger late microcode loading:
`echo 1 > /sys/devices/system/cpu/microcode/reload`
The command should return without any error message.

### Actual behaviour

Running the command gives back an error:
`echo 1 > /sys/devices/system/cpu/microcode/reload
ash: write error: No such file or directory`

Additionally, there is no indication in dmesg that a microcode update was attempted or applied.

### Expected behaviour

The command should succeed silently with no output or errors. Upon execution, the system should attempt to apply any available late microcode updates, and relevant log entries should appear in dmesg.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

dissent1, 2025-04-19:
+1
https://github.com/openwrt/openwrt/issues/18538

dissent1, 2025-04-19:
Setting microcode.force_minrev=0 in boot parameters (grub.cfg) fixes the issue, microcode gets updated.

jeandestouches, 2025-04-20:
With amd64-microcode installed and microcode.force_minrev=0
I now have: "sh: write error: File descriptor in bad state"

```
ls /lib/firmware/amd-ucode
microcode_amd.bin         microcode_amd_fam16h.bin  microcode_amd_fam19h.bin
microcode_amd_fam15h.bin  microcode_amd_fam17h.bin
```

```
[    3.998524] microcode: Current revision: 0x07000110
[    4.003572] microcode: Microcode Update Driver: v2.2.
...
[    6.443654] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
[    6.458857] init: - preinit -
sh: write error: File descriptor in bad state
[    8.758607] random: crng init done
```

```
echo 1 > /sys/devices/system/cpu/microcode/reload
ash: write error: File descriptor in bad state
```

If I omit microcode.force_minrev=0 from cmdline,
then output is the same as in previous post.
Thanks for reading.

k-gabor, 2025-04-20:
It seems this issue was introduced with kernel 6.6.81.
As @dissent1 pointed out, OpenWrt commit 0fb8d7b enables the kernel config option CONFIG_MICROCODE_LATE_FORCE_MINREV=y.

However, support for this option is not yet implemented in the upstream Linux kernel. The relevant code path lacks the necessary logic to handle this option properly.
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/x86/kernel/cpu/microcode/intel.c?h=linux-6.6.y&id=640fad833ddce708ec92640ec247d4139433dc3b#n474

dissent1, 2025-04-20:
> However, support for this option is not yet implemented in the upstream Linux kernel. The relevant code path lacks the necessary logic to handle this option properly.

The same for amd.c, so using force_minrev=0 effectively makes the routine to continue.
Have you tried earlier version of amd microcode? There was a bump
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=7a9a646c72b5c1ed0cbab931d209c8b9fa1a3803

myanesp, 2025-06-09:
Hi,

I've recently upgraded my router from a Xiaomi to a Topton x64, the same model as OP. I've installed OpenWRT twice in two different SSDs, and at the end, I'm facing random reboots. The maximum uptime that I had is a day and a half, then, it rebooted again. Searching for this problem I got here. 

Can my problem be related to this issue? When I execute `echo 1 > /sys/devices/system/cpu/microcode/reload`, I also see the message of `ash: write error: No such file or directory`.

Thanks in advance :)

k-gabor, 2025-06-09:
Hi @myanesp ,

I'm not sure that your issue is related. I don't experience random reboots, yours seems more like a power/hardware/overheating problem.
This motherboard has an unfortunate layout, the nvme slot is right above the wifi card, so the wifi card can heat up the nvme.

myanesp, 2025-06-09:
@k-gabor, that's what I thought, but the host has an external fan that keeps its temperature under 30ºC. And also that I've been running OPNsense on the same machine for 6 months or so, and never experienced this. So that's why I thought that is an OpenWRT related issue, and when searching I arrived here. I'll deep into it more.

Thanks!

MarcoMartins86, 2026-10-19:
On the early loading @feckert mentioned: #25997 adds it as an image option, `CONFIG_GRUB_MICROCODE`. It takes a list of files from `intel-microcode` or `amd64-microcode`, such as `intel-ucode/06-7a-08`, puts them in `/boot/microcode.cpio`, and has the default GRUB entry load that as an initrd. On a Celeron J4125 the kernel then logs `microcode: Updated early from: 0x00000018`, where late loading refuses the same update.

It's opt-in, for images built for a known CPU, from source or with the ImageBuilder. All of Intel's microcode is about 21 MiB, more than the 16 MiB boot partition, so official images stay as they are and keep the late loading this issue is about. AMD's files are about 300 KB in all, so an image can list every one of them.