openwrt/openwrt #18494: Wi-Fi (AW859A) Not Detected on Orange Pi Zero 3 with OpenWrt 24.10.0

krill-amorim, 2025-04-15:
### Describe the bug

Device: Orange Pi Zero 3
Architecture: armv7l (as detected by OpenWrt) - Note: The OpenWrt TOH page indicates aarch64 architecture. Please verify the correct architecture.
OpenWrt Version: 24.10.0 (downloaded from https://openwrt.org/toh/xunlong/orange_pi_zero_3)

Problem Description:
After installing the OpenWrt 24.10.0 image on my Orange Pi Zero 3, the integrated Wi-Fi interface is not detected by the system. The command `ip a` only lists the loopback (lo), Ethernet (eth0), and LAN bridge (br-lan) interfaces. No wlan0 or similar interface appears.

Steps Performed:

* **Interface List Verification:** Executed the `ip a` command to list active network interfaces. Output file atached
    
    
* **Kernel Log Analysis:** Examined the output of the `dmesg` command for errors or information related to the Wi-Fi (AW859A), drivers (mt76, etc.), or firmware. I did not find any clear indications of successful Wi-Fi driver loading or explicit Wi-Fi related errors. However, I observed multiple "request() failed for pin ..." errors for various peripherals (serial, SPI, Ethernet, MMC), suggesting a potential underlying hardware configuration issue. Output atached:
    
[dmesg-output.txt](https://github.com/user-attachments/files/19746204/dmesg-output.txt)
[ip -c a.txt](https://github.com/user-attachments/files/19746205/ip.-c.a.txt)

* **OpenWrt Device Page Verification:** Consulted the official OpenWrt page for the Orange Pi Zero 3, which confirms the Wi-Fi hardware is an Allwinner AW859A (aka CDTech 20U5622). However, there are no specific instructions on how to configure Wi-Fi on this page, suggesting that support should be native.

Additional Information:

* The Orange Pi Zero 3 board is functioning, and the Ethernet network is configured correctly (I can access the system via SSH).
* There are no visible Wi-Fi activity LEDs on the board (if any exist).

Possible Causes:

* Kernel driver for the AW859A is not included in the default image or is not loading correctly.
* Firmware for the AW859A is missing.
* Issues with the GPIO pin configuration required for Wi-Fi in the Device Tree (evidenced by the multiple pinctrl errors).
* Some other hardware initialization problem.

Request:
I would like to request assistance in identifying and resolving this issue. I need information on:

* Whether the driver for the Allwinner AW859A should be included in the OpenWrt 24.10.0 image for the Orange Pi Zero 3.
* If any specific configuration is required to enable Wi-Fi (e.g., installation of additional packages).
* How to troubleshoot the pinctrl errors that may be preventing the correct initialization of the hardware, including Wi-Fi.
* Any other information or diagnostic steps that might help identify the root cause of the problem.

Thank you for the attention and help from the development team.

Krill Amorim
Date of Detection - April 13, 2025
Date of Post Creation - April 14,2025

### OpenWrt version

24.10.0

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

sunxi/cortexa53

### Device

OrangePi Zero3

### Image kind

Official downloaded image

### Steps to reproduce

1.  Install the OpenWrt 24.10.0 image onto an SD card using the firmware available at: https://openwrt.org/toh/xunlong/orange_pi_zero_3 (using either the Firmware OpenWrt Install URL or the Firmware OpenWrt Upgrade URL as appropriate for your installation method).

2.  Boot the Orange Pi Zero 3 with the prepared SD card.

3.  Access the device via SSH.

4.  Once logged in via SSH, execute the command `ip a` to list the available network interfaces.

5.  Observe that the integrated Wireless interface (wlan0 or similar) is not listed in the output. Only the loopback (`lo`), Ethernet (`eth0`), and LAN bridge (`br-lan`) interfaces are present.

### Actual behaviour

The system boots successfully and the Ethernet interface (`eth0`) is detected and functional, allowing access via SSH. However, the integrated Wi-Fi interface (Allwinner AW859A) is not detected by the operating system. The output of the `ip a` command only lists the following network interfaces:

1: `lo`: The loopback interface.
2: `eth0`: The Ethernet interface, which is currently active and connected.
3: `br-lan`: The LAN bridge interface, associated with the Ethernet connection.

There is no `wlan0` or any other wireless interface listed, indicating that the Wi-Fi hardware is not being recognized or initialized by the system.

### Expected behaviour

Expected Behavior:

Upon booting the system after installing the OpenWrt image, the system should detect and initialize the integrated Allwinner AW859A Wi-Fi adapter as a network interface (typically named `wlan0`). The `ip a` command should list four network interfaces:

1: `lo`: The loopback interface.
2: `eth0`: The Ethernet interface.
3: `br-lan`: The LAN bridge interface.
4: `wlan0`: The wireless network interface, representing the integrated Wi-Fi adapter.

### Additional info

For comparison, I have tested the same Orange Pi Zero 3 hardware with the latest Armbian stable image. In the Armbian environment, the integrated wireless LAN interface (`wlan0`) is detected and operational out-of-the-box, functioning as plug and play. The system works correctly, confirming that the Wi-Fi hardware itself is functional.

While Armbian demonstrates the Wi-Fi capability of the device, OpenWrt remains my preferred operating system, particularly for its suitability as a travel router. I am keen to contribute to the OpenWrt project by reporting this issue and assisting in its resolution to improve the platform's support for the Orange Pi Zero 3.

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

wigyori, 2025-04-15:
No aw859a (uwe5622) driver in OpenWrt - the available driver is FUBAR. If anyone wants to work on it, feel free, I've dropped the idea of touching that after a few tries.

krill-amorim, 2025-04-15:
> Invalid Version reported. `24.10.0` Is this from a clean repository?

As detailed in the report, I downloaded the OpenWrt 24.10.0 image from the following URL:

https://openwrt.org/toh/xunlong/orange_pi_zero_3

krill-amorim, 2025-04-15:
> No aw859a (uwe5622) driver in OpenWrt - the available driver is FUBAR. If anyone wants to work on it, feel free, I've dropped the idea of touching that after a few tries.

It would be good to detail this information on the download page or in the documentation to warn future users about the potential lack of immediate Wi-Fi support.

closed 2025-04-15 by krill-amorim

krill-amorim, 2025-04-15:
Thank you all for your time.

Unfortunately, I am currently unable to dedicate time to further investigate or work on improving Wi-Fi support for the Orange Pi Zero 3 on OpenWrt.

I will proceed with using Armbian for the time being.

Perhaps in the future, I will have the opportunity to contribute some support for this device.

Once again, thank you everyone.

salmanuc, 2026-08-24:
There now appears to be a working community implementation for the AW859A/UWE5622 on the Orange Pi Zero 2/Zero 3:

https://github.com/Anieake/openwrt-orangepi-zero2-zero3-wifi-modem

It provides a custom OpenWrt 25.12 build with the required UWE5622 driver, and the author has also provided a prebuilt image. I have tested the image on an Orange Pi Zero 3 and can confirm that the built-in Wi-Fi works.

It would be great if the required UWE5622/AW859A support could eventually be upstreamed into official OpenWrt. I have also asked the author of the implementation to consider submitting the required changes upstream, so that users don’t need to rely on custom images to use the onboard Wi-Fi.