openwrt/openwrt #18495: The site https://firmware-selector.openwrt.org is not working

NOX69RUS, 2025-04-15:
### Describe the bug

If i request a custom build on this site for GL.iNet GL-MT6000, it gives an error with the text "Unsupported version: 24.10.1".

### OpenWrt version

24.10.1 (r28597-0425664679)

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

24.10.1

### Device

GL.iNet GL-MT6000

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

NOX69RUS, 2025-04-15:
from a supported device https://firmware-selector.openwrt.org/?version=24.10.1&target=mediatek%2Ffilogic&id=glinet_gl-mt6000

systemcrash, 2025-04-15:
https://firmware-selector.openwrt.org is working. 
https://sysupgrade.openwrt.org/ isn't. This bug is invalid here. Please close it.

closed 2025-04-15 by NOX69RUS

systemcrash, 2025-04-15:
There are plenty of issues already in the correct repos - don't worry :)