openwrt/openwrt #18496: Xiaomi AIoT AC2350 5GHz WiFi missing after upgrade to release 24.10.1

tgajos, 2025-04-15:
### Describe the bug

AC2350 was working well with 23.05 and earlier versions, but after upgrading to 24.10.0 or 24.10.1, 5 GHz Wi-Fi is gone. 2.4 GHz works fine. 

### OpenWrt version

r28597-0425664679

### OpenWrt release

24.10.1

### OpenWrt target/subtarget

ath79/generic

### Device

Xiaomi AIoT AC2350

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

aimacintyre, 2025-04-15:
Did you upgrade preserving the configuration?

tgajos, 2025-04-15:
> Did you upgrade preserving the configuration?

First upgrade I did with preserving configuration. It was upgrade from 23.05.5 -> 24.10.0 - 5GHz radio dissapeared.
Next I reverted back to 23.05.5 with cleaning all the configuration  - 5GHz radio started to work.
Third upgrade was after appearing 24.10.1 release, so, I upgraded from 23.05.5 -> 24.10.1 with cleaning all the configuration - 5GHz radio dissapeared again.

ParronS3, 2025-04-19:
Seems the same problem as #17582  I´m having same problem too. And trying to solve it.

ParronS3, 2025-04-19:
in the other issue is the solution.
SSHed into the router CD=> /lib/firmware/ath10k/QCA9984/hw1.0, deleted board-2.bin. Wgetted the one linked there and reboot. Working like a charm.

mcurly, 2025-04-29:
Hi,

I also used another workaround, by copying the caldata file from [this script](https://github.com/openwrt/openwrt/blob/openwrt-23.05/target/linux/ath79/generic/base-files/etc/hotplug.d/firmware/11-ath10k-caldata) , and replacing the one in `/etc/hotplug.d/firmware/11-ath10k-caldata` with the one from the former url.

This workaround/solution is (among other locations) present in this post [here](https://github.com/openwrt/openwrt/issues/17175#issuecomment-2531723181).

If you follow the above post from @zekica , there is a direct link to the same exact caldata file that could/should replace the one in the above `/etc/hotplug.d/firmware/` directory.

I will call another Xiaomi-AC2350 owner, here ( @bitfarmer ) from [this post](https://github.com/openwrt/openwrt/issues/14541#issuecomment-2821024761), so that, if he would feel like testing any of the workarounds in this thread, here, maybe he could reproduce the same results as I (and others, I believe), had.

Also, (and finally), I would like to summon / call the attention of the OpenWRT developers here to know if they are aware of this issue and if a more definitive solution might be underway.

@igor-dyatlov issued a [pull-request](https://github.com/openwrt/firmware_qca-wireless/pull/83/commits/89047083e14b39774b931a66ede7cb0a1a2cfcfb) that the user @cujanovic mentioned. My abilities to test pull requests are limited. So I can't vouch for the PR, yet, since this issue is/was still present in latest 24.10.1 stable version, I am adding myself to the 'lobbiers' 😄 for to see if this gets implemented in a short period of time.

Do apologize if this attitude of mine seems hasty.

Cheers!

bitfarmer, 2025-04-29:
Using a clean install of **24.10.1** to circumvent earlier changes I might have made to previously fix this, I have tried just now. 

Besides taking the script which replaces '/etc/hotplug.d/firmware/11-ath10k-caldata', I noticed that within that file,  'xiaomi,aiot_ac2350' it wants to load '/lib/firmware/ath10k/QCA9888/hw2.0/board.bin'.

So right now building that directory and placing the file 'board-xiaomi_ac2350.qca9988' renamed as board.bin in it

Let's see what happens on reboot…

bitfarmer, 2025-04-29:
That didn't work. Right now this is the layout
`root@stargate-AC2350:~# ls -lR /lib/firmware/ath10k/
/lib/firmware/ath10k/:
drwxr-xr-x    3 root     root             0 Apr 29 21:43 QCA9888
drwxr-xr-x    1 root     root             0 Apr 29 21:43 QCA9984
-rw-r--r--    1 root     root         12064 Apr 29 21:20 pre-cal-pci-0000:00:00.0.bin

/lib/firmware/ath10k/QCA9888:
drwxr-xr-x    2 root     root             0 Apr 29 21:43 hw2.0

/lib/firmware/ath10k/QCA9888/hw2.0:
-rw-r--r--    1 root     root         12148 Apr 29 21:28 board.bin

/lib/firmware/ath10k/QCA9984:
drwxr-xr-x    1 root     root             0 Apr 29 21:36 hw1.0

/lib/firmware/ath10k/QCA9984/hw1.0:
-rw-r--r--    1 root     root        281312 Apr 13 17:38 board-2.bin
-rw-r--r--    1 root     root        621724 Apr 13 17:38 firmware-5.bin`

and 

`root@stargate-AC2350:~# ls -l /etc/hotplug.d/firmware/
-rw-r--r--    1 root     root          3548 Apr 13 17:38 10-ath9k-eeprom
-rw-r--r--    1 root     root           568 Apr 29 21:19 11-ath10k-caldata`

but still this is what DMESG indicates:

> [   20.523203] ath10k 6.10 driver, optimized for CT firmware, probing pci device: 0x46.
[   20.533397] ath10k_pci 0000:00:00.0: enabling device (0000 -> 0002)
[   20.540187] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   22.553229] ath10k_pci 0000:00:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[   22.563542] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   22.577940] ath10k_pci 0000:00:00.0: firmware ver 10.4b-ct-9984-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7ea63dc5
[   24.972498] ath10k_pci 0000:00:00.0: failed to fetch board data for bus=pci,bmi-chip-id=0,bmi-board-id=16 from ath10k/QCA9984/hw1.0/board-2.bin
[   25.493901] ath10k_pci 0000:00:00.0: failed to fetch board-2.bin or board.bin from ath10k/QCA9984/hw1.0
[   25.503683] ath10k_pci 0000:00:00.0: failed to fetch board file: -12
[   25.521919] ath10k_pci 0000:00:00.0: could not probe fw (-12)

I'm afraid that I don't have sufficient familiarity with these drivers but I do notice that it is still trying to load from the 'QCA9984/hw1.0' path.

Any ideas!

mcurly, 2025-04-29:
>  `/etc/hotplug.d/firmware/11-ath10k-caldata` I noticed that within that file (...) it wants to load 
> `/lib/firmware/ath10k/QCA9888/hw2.0/board.bin`

Hi, @bitfarmer , can you please point where you see the reference to QCA9888 directory? 

[image]

I only caught this reference which is QCA9984... Could that be the source of your outcome?

mcurly, 2025-04-29:
> That didn't work. Right now this is the layout (...)
> 
> Any ideas!

Hi again, @bitfarmer , I am guessing that you tried to follow both of the workarounds at the same time. 

And I did not explicitly mentioned to use both of them simultaneously, and I mentioned `any` although I only tested the `caldata` one.

And I did not explicitly mentioned to use both of them simultaneously. 

Which could be the cause of your issue, right now. 

How is your router? 

Is it yet operable? 

Did you make any backup?

bitfarmer, 2025-04-30:
Good morning,

Perhaps I was rushing last night, I'm in  UTC land here but the location where I got the script that refers to QCA9988 is the one in your original message:  "@igor-dyatlov issued a [pull-request](https://github.com/openwrt/firmware_qca-wireless/pull/83/commits/89047083e14b39774b931a66ede7cb0a1a2cfcfb) that the user @cujanovic mentioned." _You have to choose view file whereupon you will given the opportunity to download it. It's not very intuitive I'll admit._

Also since it was pretty late I didn't try all options (or indeed use my brain to analyse) but just tried applying one of them.

I'll reexamine the whole thing later today and report. (that will be in about 4 hours though.

Later…

bitfarmer, 2025-04-30:
_One further point… This router is still in service. I'll replace it with another one when I return which will make things a bit easier_

mcurly, 2025-04-30:
> _One further point… This router is still in service. (...)

Hello, @bitfarmer , good afternoon.

Good news then (I think, at least)...

Yes maybe we both were hasty. (I, for one, admit I was hasty, to believe that my language was unequivocal, which, in good terms, no one can admit that).

And, I think that you, @bitfarmer as you have said, were rushing as well. 

But, beyond that, I am wondering where could it be the origin of your decision to put that `bin` file over at the `/lib/firmware/ath10k/QCA9888/hw2.0/board.bin` location.

Reading, your [post](https://github.com/openwrt/openwrt/issues/18496#issuecomment-2840213939) , (perhaps in rush, as well), I jumped into the conclusion that you had used **BOTH** of the named solutions. And something (which I don't follow or don't fully understand) has gone weary. I thought that you had placed the given `caldata` file inside `/etc/hotplug.d/firmware/` folder, and, **SIMULTANEOUSLY**, had created that directory (either `QCA9888` or any other with a similar name...) and had placed the `board.bin` file that you had downloaded from the @igor-dyatlov 's PR.

I am 😲 and 🔇 What can I do. I know I did solve/workaround my issue by using the `caldata` file. I also know that others have followed the `board.bin` file direction. And I am a bit surprised there would be this outcome... So... tell me @bitfarmer ...

What will you do and what would you do if you were in my shoes?

I just hope that in the end, this turns out alright for both of us.

Sincerely,

bitfarmer, 2025-04-30:
I went back to square one this afternoon and tried the 

> I also used another workaround, by copying the caldata file from [this [...]

 First, inserting the much larger '11-ath10k-caldata' which this time refers 

> /lib/firmware/ath10k/QCA9984/hw1.0/board.bin

It should be noted that there is only a board-2.bin and no board.bin file present. Was I to get the board.bin file elsewhere?!

Any that didn't produce any results either so I decided to go back to V23.05.5 and pull the files there

### V23.05.5 Installation
```
# ls -l /etc/hotplug.d/firmware/
-rw-r--r--    1 root     root          3882 Sep 23  2024 10-ath9k-eeprom
-rw-r--r--    1 root     root          7534 Sep 23  2024 11-ath10k-caldata

# ls -lR /lib/firmware/ath10k/
/lib/firmware/ath10k/:
drwxr-xr-x    1 root     root             0 Sep 23  2024 QCA9984
-rw-r--r--    1 root     root         12064 Apr 30 16:34 pre-cal-pci-0000:00:00.0.bin

/lib/firmware/ath10k/QCA9984/hw1.0:
-rw-r--r--    1 root     root        281312 Sep 23  2024 board-2.bin
lrwxrwxrwx    1 root     root            49 Apr 30 16:34 board.bin -> /lib/firmware/ath10k/pre-cal-pci-0000:00:00.0.bin
-rw-r--r--    1 root     root        621724 Sep 23  2024 firmware-5.bin

# sha256sums
9a04d0c801c69db16399c47e5e65ef6311da2785918b26fa83b9392e83b81e64  /lib/firmware/ath10k/QCA9984/hw1.0/board-2.bin
7bfe5bf7c38532fa57db62ebc56ec625583928d5d4736475d5dec4d4ae031154  /lib/firmware/ath10k/QCA9984/hw1.0/firmware-5.bin
2eb16ddd722a2bb02ed8a04858fa74233f69f11d5153fe2886f9eec0b975f4b0  /lib/firmware/ath10k/QCA9984/hw1.0/board.bin
```
Interesting to note that board.bin is symlinked to 'pre-cal-pci-0000:00:00.0.bin' but as I have said before I am I'm not familiar with the methods used here.

Comparing these files with **24.10.1** apart from the missing board.bin, the other files are the same!

```
Shasums
9a04d0c801c69db16399c47e5e65ef6311da2785918b26fa83b9392e83b81e64  /lib/firmware/ath10k/QCA9984/hw1.0/board-2.bin
7bfe5bf7c38532fa57db62ebc56ec625583928d5d4736475d5dec4d4ae031154  /lib/firmware/ath10k/QCA9984/hw1.0/firmware-5.bin
```

I'm going to try fitting out v24 with the missing files and the older 11-ath10k-caldata.

bitfarmer, 2025-04-30:
I didn't see you latest post @mcurly until i had finished my testing and writing it up. I am just about to test the outfitting of **v24.10.1** with the files extracted from **v23.05.5**. Obviously I took the router out of service as well so I can hack away on it.

_I think I am going to have to do some more learning if that doesn't work_.

mcurly, 2025-04-30:
> I went back to square one this afternoon and tried the
> 
(...)
Wait, @bitfarmer

mcurly, 2025-04-30:
How can you be sure that you went to square one? @bitfarmer ?

What exactly did you do to be sure you were on `square one`?

bitfarmer, 2025-04-30:
> How can you be sure that you went to square one? [@bitfarmer](https://github.com/bitfarmer) ?
> 
> What exactly did you do to be sure you were on `square one`?

'Square One' is just a euphemism for _starting from the beginning again_ :-)

mcurly, 2025-04-30:
OK I know that it is an euphemism and what it generally means... I am not a bot if you might be wondering... What did you do to be sure you were on the so called square one? Did you do a reset?

Also I am getting the list of my router's files and their hashes (sha256sums) and I am a bit surprised at how different they are...

For instance...

[image]

If you take a closer look at the `ls -l` and `ls -lR` command results can you see any difference?

mcurly, 2025-04-30:
I am  particularly puzzled by the size (different size, that is) of  `10-ath9k-eeprom` which is `3548` bytes, more exactly `3548 Apr 13 17:38 10-ath9k-eeprom` (that is what the listing says) and the fact that `pre-cal-pci-0000:00:00.0.bin` is `null` size.  `0 Apr 30 19:58 pre-cal-pci-0000:00:00.0.bin` whilst yours is `12064` bytes...

bitfarmer, 2025-04-30:
To answer your question, I literally reinstalled v23 and checked the files and then I re-inserted the missing ones in a fresh install of v24. I guess that counts as a reset.

It made no difference,  i am still getting a load failure

```
ath10k_pci 0000:00:00.0: failed to fetch board data for bus=pci,bmi-chip-id=0,bmi-board-id=16 from ath10k/QCA9984/hw1.0/board-2.bin
ath10k_pci 0000:00:00.0: failed to fetch board-2.bin or board.bin from ath10k/QCA9984/hw1.0
ath10k_pci 0000:00:00.0: failed to fetch board file: -12
```

So right now I am examing the sources to see if anything stands out. Unfortunately I can't do a 'git blame' on the file because the file itself doesn't appear to be part of the repo.

By the way it appears that only your board.bin has a different shasum, the other two are the same as mine. As I said my board.bin which is the link to `/lib/firmware/ath10k/pre-cal-pci-0000:00:00.0.bin` came from my install of v23.05.5

mcurly, 2025-04-30:
> To answer your question, I literally reinstalled v23 and checked the files and then I re-inserted the missing ones in a fresh install of v24. I guess that counts as a reset.

Hello again @bitfarmer .

Sorry for this late reply. 

Well I'm not that sure if it could count as a reset it not. 

I mean what you did... 

The reason for my late reply is that my router is acting up. 

I mean, by acting up, the fact that its wireless 5g feature, wasn't performing consistently. 

Not even with the above `caldata`workaround. 

At this point I am going to ask you: 

Did you, by chance, turn your router off from the current? After upgrading I mean, when it settled and presented/greeted you with the web `Luci` interface? 

Odd enough, today I did. I did it because the router stopped activating its AC Wireless capability... again... Even after placing the `caldata` file as above instructed... 

So I decided to downgrade to previous stable branch, keeping the apparently working settings with the newest stable branch, 24.10, that is 

So, right now, my router is operational, but with version 23.05.5.

Can anyone else share their thoughts, please? 

Thank you.

bitfarmer, 2025-05-01:
> Did you, by chance, turn your router off from the current? After upgrading I mean, when it settled and presented/greeted you with the web `Luci` interface?

Ok, to clarify matters, I did perform a 'power-on-reset' which didn't change the state of things at all. 

> So, right now, my router is operational, but with version 23.05.5.

Mine is the same, works perfectly on v23 but not v24. I did note that somehow the caldata file got completely butchered between the two releases and that must have caused multiple routers to have 5G failures.

I'm not wasting any more time on this but instead, with my limited time, I am going to comb through the sources to see where this happened.

mcurly, 2025-05-01:
I will keep using 23.05.5 until there are improvements on this situation. 

Cheers

cujanovic, 2025-05-02:
> I will keep using 23.05.5 until there are improvements on this situation.
> 
> Cheers

Why are you not using this method:

https://github.com/openwrt/openwrt/issues/17582#issuecomment-2780666476

Its working stable for me for 2-3 weeks now.

mcurly, 2025-05-02:
> > I will keep using 23.05.5 until there are improvements on this situation.
> > Cheers
> 
> Why are you not using this method:
> 
> [#17582 (comment)](https://github.com/openwrt/openwrt/issues/17582#issuecomment-2780666476)
> 
> Its working stable for me for 2-3 weeks now.

Hi @cujanovic , is that solution persistent across unplugging the router from the current?

cujanovic, 2025-05-02:
> Hi [@cujanovic](https://github.com/cujanovic) , is that solution persistent across unplugging the router from the current?

Yes

mcurly, 2025-05-02:
> > > > I will keep using 23.05.5 until there are improvements on this situation.
> > > > Cheers
> > > 
> > > 
> > > Why are you not using this method:
> > > [#17582 (comment)](https://github.com/openwrt/openwrt/issues/17582#issuecomment-2780666476)
> > > Its working stable for me for 2-3 weeks now.
> > 
> > 
> > Hi [@cujanovic](https://github.com/cujanovic) , is that solution persistent across unplugging the router from the current?
> 
> Yes, but if you update to other version of 24.10.x you should repeat the download and copy.

Ok I'll give that solution a try and post back. 

Thank you @cujanovic

mcurly, 2025-05-04:
> Ok I'll give that solution a try and post back.
>
> Thank you [@cujanovic](https://github.com/cujanovic)

Hello again. 

I'm sorry but I must retract myself for the above comments. I've decided with the Xiaomi router to downgrade. 

I have a total of 3 OpenWrt routers in my place.  It seems to me that only one of them is behaving airtight with latest 24.10.x series. 

The others weren't. Among the ones with oddities was/is Xiaomi AC 2350.

I consider 24.10.x series no to be quite suitable yet for those routers (Xiaomi AC 2350 and also Cudy WR2100). I will going to e wait for further news and when I have the time, get back to the testing with the new series. 

Thank you so much for all your attention and support. 

See you soon

geldot, 2025-05-14:
For the record, proposed fix is at [openwrt/firmware_qca-wireless#83](https://github.com/openwrt/firmware_qca-wireless/pull/83). Hope it'll make it into 24.10.2.

gargullia, 2025-05-18:
@robimarko https://github.com/openwrt/firmware_qca-wireless/pull/83 Is there any hope that this will merge into 24.10.2?

geldot, 2025-06-26:
@gargullia It landed in main but wasn't backported to 24.10.2, so is still not working in the latest release per https://github.com/openwrt/openwrt/issues/17582#issuecomment-3009013538

rapcore2, 2025-07-12:
OpenWrt 24.10.2 r28739-d9340319c6 / LuCI openwrt-24.10 branch 25.186.44806~9ceef01 - not working.