openwrt/openwrt #18500: kernel hang at boot

borkra, 2025-04-15:
### Describe the bug

```
AND read: device 0 offset 0x3200000, size 0x600000
 6291456 bytes read: OK
## Booting kernel from Legacy Image at 02000000 ...
   Image Name:   ARM OpenWrt Linux-6.6.87
   Created:      2025-04-15  11:50:33 UTC
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4201223 Bytes = 4 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK
OK

Starting kernel ...
```

### OpenWrt version

r29221-7921e48d43

### OpenWrt release

SNAPHOT 

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

WRT1900ACv2

### Image kind

Self-built image

### Steps to reproduce

update to the latest image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

borkra, 2025-04-15:
The problem caused by patches by @Ansuel  

from this pull request: https://github.com/openwrt/openwrt/pull/18464

https://github.com/openwrt/openwrt/commit/7843f21c5120c787454c6095ab87209a22cd3a1d
https://github.com/openwrt/openwrt/commit/bed637d9d30df36a518608ae827bc47124b48614

Please fix it.

Ansuel, 2025-04-15:
@borkra thank for testing.  Is it ok if I send you some patch and you test some images for me?

borkra, 2025-04-15:
Yes, I can do that.

stklcode, 2025-04-15:
Can confirm this issue on a _Turris Omnia_, too. Picked your patch and rebuilt ...

Problem seems to be fixed, up and running again on _r28739+505-69890e16b3_ 😄

Ansuel, 2025-04-15:
@anomeome do you mean that it also worked before or it was broken and works with the fix patch?

Anyway yes I have other patch to test since I want to send this fix upstream and wants to check if we can remove also other piece.

anomeome, 2025-04-15:
Was borked, patch resolved issue.

borkra, 2025-04-15:
@Ansuel yes, this patch fixes the issue

Ansuel, 2025-04-15:
try https://gist.github.com/Ansuel/30047e5b8db63c4dafe14f392176c140

anomeome, 2025-04-15:
```
patch: **** malformed patch at line 157: @@ -116,7 +116,7 @@ SECTIONS

Patch failed!  Please fix /home/kc/wrtpac/source/target/linux/generic/backport-6.6/200-v6.11-ARM-9404-1-arm32-enable-HAVE_LD_DEAD_CODE_DATA_ELIMI.patch!

```

Ansuel, 2025-04-15:
Mhhh on second thought I currently lack arm board and those are way too much test to do remotely. So I will just restore the original patch and postpone the upstream later.

flipreverse, 2025-04-15:
@Ansuel FYI, fb4040 and c2600 are both affected. I'm currently building an image for c2600 using the first version of your patch.

borkra, 2025-04-15:
That is actually very strange: MVEBU is Cortex A9 based, it's the most used CPU in the family. I doubt upstream implementation was not tested on it, something else is missing

commit 71446d80 (2025-04-15): generic: fix broken ARM gc sections patch

Upstream version of ARM gc sections skip eeping some section. It was
reported some kernel load hang hence restore what we original did and
introduce a new patch that add the additional entry on top of the
upstream version.

Fixes: #18500
Fixes: 7843f21c5120 ("generic: replace ARM gc sections patch with upstream version")
Tested-by: Stefan Kalscheuer <stefan@stklcode.de> (Turris Omnia)
Link: https://github.com/openwrt/openwrt/pull/18503
closed 2025-04-15 by openwrt-bot