openwrt/openwrt #18502: I can't upgrade to 24.10.1 from asu

castillofrancodamian, 2025-04-15:
### Describe the bug

[image]

[image]

[image]

[image]

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

MERCUSYS MR70X v1

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

newpipe, 2025-04-16:
I Can confirm this issue on these two devices, both using Openwrt 24.10.0:

- Xiaomi Mi Router 3G
- MikroTik cAP ac

broizter, 2025-04-16:
Related https://github.com/openwrt/asu/issues/1327.