openwrt/openwrt #18518: Cannot update Linksys E8450 (UBI) to 24.10.1

svenopdehipt, 2025-04-17:
### Describe the bug

I cannot update the Linksys E8450 (UBI) from version 24.10.0 to 24.10.1. I tested it with a custom images (with custom packages) and the default images. The version stays on version 24.10.0.

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

anarsoul, 2025-04-25:
I have the same issue

anarsoul, 2025-04-25:
It looks like it's caused by missing `fitblk` package in original installation. Try running `opkg update; opkg install fitblk`

svenopdehipt, 2025-04-26:
This did work. But the update destroyed some of my config (network devices were split so that there where to with the same name and the lan connection didn't work anymore. And the routing didn't work anymore so that I had to add the routes manually.).

closed 2025-04-26 by svenopdehipt

anarsoul, 2025-04-27:
> But the update destroyed some of my config

I'm not a openwrt dev, but general recommendation is to backup the configs before upgrading