openwrt/openwrt #18520: Netgear WAX202 missing green:lanX lets after upgrade to 24.10.0

MightySlaytanic, 2025-04-17:
### Describe the bug

I've just upgraded my WAX202 to 24.10.0 and the following settings do not work anymore because green:lan1 green:lan2 and green:lan3 do not exist anymore. I can find only orange:lanX led names:

```
config led 'led_lan1'
	option name 'LAN1'
	option sysfs 'green:lan1'
	option trigger 'none'
	option default '0'

config led 'led_lan2'
	option name 'LAN2'
	option sysfs 'green:lan2'
	option trigger 'none'
	option default '0'

config led 'led_lan3'
	option name 'LAN3'
	option sysfs 'green:lan3'
	option trigger 'none'
	option default '0'
```

It seems something like bug #14213. 

```
root@WAX202:/# find . -name brightness
./sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/leds/mt76-phy0/brightness
./sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/leds/mt76-phy1/brightness
./sys/devices/platform/leds/leds/green:wifi5g/brightness
./sys/devices/platform/leds/leds/green:wifi2g/brightness
./sys/devices/platform/leds/leds/orange:power/brightness
./sys/devices/platform/leds/leds/green:net/brightness
./sys/devices/platform/leds/leds/orange:lan3/brightness
./sys/devices/platform/leds/leds/orange:lan1/brightness
./sys/devices/platform/leds/leds/blue:net/brightness
./sys/devices/platform/leds/leds/green:power/brightness
./sys/devices/platform/leds/leds/blue:wifi5g/brightness
./sys/devices/platform/leds/leds/orange:lan2/brightness
./sys/devices/platform/leds/leds/blue:wifi2g/brightness
```

### OpenWrt version

r28427-6df0e3d02a

### OpenWrt release

24.10.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

Netgear WAX202

### Image kind

Official downloaded image

### Steps to reproduce

Can not find anymore green:lanX led names when defining LED rules on /cgi-bin/luci/admin/system/leds

### Actual behaviour

Led names green:lan1 green:lan2 and green:lan3 are missing, only orange:lan1 orange:lan2 and orange:lan3 are found for lan

### Expected behaviour

find also green:lan1 green:lan2 and green:lan3 in order to be able to shutdown green lights on LAN-leds

### Diffconfig

```text

```

### Terms

- [x] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2025-04-18:
Look at: https://github.com/openwrt/openwrt/commit/fc33c41c21362b7186aa051a2140623943fa3143

MightySlaytanic, 2025-04-19:
> Look at: https://github.com/openwrt/openwrt/commit/fc33c41c21362b7186aa051a2140623943fa3143

I can see that green:lanX have been replaced with orange:lanX but this means that I can not force off green leds, I can only force orange led on lan ports to be off or on, but green leds now keep blinking with network activity. Does it mean that we won't be able anymore to shutdown green leds?

neheb, 2025-04-20:
Nope. switch LED control is missing from the DSA driver. @arinc9 might have more information.

matrix200, 2025-04-25:
Same problem here.
I understand the reasoning given, but I think, the option to completely turn them off should still be available.